dresden-elektronik/deconz-rest-plugin #8102: _TZE204_ex3rcdha Tuya Zigbee mmwave Radar Detektor Luminanz

4MeInfo, 2025-01-23:
### Is there already an existing issue for this?

- [x] I have searched the existing issues and there is none for my device

### Product name

ZY-ZHPS01-5.8G

### Manufacturer

_TZE204_ex3rcdha

### Model identifier

TS0601

### Device type to add

Sensor

### Node info

[image]

### Endpoints and clusters

[image]

### Basic

[image]

### Further relevant clusters

###Green Power

[image]

###Groups

[image]

[image]

###OTAU

[image]

[image]

###Scenes

[image]

[image]

[image]

[image]

###Time

[image]

4MeInfo, 2025-01-23:
Got this from Aliexpress https://de.aliexpress.com/item/1005007334728544.html?spm=a2g0o.order_list.order_list_main.23.2bb05c5fFW2KkG&gatewayAdapt=glo2deu

Seemone, 2025-02-15:
What other information is needed?

Seemone, 2025-03-10:
Keeping this alive as no feedback has been given.

4MeInfo, 2025-03-13:
@Smanar can you help please ?

Smanar, 2025-03-14:
Can start with this DDF

```
{
  "schema": "devcap1.schema.json",
  "uuid": "eadb2b6f-aeb8-4695-923d-ba0eb42fd003",
  "manufacturername": "_TZE204_ex3rcdha",
  "modelid": "TS0601",
  "vendor": "Tuya",
  "product": "Tuya Zigbee mmwave Radar Detektor Luminanz",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_LIGHT_LEVEL_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0400"
      ],
      "items": [
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "refresh.interval": 86400,
          "read": {
            "at": "0x0001",
            "cl": "0x0000",
            "ep": 1,
            "fn": "zcl:attr"
          },
          "parse": {
            "at": "0x0001",
            "cl": "0x0000",
            "ep": 1,
            "fn": "zcl:attr",
            "script": "tuya_swversion.js"
          }
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/dark"
        },
        {
          "name": "config/tholddark"
        },
        {
          "name": "config/tholdoffset"
        },
        {
          "name": "state/daylight"
        },
        {
          "name": "state/lightlevel"
        },
        {
          "name": "state/lux",
          "read": {
            "fn": "none"
          },
          "parse": {
            "dpid": 12,
            "script": "../generic/illuminance_cluster/lux_to_lightlevel.js",
            "fn": "tuya"
          },
          "default": 0
        }
      ]
    },
    {
      "type": "$TYPE_PRESENCE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0xef00"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "refresh.interval": 86400,
          "read": {
            "at": "0x0001",
            "cl": "0x0000",
            "ep": 1,
            "fn": "zcl:attr"
          },
          "parse": {
            "at": "0x0001",
            "cl": "0x0000",
            "ep": 1,
            "fn": "zcl:attr",
            "script": "tuya_swversion.js"
          }
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/tuya_unlock"
        },
        {
          "name": "config/delay",
          "read": {
            "fn": "none"
          },
          "write": {
            "dpid": 104,
            "dt": "0x2b",
            "eval": "Item.val;",
            "fn": "tuya"
          },
          "parse": {
            "dpid": 104,
            "eval": "Item.val = Attr.val;",
            "fn": "tuya"
          },
          "default": 1
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "config/sensitivity",
          "range": [
            0,
            10
          ],
          "read": {
            "fn": "none"
          },
          "write": {
            "dpid": 105,
            "dt": "0x2b",
            "eval": "Item.val;",
            "fn": "tuya"
          },
          "parse": {
            "dpid": 105,
            "eval": "Item.val = Attr.val;",
            "fn": "tuya"
          },
          "default": 5
        },
        {
          "name": "state/presence",
          "read": {
            "fn": "none"
          },
          "parse": {
            "dpid": 101,
            "eval": "Item.val = Attr.val;",
            "fn": "tuya"
          },
          "default": false
        }
      ]
    }
  ]
}

```

For the Luminosity the DD need to be in the "tuya" folder.

4MeInfo, 2025-03-15:
The LightLevel works perfectly with this config, but presence is not detected.

Smanar, 2025-03-16:
Strange with what I have as info, the dpid used for occupancy is 101.
Nothing special on logs ?

4MeInfo, 2025-03-16:
Here is the API response 

> "2": {
>         "capabilities": {
>             "sleeper": false
>         },
>         "config": {
>             "on": true,
>             "reachable": true,
>             "tholddark": 12000,
>             "tholdoffset": 7000
>         },
>         "ddf_hash": null,
>         "ddf_policy": "latest_prefer_stable",
>         "etag": "af453f8d60ee67de10add110438d5f17",
>         "lastannounced": null,
>         "lastseen": "2025-03-16T19:11Z",
>         "manufacturername": "_TZE204_ex3rcdha",
>         "modelid": "TS0601",
>         "name": "LightLevel 2",
>         "nwkaddress": 51605,
>         "state": {
>             "dark": true,
>             "daylight": false,
>             "lastupdated": "2025-03-16T19:11:42.503",
>             "lightlevel": 11462,
>             "lux": 14
>         },
>         "swversion": "1.0.10",
>         "type": "ZHALightLevel",
>         "uniqueid": "a4:c1:38:b2:8b:68:08:e2-01-0400"
>     },
>     "3": {
>         "capabilities": {
>             "sleeper": false
>         },
>         "config": {
>             "delay": 1,
>             "on": true,
>             "reachable": true,
>             "sensitivity": 5
>         },
>         "ddf_hash": null,
>         "ddf_policy": "latest_prefer_stable",
>         "etag": "97b54d7e53c498741f7c64a888ccc75b",
>         "lastannounced": null,
>         "lastseen": "2025-03-16T19:11Z",
>         "manufacturername": "_TZE204_ex3rcdha",
>         "modelid": "TS0601",
>         "name": "Presence 3",
>         "nwkaddress": 51605,
>         "state": {
>             "lastupdated": "2025-03-16T19:10:31.031",
>             "presence": false
>         },
>         "swversion": "1.0.10",
>         "type": "ZHAPresence",
>         "uniqueid": "a4:c1:38:b2:8b:68:08:e2-01-ef00"
>     }
> 

And here the log :

> 20:15:06:190 Daylight now: nauticalDusk, status: 220, daylight: 0, dark: 1
> 20:15:07:281 [Master] config changed, read parameters
> 20:15:07:282 [Master] read param with arg 0x18
> 20:15:07:282 [Master] read param with arg 0x13
> 20:15:07:282 [Master] read param with arg 0x13
> 20:15:07:282 [Master] read param with arg 0x13
> 20:15:07:297 Device protocol version: 0x010C
[... 265 lines cut ...]
> const lux = Attr.val;
> var ll = 0;
> 
> if (lux > 0 && lux < 0xffff) {
>     ll = Math.round(10000 * Math.log10(lux) + 1);
> }
> 
> R.item('state/lightlevel').val = ll;
> R.item('state/dark').val = ll <= tholddark;
> R.item('state/daylight').val = ll >= tholddark + tholdoffset;
> 
> Item.val = lux;
>  --> 23
> 20:18:01:882 TY_DATA_REPORT: seq 8, dpid: 0x0C, type: 0x02, length: 4, val: 23
> 20:18:01:882 Websocket 192.168.178.70:57185 send message: {"attr":{"ddf_hash":null,"ddf_policy":"latest_prefer_stable","lastannounced":null,"lastseen":"2025-03-16T19:18Z","manufacturername":"_TZE204_ex3rcdha","modelid":"TS0601","name":"Presence 3","nwkaddress":51605,"swversion":"1.0.10","type":"ZHAPresence","uniqueid":"a4:c1:38:b2:8b:68:08:e2-01-ef00"},"e":"changed","id":"3","r":"sensors","t":"event","uniqueid":"a4:c1:38:b2:8b:68:08:e2-01-ef00"} (ret = 391)
> 20:18:01:883 Websocket 192.168.178.70:57185 send message: {"attr":{"ddf_hash":null,"ddf_policy":"latest_prefer_stable","lastannounced":null,"lastseen":"2025-03-16T19:18Z","manufacturername":"_TZE204_ex3rcdha","modelid":"TS0601","name":"LightLevel 2","nwkaddress":51605,"swversion":"1.0.10","type":"ZHALightLevel","uniqueid":"a4:c1:38:b2:8b:68:08:e2-01-0400"},"e":"changed","id":"2","r":"sensors","t":"event","uniqueid":"a4:c1:38:b2:8b:68:08:e2-01-0400"} (ret = 395)
> 20:18:01:883 Websocket 192.168.178.70:57185 send message: {"e":"changed","id":"2","r":"sensors","state":{"dark":false,"daylight":false,"lastupdated":"2025-03-16T19:18:01.882","lightlevel":13618,"lux":23},"t":"event","uniqueid":"a4:c1:38:b2:8b:68:08:e2-01-0400"} (ret = 203)
> 20:18:06:190 Daylight now: nauticalDusk, status: 220, daylight: 0, dark: 1
> 20:18:10:690 sql exec SELECT conf FROM zbconf ORDER BY rowid desc limit 1
> 20:18:16:190 Daylight now: nauticalDusk, status: 220, daylight: 0, dark: 1
> 20:18:17:190 DB save zll database items 0x00000081
> 20:18:17:190 DB saved in 0 ms
> 20:18:19:834 Websocket 192.168.178.70:57185 send message: {"attr":{"ddf_hash":null,"ddf_policy":"latest_prefer_stable","lastannounced":null,"lastseen":"2025-03-16T19:18Z","manufacturername":"_TZE204_ex3rcdha","modelid":"TS0601","name":"Smart plug 1","nwkaddress":51605,"swversion":null,"type":"Smart plug","uniqueid":"a4:c1:38:b2:8b:68:08:e2-01"},"e":"changed","id":"1","r":"lights","t":"event","uniqueid":"a4:c1:38:b2:8b:68:08:e2-01"} (ret = 377)
> 20:18:26:189 Daylight now: nauticalDusk, status: 220, daylight: 0, dark: 1
> 20:18:36:189 Daylight now: nauticalDusk, status: 220, daylight: 0, dark: 1

Smanar, 2025-03-17:
Better to use Pastebin, to avoid huge post like his one, but you have catch the good part 
```

20:17:17:636 TY_DATA_REPORT: seq 1, dpid: 0x65, type: 0x04, length: 1, val: 0
20:17:17:636 TY_DATA_REPORT: seq 1, dpid: 0x65, type: 0x04, length: 1, val: 0
20:17:17:636 a4:c1:38:b2:8b:68:08:e2-01-ef00/state/presence expression: Item.val = Attr.val; --> 0
20:17:17:637 TY_DATA_REPORT: seq 1, dpid: 0x65, type: 0x04, length: 1, val: 0
20:17:17:637 TY_DATA_REPORT: seq 1, dpid: 0x65, type: 0x04, length: 1, val: 0
20:17:17:637 Websocket 192.168.178.70:57185 send message: {"e":"changed","id":"3","r":"sensors","state":{"lastupdated":"2025-03-16T19:17:17.636","presence":false},"t":"event","uniqueid":"a4:c1:38:b2:8b:68:08:e2-01-ef00"} (ret = 162)
```

0x65 = 101.
If the sensor detect something it need to make the same report but with different value.
Or If I m wrong it ca, use other dpid

4MeInfo, 2025-03-22:
Hi @Smanar,
I used the last days to test the config. It worked but the presence state must be inverted.
Can you send me please the updated json with inverted state?
Thanks a lot for your help

4MeInfo, 2025-03-23:
So this is the working DDF
[_TZE204_ex3rcdha.json](https://github.com/user-attachments/files/19407670/_TZE204_ex3rcdha.json)

```
{
  "schema": "devcap1.schema.json",
  "uuid": "eadb2b6f-aeb8-4695-923d-ba0eb42fd003",
  "manufacturername": "_TZE204_ex3rcdha",
  "modelid": "TS0601",
  "vendor": "Tuya",
  "product": "Tuya Zigbee mmwave Radar Detektor Luminanz",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_LIGHT_LEVEL_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0400"
      ],
      "items": [
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "refresh.interval": 86400,
          "read": {
            "at": "0x0001",
            "cl": "0x0000",
            "ep": 1,
            "fn": "zcl:attr"
          },
          "parse": {
            "at": "0x0001",
            "cl": "0x0000",
            "ep": 1,
            "fn": "zcl:attr",
            "script": "tuya_swversion.js"
          }
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/dark"
        },
        {
          "name": "config/tholddark"
        },
        {
          "name": "config/tholdoffset"
        },
        {
          "name": "state/daylight"
        },
        {
          "name": "state/lightlevel"
        },
        {
          "name": "state/lux",
          "read": {
            "fn": "none"
          },
          "parse": {
            "dpid": 12,
            "script": "../generic/illuminance_cluster/lux_to_lightlevel.js",
            "fn": "tuya"
          },
          "default": 0
        }
      ]
    },
    {
      "type": "$TYPE_PRESENCE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0xef00"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "refresh.interval": 86400,
          "read": {
            "at": "0x0001",
            "cl": "0x0000",
            "ep": 1,
            "fn": "zcl:attr"
          },
          "parse": {
            "at": "0x0001",
            "cl": "0x0000",
            "ep": 1,
            "fn": "zcl:attr",
            "script": "tuya_swversion.js"
          }
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/tuya_unlock"
        },
        {
          "name": "config/delay",
          "read": {
            "fn": "none"
          },
          "write": {
            "dpid": 104,
            "dt": "0x2b",
            "eval": "Item.val;",
            "fn": "tuya"
          },
          "parse": {
            "dpid": 104,
            "eval": "Item.val = Attr.val;",
            "fn": "tuya"
          },
          "default": 1
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "config/sensitivity",
          "range": [
            0,
            10
          ],
          "read": {
            "fn": "none"
          },
          "write": {
            "dpid": 105,
            "dt": "0x2b",
            "eval": "Item.val;",
            "fn": "tuya"
          },
          "parse": {
            "dpid": 105,
            "eval": "Item.val = Attr.val;",
            "fn": "tuya"
          },
          "default": 5
        },
        {
          "name": "state/presence",
          "read": {
            "fn": "none"
          },
          "parse": {
            "dpid": 101,
            "eval": "Item.val = !Attr.val;",
            "fn": "tuya"
          },
          "default": false
        }
      ]
    }
  ]
}

```

Mimiix, 2025-03-23:
@Smanar can you make a PR?

Smanar, 2025-03-23:
Done https://github.com/dresden-elektronik/deconz-rest-plugin/pull/8144

closed 2025-03-26 by manup