dresden-elektronik/deconz-rest-plugin #8103: _TZ3000_wut53hfm  TSO203 Door Window Contact

pierrecinema, 2025-01-24:
### Is there already an existing issue for this?

- [x] I have searched the existing issues and there is none for my device

### Product name

Door Contact Avatto?

### Manufacturer

_TZ3000_wut53hfm

### Model identifier

TSO203

### Device type to add

Sensor

### Node info

[image]

### Endpoints and clusters

[image]

### Basic

[image]

### Further relevant clusters

### Power Configuration

### On/Off

### Level Control

### Color Control

### Thermostat

### Simple Metering

### Electrical Measurement

### Any other cluster of relevance/interest

Smanar, 2025-01-24:
Hello, nothing specil on your device, can you just try to adding it to this DDF https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/devices/tuya/_TZ3000_TS0203_door_sensor.json
Need to add the manufacture name AND the model id, even they are same, then restart deconz.

pierrecinema, 2025-01-29:
It works

Thanks

Smanar, 2025-01-29:
Do you want to make a PR to add it, or want I make it ?

pierrecinema, 2025-01-29:
Can you please do that. Thanks

Smanar, 2025-01-30:
Done > https://github.com/dresden-elektronik/deconz-rest-plugin/pull/8108

closed 2025-02-05 by manup