dresden-elektronik/deconz-rest-plugin #8105: _TZE200_n8dljorx TS0601 Door Window Contact Sensor

Jan-Unger, 2025-01-28:
### Is there already an existing issue for this?

- [x] I have searched the existing issues and there is none for my device

### Product name

Tuya

### Manufacturer

_TZE200_n8dljorx

### Model identifier

TS0601

### Device type to add

Sensor

### Node info

[image]

### Endpoints and clusters

[image]

### Basic

[image]

### Further relevant clusters

### Power Configuration

### On/Off

### Level Control

### Color Control

### Thermostat

### Simple Metering

### Electrical Measurement

### Any other cluster of relevance/interest

csp80, 2025-02-07:
I also have problems with the model TS0601_TZE200_n8dljorx 
closing and opening the sensor works but unfortunately the light intensity is not displayed.

Strangely enough, with model TS0601_TZE200_pay2byax
everything works without any problems.

[image]
[image]

Checkdat, 2025-02-08:
Same issue here.

Smanar, 2025-02-09:
Hello you have added the model "_TZE200_n8dljorx" to the DDF _TZE200_pay2byax_openclose_sensor.json ?

The TS0601_TZE200_pay2byax use Dpid 0x01 for contact, 0x02 for battery and 0x101 ofr luminosity.
But the _TZE200_n8dljorx don't have luminosity from Z2M and probbly need the tuya uncloc, sequence.

brientim, 2025-05-03:
Request this be reopened and add "_TZE200_n8dljorx" to... 
https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/devices/tuya/_TZE200_pay2byax_openclose_sensor.json

Smanar, 2025-05-04:
And adding the model "_TZE200_n8dljorx" to the DDF _TZE200_pay2byax_openclose_sensor.json  is not enought to make device working ?
As this one don't have luminosity, the same DDF can be used.

brientim, 2025-05-04:
Yes, adding the model "_TZE200_n8dljorx" to the DDF _TZE200_pay2byax_openclose_sensor.json will enable the full functionality of the sensor model without luminosity. 

In my opinion the correct appending "_TZE200_n8dljorx" to the extant DDF is the most appropriate outcome.

Smanar, 2025-05-04:
Same for me, do you want to make the PR or want I make it ?

brientim, 2025-05-04:
Can you please raise PR

brientim, 2025-05-04:
Bundle tested and working correctly. The attributes with "Null" values has been raised as a distinct issue #8208 and this does not impact operationally. 

                    {
    "capabilities": {
        "sleeper": true
    },
    "config": {
        "battery": 100,
        "on": true,
        "reachable": true
    },
    "ddf_hash": "b657ad79ddd58a88fbaf8d7c037a060d53f444b600f269b03fa2d8101473c70a",
    "ddf_policy": "latest_prefer_stable",
    "etag": "42c7234902b0d5e99f23cadff3bbff09",
    "lastannounced": null,
    "lastseen": "2025-05-04T22:04Z",
    "manufacturername": "_TZE200_n8dljorx",
    "modelid": "TS0601",
    "name": "TS0601 (2)",
    "nwkaddress": 40837,
    "state": {
        "lastupdated": "2025-05-04T22:05:02.885",
        "open": false
    },
    "swversion": null,
    "type": "ZHAOpenClose",
    "uniqueid": "a4:c1:38:53:8c:f6:fb:61-01-ef00",
    "zonetype": null
}

closed 2025-05-06 by manup