dresden-elektronik/deconz-rest-plugin #8107: Save source routing settings

mn-90, 2025-01-30:
### Does the issue really belong here?

- [x] I definitively want to report a bug within deCONZ or its REST-API

### Is there already an existing issue for this?

- [x] I have searched the existing issues and there is none for the bug at hand

### Describe the bug

When running deconz within home assistant and a conbee II stick, i can activate source routing via the ui, amd it works perfectly fine. After a restart of ha, the source routing box is unchecked again, so the setting seems to be lost during restart.

### Steps to reproduce the behavior

1. Open deconz within home assistant
2. activate source routing
3. restart ha
4. check source routing setting

### Expected behavior

The source routing setting should be persistent

### Environment

- Host system: Raspberry Pi
- Running method: Home Assistent deCONZ Add-on
- Firmware version:26720700
- deCONZ version (not Home assistant Addon version!): 2.28.1
- Device:  ConBee II
- Do you use an USB extension cable: yes
- Is there any other USB or serial devices connected to the host system? If so: Which? No

mtressl, 2025-02-11:
Check the file permissions of your `config.ini` file, there might be problems so the changes through the UI can not be saved.

mn-90, 2025-02-11:
I never touched this file, so if this is created by the app, there should not be any permission issues.

What i would like to understand: when doing a restart of home assistant, will deconz be properly closed? I found those settings will not be written when i Change them, but only when the controller is destructed. Can that be the root cause?

https://github.com/dresden-elektronik/deconz/issues/8

manup, 2025-02-17:
Hi thanks for reporting. I'm closing this issue so we can track it in <https://github.com/dresden-elektronik/deconz/issues/8>

closed 2025-02-17 by manup