dresden-elektronik/deconz-rest-plugin #8119: LEDVANCE PLUG EU EM T (with metering)

fredl99, 2025-02-11:
### Is there already an existing issue for this?

- [x] I have searched the existing issues and there is none for my device

### Product name

SMART+ PLUG EU 

### Manufacturer

LEDVANCE

### Model identifier

PLUG EU EM T

### Device type to add

None

### Node info

[image]

### Endpoints and clusters

[image]

### Basic

[image]

### Further relevant clusters

### On/Off
[image]

### Simple Metering
[image]

### Electrical Measurement
[image]
[image]

### Any other cluster of relevance/interest

fredl99, 2025-02-11:
This is the new 3rd generation of these devices. Now with metering.

fredl99, 2025-02-12:
I just saw request #8082 which seems to be identical hardware. I was mislead by the 'COMPACT' model name there. Sorry for the confusion!

However, as I'm about to replace my 2nd generation plugs by these newer ones in order to get measurements, I'd appreciate any help. As mentioned in issue #8082, the new devices are supported by zigbee2mqtt already. Maybe that could be an additional source of information. Unfortunately I'm not experienced enough to solve it on my own. Thanks in advance!

Lempomat, 2025-03-06:
Same Problem. I don't see energy meter. How can I add it?