dresden-elektronik/deconz-rest-plugin #8120: Pilot wire heating module - Powernity PO-BOCO-ELEC

Sekenfo, 2025-02-12:
### Is there already an existing issue for this?

- [x] I have searched the existing issues and there is none for my device

### Product name

Powernity - PO-BOCO-ELEC

### Manufacturer

_TZE204_d6i25bwg

### Model identifier

TS0601

### Device type to add

Other

### Node info

[image]

### Endpoints and clusters

[image]

### Basic

[image]

### Further relevant clusters

### Power Configuration

### On/Off

### Level Control

### Color Control

### Thermostat

### Simple Metering

### Electrical Measurement

### Any other cluster of relevance/interest

Sekenfo, 2025-02-12:
There have been requests to add hardware support, but is it possible for a beginner to learn from examples or other similar devices ?

Smanar, 2025-02-13:
The problem is there is not similar device for this one ^^.
Standard pilot wire are not easy to add, so if they are from tuya ....

Sekenfo, 2025-02-13:
Okay, so I can't expect a DDF file for deconz ?
Or should I wait for a “professional” (I have no idea what I'm talking about ;-) ) to come by ?

closed 2025-03-15 by github-actions[bot]