dresden-elektronik/deconz-rest-plugin #8122: Aqara Smoke Detector SD-S01E

TheBisk, 2025-02-12:
### Is there already an existing issue for this?

- [x] I have searched the existing issues and there is none for my device

### Product name

Aqara Smoke Detector SD-S01E

### Manufacturer

Aqara Lumi

### Model identifier

SD-S01E

### Device type to add

Sensor

### Node info

[image]

### Endpoints and clusters

[image]

### Basic

[image]

### Further relevant clusters

### Identify
[image]

### Power configuration

[image]

### Lumi

[image]

[image]

[image]

### OTAU

[image]

TheBisk, 2025-02-12:
Hi, 
i tryed this DDF https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5864#issuecomment-1500942502,

TheBisk, 2025-02-12:
[image]

aramcap, 2025-04-21:
Any response? I've recently bought it and I have the same problem: Deconz see the device but Phoscon App, between API, no. Please, reopen the issue!

aramcap, 2025-04-23:
@ebaauw can you help us?

gitfvb, 2025-05-05:
That would be very helpful, if this device could be supported in future. I also bought one.

3sdg, 2025-05-06:
Is it normal that this issue is closed but the device is not supported yet. There were several releases between the device request and today, but not supported.
Can someone reopen the issue?

Smanar, 2025-05-06:
If someone want to make a try, it seem this device don't use the IAS cluster, but only the Lumi one.
For exemple for the smoke detection the code can be on the DDF

```
        {
          "name": "state/fire",
          "parse": {
            "fn": "zcl:attr",
            "ep": 1,
            "cl": "0xFCC0",
            "mf": "0x115F",
            "at": "0x013a",
            "eval": "Item.val = Attr.val;"
          }
        },
```

And this cluster generaly don't need bind/report.

gitfvb, 2025-05-08:
I am happy to test @Smanar , but haven't done this before. Not really sure how to do that.

Smanar, 2025-05-09:
Depend of your OS, you just need to create a text file called what_you_want.json with the DDF contain in a "devices" folder.
https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/DDF-cheat-sheet
But on some OS like docker, need more gymastic.

Here a more complete DDF.
```

{
  "schema": "devcap1.schema.json",
  "manufacturername": "$MF_LUMI",
  "modelid": "lumi.sensor_smoke.acn03",
  "vendor": "Xiaomi Mi",
  "product": "Aqara Smoke Detector SD-S01E",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_FIRE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0xfc00"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "parse": {
            "at": "0xff01",
            "ep": 1,
            "fn": "xiaomi:special",
            "idx": "0x08",
            "script": "xiaomi_swversion.js"
          },
          "read": {
            "fn": "none"
          }
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "parse": {
            "at": "0xff01",
            "fn": "xiaomi:special",
            "idx": "0x01",
            "script": "xiaomi_battery.js"
          }
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/fire",
          "parse": {
            "fn": "zcl:attr",
            "ep": 1,
            "cl": "0xFCC0",
            "mf": "0x115F",
            "at": "0x013a",
            "eval": "Item.val = Attr.val;"
          },
          "read": {
            "fn": "none"
          }
        },
        {
          "name": "state/lastupdated"
        }
      ]
    }
  ]
}
```

gitfvb, 2025-05-09:
Thanks @Smanar 

Seems doable with my raspi (without containers). Will try to achieve that over the weekend.

Smanar, 2025-05-09:
So yes, will be easy for you, just choose one of the 2 folders "devices".
Then restart deconz, DDF are loaded only at start. After you device will be reconised if you try to include it.

But IDK if it will work ^^.

gitfvb, 2025-05-12:
Hi @Smanar , first connection worked like a charm. I have tested a few things, but noticed those points:

- The battery power is shown as 0 %
- The lastupdated state is shown at `none`
- It sent regular updates until yesterday night. I will wait a few more days to see if it still sends heartbeats

Here can you see a screenshot from iobroker:

[image]

And here one from phoscon:

[image]

So long story short, the DDF is not perfect yet, but a good start. I am still trying to "decrypt" the DDF to understand it better. But I am making progress :-)

In the next days, I will try to simulate a fire alarm. Then I can report more.

Smanar, 2025-05-13:
Yeah for the battery it seem the device use the classic cluster so can remove

```
        {
          "name": "config/battery",
          "parse": {
            "at": "0xff01",
            "fn": "xiaomi:special",
            "idx": "0x01",
            "script": "xiaomi_battery.js"
          }
        },
```

And use instead

```
        {
          "name": "config/battery",
          "refresh.interval": 43265,
          "parse": {
            "at": "0x0020",
            "cl": "0x0001",
            "ep": 1,
            "eval": "const vmin = 44; const vmax = 60; var bat = Attr.val; if (bat > vmax) { bat = vmax; } else if (bat < vmin) { bat = vmin; } bat = ((bat - vmin) / (vmax - vmin)) * 100; if (bat > 100) { bat = 100; } else if (bat <= 0) { bat = 1; } Item.val = bat;",
            "fn": "zcl:attr"
          },
          "read": {
            "at": "0x0020",
            "cl": "0x0001",
            "ep": 1,
            "fn": "zcl:attr"
          }
        },
```

And add binding
```
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0020",
          "dt": "0x20",
          "min": 3600,
          "max": 43200,
          "change": "0x00000002"
        }
      ]
    },
```

The device can detect "fire" ? Because the "lastupdated" need to be updated on fire detection.

gitfvb, 2025-05-16:
Ah, thanks @Smanar , after updating the ddf and reconnect of the smoke detector, I am getting the battery power level:

[image]

And this is how it looks like in iobroker

[image]

I still need to test it with smoke. But it looks promising. Also new starts of the raspi recognised the smoke detector again without problems with the fire state I have read about in other issues.

Smanar, 2025-05-17:
Can try too

```
        {
          "name": "config/selftest",
          "parse": {
            "fn": "zcl:attr",
            "ep": 1,
            "cl": "0xFCC0",
            "mf": "0x115F",
            "at": "0x0127",
            "eval": "Item.val = Attr.val;"
          },
          "read": {
            "fn": "none"
          },
          "write": {
            "at": "0x0127",
            "cl": "0xFCC0",
            "dt": "0x10",
            "ep": 1,
            "eval": "Item.val",
            "fn": "zcl:attr",
            "mf": "0x115F"
          },
          "values": [
            [
              true,
              "Initiate self-test"
            ],
            [
              false,
              "Self-test not initiated"
            ]
          ]
        },
```

gitfvb, 2025-05-17:
That looks promising @Smanar 

Will test this evening. Do you see something to trigger an alarm? That way I could trigger multiple devices when smoke is detected on one.

Smanar, 2025-05-17:
I never try it since this PR https://github.com/dresden-elektronik/deconz-rest-plugin/pull/7839
But can make a try, for that you need to create a new entry in the DDF to have a "light" entry and use state/alert=select

```
    {
      "type": "$TYPE_WARNING_DEVICE",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "parse": {
            "at": "0xff01",
            "ep": 1,
            "fn": "xiaomi:special",
            "idx": "0x08",
            "script": "xiaomi_swversion.js"
          },
          "read": {
            "fn": "none"
          }
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/alert",
          "read": {
            "fn": "none"
          },
          "write": {
            "at": "0x013d",
            "cl": "0xFCC0",
            "dt": "0x10",
            "ep": 1,
            "eval": "if (Item.val=='select') { 1 } else if (Item.val=='none') { 0 }",
            "fn": "zcl:attr",
            "mf": "0x115F"
          },
          "parse": {
            "fn": "zcl:attr",
            "ep": 1,
            "cl": "0xFCC0",
            "mf": "0x115F",
            "at": "0x013d",
            "eval": "Item.val = Attr.val;"
          },
        },
        {
          "name": "state/reachable"
        }
      ]
    }
```

scaarup, 2025-06-04:
I am getting "import failed" all the time:

[image]

On HAOS with the deconz addon. Has anyone managed to get it to work?

Thank you

gitfvb, 2025-06-04:
Hi @scaarup , this is the latest json I am testing with. I have imported it directly via winscp over ssh

Some of the functionality is not working yet, but we cannot be far off.

```json
{
  "schema": "devcap1.schema.json",
  "manufacturername": "$MF_LUMI",
  "modelid": "lumi.sensor_smoke.acn03",
  "vendor": "Xiaomi Mi",
  "product": "Aqara Smoke Detector SD-S01E",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_FIRE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0xfc00"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "parse": {
            "at": "0xff01",
            "ep": 1,
            "fn": "xiaomi:special",
            "idx": "0x08",
            "script": "xiaomi_swversion.js"
          },
          "read": {
            "fn": "none"
          }
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "refresh.interval": 43265,
          "parse": {
            "at": "0x0020",
            "cl": "0x0001",
            "ep": 1,
            "eval": "const vmin = 20; const vmax = 30; var bat = Attr.val; if (bat > vmax) { bat = vmax; } else if (bat < vmin) { bat = vmin; } bat = ((bat - vmin) / (vmax - vmin)) * 100; if (bat > 100) { bat = 100; } else if (bat <= 0) { bat = 1; } Item.val = bat;",
            "fn": "zcl:attr"
          },
          "read": {
            "at": "0x0020",
            "cl": "0x0001",
            "ep": 1,
            "fn": "zcl:attr"
          }
        },
        {
          "name": "config/selftest",
          "parse": {
            "fn": "zcl:attr",
            "ep": 1,
            "cl": "0xFCC0",
            "mf": "0x115F",
            "at": "0x0127",
            "eval": "Item.val = Attr.val;"
          },
          "read": {
            "fn": "none"
          },
          "write": {
            "at": "0x0127",
            "cl": "0xFCC0",
            "dt": "0x10",
            "ep": 1,
            "eval": "Item.val",
            "fn": "zcl:attr",
            "mf": "0x115F"
          },
          "values": [
            [
              true,
              "Initiate self-test"
            ],
            [
              false,
              "Self-test not initiated"
            ]
          ]
        },
        {
          "bind": "unicast",
          "src.ep": 1,
          "cl": "0x0001",
          "report": [
            {
              "at": "0x0020",
              "dt": "0x20",
              "min": 3600,
              "max": 43200,
              "change": "0x00000002"
            }
          ]
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/fire",
          "parse": {
            "fn": "zcl:attr",
            "ep": 1,
            "cl": "0xFCC0",
            "mf": "0x115F",
            "at": "0x013a",
            "eval": "Item.val = Attr.val;"
          },
          "read": {
            "fn": "none"
          }
        },
        {
          "name": "state/lastupdated"
        }
      ]
    },
    {
      "type": "$TYPE_WARNING_DEVICE",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "parse": {
            "at": "0xff01",
            "ep": 1,
            "fn": "xiaomi:special",
            "idx": "0x08",
            "script": "xiaomi_swversion.js"
          },
          "read": {
            "fn": "none"
          }
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/alert",
          "read": {
            "fn": "none"
          },
          "write": {
            "at": "0x013d",
            "cl": "0xFCC0",
            "dt": "0x10",
            "ep": 1,
            "eval": "if (Item.val=='select') { 1 } else if (Item.val=='none') { 0 }",
            "fn": "zcl:attr",
            "mf": "0x115F"
          },
          "parse": {
            "fn": "zcl:attr",
            "ep": 1,
            "cl": "0xFCC0",
            "mf": "0x115F",
            "at": "0x013d",
            "eval": "Item.val = Attr.val;"
          }
        },
        {
          "name": "state/reachable"
        }
      ]
    }
  ]
}
```

scaarup, 2025-06-05:
Thank you @gitfvb - now I got something :)

Smanar, 2025-06-05:
So sending "select" to "state/alert" don't trigger the siren ?

scaarup, 2025-06-10:
> So sending "select" to "state/alert" don't trigger the siren ?

That is correct.
Actually it just turns off again: 

[image]

Smanar, 2025-06-16:
The xml is the general.xml file > [general.xml](https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/general.xml)
It's the file used by deconz to display attribute and cluster
If you select the cluster 0xFFC0, you will found Xiaomi/Aqara attribute and command

But you miss the one for this device

You are in this part (around line 5789)
```
    <!-- LUMI -->
    <cluster id="0xfcc0" name="Lumi specific" mfcode="0x115f">
      <description>Lumi specific attributes.</description>
```

You can add this line around line 5849 (after a deconz restart)

`<attribute id="0x0014B" name="linkage alarm " type="u8" mfcode="0x115f" access="rw" required="m"></attribute>`

To have it visible in the GUI.

IDK your OS but on normal OS the file is `/usr/share/deCONZ/zcl/general.xml`

closed 2025-07-15 by github-actions[bot]

gitfvb, 2025-07-20:
Thanks @Smanar , should the switch appear on this screen after a restart with `sudo systemctl restart deconz`?

[image]

And should it be

```
<attribute id="0x0014B" name="linkage alarm " type="u8" mfcode="0x115f" access="rw" required="m"></attribute>
```

or 

```
<attribute id="0x014B" name="linkage alarm" type="u8" mfcode="0x115f" access="rw" required="m"></attribute>
```

?

Smanar, 2025-07-21:
The switch ? you mean the siren ? IDK if phoscon support them, if you haven't third app you can take a look direclty in the API using Phoscon/help/API Information/light (for the siren)

The problem is to trigger the siren alone.
There is this comment on another forum
`During the absence of a smoke alarm, the buzzer can be manually alarmed ("alarm") and disalarmed ("mute"), but for this "linkage_alarm" option must be enabled`

So need to make a try with this option "off" and "on".

The difference is only the space ? So it's just cosmetic, don't worry for it, but you need to restart deconz after having edited the xml.
The fonction will be visible on the GUI after when you select the cluster 0xfcc0

Smanar, 2025-10-23:
On the GUI
On the node
On Cluster 0xfcc0 "Lumi specific"
On Attribute list (the table)

I can share a more complete code if you are not sure where to put

`<attribute id="0x014B" name="linkage alarm" type="u8" mfcode="0x115f" access="rw" required="m"></attribute>`

Smanar, 2025-10-25:
Your device is probably not reconised, so it will be not visible in Phoscon/Domoticz but it need to be visible on Deconz, the GUI application (not a webapp).
With a name like 0xXXXX  = not reconised.

Are you using the DDF, as the actual deconz version don't have support for this device.

dbocart, 2025-10-26:
yes is not connect, The connect button is not clikable

and is not docker installation

i add smoke.json on /usr/share/deCONZ/devices/xiaomi but always not detect (after restart pi)

i am on Xserver via ssh (mobaterme), 

[image]

[image]

[image]

For little history, at start,  i have domoticz on PI1b since more than 10 years,i add zigbee with deconz since many year now, there is 2 years, domoticz don't work after update, pi1b is too old for that, after many time, domoticz was ok but deconz was ko, i have migrated deconz on another pi (pi3b), now domoticz is always on pi1b and deconz on pi3b, but it's work since 2 years without issue
i have the issue only for add this smoke detector

Smanar, 2025-10-26:
>yes is not connect, The connect button is not clikable

Take care, there is something not normal. If Deconz is not connected you can't have your zigbee network working, I suspect you are using 2 deconz instance, for exemple the headless and the second one with the GUI. As the conbee is "reserved" by the first one, you can't use it too for the second one.

>The button do not start Alarm, it's possible to have that or not ?

It's the actual issue, there is a "warning device" entry, but it don't work.
Probably about the attribute 0x014B.
We need someone with a working GUI to make test with different setting for this attribute.

Smanar, 2025-10-28:
But I m reading the doc from Z2M
https://www.zigbee2mqtt.io/devices/JY-GZ-01AQ.html

There is no command to trigger the siren ?

Edit:
Found it

`During the absence of a smoke alarm, the buzzer can be manually alarmed ("alarm") and disalarmed ("mute"), but for this "linkage_alarm" option must be enabled. Value will not be published in the state. It's not possible to read (/get) this value. To write (/set) a value publish a message to topic zigbee2mqtt/FRIENDLY_NAME/set with payload {"buzzer": NEW_VALUE}. The possible values are: mute, alarm`

hermsi94, 2026-01-18:
I was able to paste the JSON into the folder and the smoke detector was detected. It is also displayed under Lights as a switchable device, but manual triggering is not yet possible/nothing happens. I am running the GUI version. Can I help in any way to trigger manual activation/specify register values?

Manual triggering would be helpful in order to activate other smoke detectors, e.g., all those on the same floor or in the building, when one smoke detector is triggered.

Smanar, 2026-01-18:
> During the absence of a smoke alarm, the buzzer can be manually alarmed ("alarm") and disalarmed ("mute"), but for this "linkage_alarm" option must be enabled. Value will not be published in the [...]

First enable linkage alarm
`<attribute id="0x014B" name="linkage alarm" type="u8" mfcode="0x115f" access="rw" required="m"></attribute>`

Then play with Test Manualx (with value 15361 or 15360) or Test ManualY (with value 0 or 1)
Can ask value first, to see defaut one, before trying to change them.

hermsi94, 2026-01-18:
> First enable linkage alarm `<attribute id="0x014B" name="linkage alarm" type="u8" mfcode="0x115f" access="rw" required="m"></attribute>`
>
> Then play with Test Manualx (with value 15361 or 15360) or [...]

0x013E with 15360 or 15361 can be written, but has no effect. Strangely, it could not be read: "reading failed"

0x0126 can be written, but nothing happens. This register can also be read.

What surprised me is that reading (0x0126) or writing (0x0126 & 0x013E) often did not work and only after removing and reinserting the battery could the register be set, then again not, so very randomly (or can they only be set within a certain period of time after startup?). Is this normal for Zigbee devices? 

In any case, I tried all 4 combinations and it didn't work.

I used the upper read/write buttons. I don't know what the lower ones do (I'm not very familiar with the GUI^^)
[image]