dresden-elektronik/deconz-rest-plugin #8128: ZemiSmart ZRM4 (Tuya clone TZ3000_TS0044)

brientim, 2025-03-01:
### Is there already an existing issue for this?

- [x] I have searched the existing issues and there is none for my device

### Product name

ZemiSmart ZRM4 4 button remote

### Manufacturer

_TZ3000_xwuveizv, _TZ3000_wkai4ga5

### Model identifier

TS0044

### Device type to add

Switch

### Node info

The ZemiSmart ZRM4 4 button remote is another Tuya clone with various Manufacture Names depending on the batch. 
It has the same capabilities and functions as detailed in tuya/_TZ3000_TS0044_4gang_remote.json

This request covers two additions: 
_TZ3000_xwuveizv, 
_TZ3000_wkai4ga5

[image]

### Endpoints and clusters

As per tuya/_TZ3000_TS0044_4gang_remote.json

[image]

### Basic

[image]

### Further relevant clusters

[image]

### Power Configuration

### On/Off

### Level Control

### Color Control

### Thermostat

### Simple Metering

### Electrical Measurement

### Any other cluster of relevance/interest

brientim, 2025-03-10:
@Smanar this is an update to tuya/_TZ3000_TS0044_4gang_remote.json 
I note that there was another PR raised to include additional devices support as detailed in PR8134 which could be merged if this these two _TZ3000_xwuveizv, _TZ3000_wkai4ga5 were included...
Reference https://github.com/dresden-elektronik/deconz-rest-plugin/pull/8134

tadeaustria, 2025-03-11:
Shall I add them to the PR?

brientim, 2025-03-11:
> Shall I add them to the PR?

That would be good to rap them up in the one PR

brientim, 2025-03-11:
Dan,The PR for the switches has been merged and should be in the next release. ZemiSmart ZRM4 (Tuya clone TZ3000_TS0044) · Issue #8128 · dresden-elektronik/deconz-rest-plugingithub.comRegardsTim Brien