dresden-elektronik/deconz-rest-plugin #8129: Ubisys R0

jan666, 2025-03-03:
### Is there already an existing issue for this?

- [x] I have searched the existing issues and there is none for my device

### Product name

R0

### Manufacturer

Ubisys

### Model identifier

R0 (5501)

### Device type to add

Other

### Node info

This is just a router, similar to the IKEA Signal Repeater. https://www.ubisys.de/en/products/additional-options/router-r0/. But it can be used to used the Ubisys App via Bluetooth to control Ubisys devices via Bluetooth.

### Endpoints and clusters

[image]

### Basic

[image]

### Further relevant clusters

### OTAU

[image]

jan666, 2025-03-25:
Keeping active

mattreim, 2025-04-04:
This DDF is based on this device #6785 and if it works please open a PR.

[r0_5501.json](https://github.com/user-attachments/files/19602380/r0_5501.json)

jan666, 2025-04-27:
Will Test the DDF soon

mattreim, 2025-05-06:
@jan666 It will be included in the next version and this issue can now be closed.

manup, 2025-05-17:
@jan666 could you test if the new DDF is working?

jan666, 2025-05-17:
Works in Version 2.30.2

closed 2025-05-17 by jan666