dresden-elektronik/deconz-rest-plugin #8131: Tuya 4 Gang Remote TS0044 _TZ3000_j61x9rxn

tadeaustria, 2025-03-07:
### Is there already an existing issue for this?

- [x] I have searched the existing issues and there is none for my device

### Product name

Tuya 4 Gang ZigBee Wireless 12 Scene Switch

### Manufacturer

Tuya

### Model identifier

TS0044

### Device type to add

Remote

### Node info

[image]

### Endpoints and clusters

[image]

### Basic

[image]

### Further relevant clusters

### Power Configuration

### On/Off

### Level Control

### Color Control

### Thermostat

### Simple Metering

### Electrical Measurement

### Any other cluster of relevance/interest

tadeaustria, 2025-03-07:
should be this one https://www.zigbee2mqtt.io/devices/TS0044_1.html

Smanar, 2025-03-07:
Hello, try with adding your model to the list

Here the classic one > https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/devices/tuya/_TZ3000_TS0044_4gang_remote.json
Here the locked one https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/devices/tuya/_TZ3000_TS0044_4gang_remote_locked.json

But to have return for remote switch you need too add your model in this file https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/button_maps.json

Here will be more complex according your OS, wich one you have ?

tadeaustria, 2025-03-08:
Hi,

Thank you for your input. I've already found the _device files_, but I missed the _button_maps.json_. I've added the model in the button_maps and now the device is detected correctly and I can program it in the Phoscon App.

I'm using a headless raspberrypi OS.

But maybe one further question. The switch would have double click capabilities but I cannot access them via the Phoscon App. Is this a limitation of the Phoscon App?

Thank you!

Smanar, 2025-03-08:
>But maybe one further question. The switch would have double click capabilities but I cannot access them via the Phoscon App. Is this a limitation of the Phoscon App?

It's possible, to be sure you can check the "event" using Phoscon/help/API information/event or directly the device json using Phoscon/help/API information/sensors/Device_id

You need to found a "state/buttonevent" field with a number as form X00Y, X is the button number, Y the event.

BTW do you want to make a PR to submit the device officialy, or want I make it (so tell me wich one DDF you have used)

barsk, 2025-03-09:
Chiming in on this issue. I too have the same device and hoping for an official inclusion in a coming release as I might be adding more of this one.
I'll lokk into how to include the device with the already defined one that exists as you mention above. A bit new to deconz and DDFs, but I think I understand how to do it.

Smanar, 2025-03-09:
It's a "classic" device, so I think it will work without problem and the PR can be valided fastly, I just need to know if all is working and if he want to make the PR or want I make it.

barsk, 2025-03-09:
Hmm, not sure what the difference between the normal and the "locked" tuya device is? Which json file to edit?

I added the manufacturer id _TZ3000_j61x9rxn to the **normal** variant as:
`{
  "schema": "devcap1.schema.json",
  "uuid": "da3c3112-7dcd-419f-9e4b-cc9f32bf6e56",
  "manufacturername": [
    "_TZ3000_ygvf9xzp",
    "_TZ3000_ee8nrt2l",
    "_TZ3000_dziaict4",
    "_TZ3000_mh9px7cq",
    "_TZ3000_b7bxojrg",
    "_TZ3000_ufhtxr59",
    "_TZ3000_abci1hiu",
    "_TZ3000_wkai4ga5",
    "_TZ3000_wbfgbpxq",
    "_TZ3000_uaa99arv",
    "_TZ3000_j61x9rxn"
  ],`

and in button_maps.json I added as:
`  
"TuyaDimmerMap": {
            "vendor": "Tuya",
            "doc": "4-gang remote",
            "modelids": ["_TZ3000_xabckq1v", "_TZ3000_j61x9rxn"],
            "map": [
                [1, "0x01", "ONOFF", "ON", "0", "S_BUTTON_1", "S_BUTTON_ACTION_S
HORT_RELEASED", "B1"],
                [1, "0x01", "ONOFF", "OFF", "1", "S_BUTTON_2", "S_BUTTON_ACTION_
SHORT_RELEASED", "B2"],
                [1, "0x01", "LEVEL_CONTROL", "STEP", "0", "S_BUTTON_3", "S_BUTTO
N_ACTION_SHORT_RELEASED", "B3"],
                [1, "0x01", "LEVEL_CONTROL", "STEP", "1", "S_BUTTON_4", "S_BUTTO
N_ACTION_SHORT_RELEASED", "B4"],
                [1, "0x01", "LEVEL_CONTROL", "MOVE", "0", "S_BUTTON_3", "S_BUTTO
N_ACTION_HOLD", "B3H"],
                [1, "0x01", "LEVEL_CONTROL", "STOP", "0", "S_BUTTON_3", "S_BUTTO
N_ACTION_SHORT_RELEASED", "B3R"],
                [1, "0x01", "LEVEL_CONTROL", "MOVE", "1", "S_BUTTON_4", "S_BUTTO
N_ACTION_HOLD", "B4L"],
                [1, "0x01", "LEVEL_CONTROL", "STOP", "1", "S_BUTTON_4", "S_BUTTO
N_ACTION_SHORT_RELEASED", "B4R"]
            ]
        },
`
Then restarted deconz "systemctl restart deconz-gui". Removed the device-node in GUI and added again in Phoscon. Unfortunately no change, still undetected in Phoscon and in the deconz gui it is shown as before.

What am I doing wrong here?

Smanar, 2025-03-09:
The difference is a request, the tuya unlock sequence, somes devices need it to work.
on the button_maps.json you just have to add the model on this part

```
        "Tuya3gangMap": {
            "vendor": "Tuya",
            "doc": "3-gang remote",
            "modelids": ["_TZ3000_yw5tvzsk", "_TZ3000_czuyt8lz","_TZ3000_ee8nrt2l", "_TZ3000_ygvf9xzp", "_TZ3000_t8hzpgnd", "_TZ3000_wkai4ga5", "_TZ3000_bi6lpsew", "_TZ3400_keyjhapk", "_TYZB02_key8kk7r", "_TZ3400_keyjqthh", "_TZ3400_key8kk7r", "_TZ3000_vp6clf9d", "_TYZB02_keyjqthh", "_TZ3000_peszejy7", "_TZ3000_qzjcsmar", "_TZ3000_owgcnkrh", "_TZ3000_adkvzooy", "_TZ3000_arfwfgoa", "_TZ3000_a7ouggvs", "_TZ3000_rrjr1q0u", "_TZ3000_abci1hiu", "_TZ3000_dfgbtub0", "_TZ3000_rco1yzb1", "_TZ3000_gbm10jnj", "_TZ3000_5e235jpa", "_TZ3000_sj7jbgks", "_TZ3000_w8jwkczz", "_TZ3000_dziaict4", "_TZ3000_famkxci2", "_TZ3000_itb0omhv", "_TZ3000_tzvbimpq", "_TZ3000_mh9px7cq", "_TZ3000_b7bxojrg", "_TZ3000_ufhtxr59", "_TZ3000_4upl1fcj", "_TZ3000_i3rjdrwu", "_TZ3000_w3c7ouru", "_TZ3000_abci1hiu","_TZ3000_wqcbzbae", "_TZ3000_pkeqinnt", "_TZ3000_uaa99arv", "_TZ3000_wbfgbpxq", "_TZ3000_fa9mlvja", "_TZ3000_tk3s5tyg", "_TZ3000_mrpevh8p", "_TZ3000_qgwcxxws", "_TZ3000_yj6k7vfo", "_TZ3000_axpdxqgu", "_TZ3000_a4xycprs", "_TZ3000_u3nv1jwk", "_TZ3000_fkvaniuu"],
            "map": [
```

and yes, the name is not good ^^, it's for all tuya, not only 3 gangs

barsk, 2025-03-09:
Thanks, it works now as an "unlocked" variant. I can add it in Phoscon and configure the switch buttons for Phoscon light groups.

But, when  I try to handle the device in Home Assistant I only get the battery status, no buttons events to create automations on - such as switch on/off lights on single, double or long press. Why is HA not discovering my device properly from the deConz integration? I have som Ikea switches and they work fine with all the events programmable.

tadeaustria, 2025-03-09:
> It's a "classic" device, so I think it will work without problem and the PR can be valided fastly, I just need to know if all is working and if he want to make the PR or want I make it.

Sorry, was quite busy today. I created the PR.

tadeaustria, 2025-03-09:
> It's possible, to be sure you can check the "event" using Phoscon/help/API information/event or directly the device json using Phoscon/help/API information/sensors/Device_id
> 
> You need to found a "state/buttonevent" field with a number as form X00Y, X is the button number, Y the event.

hm... I am really new to this. With API information you mean to directly access the REST-API? I probably have to read more on that in https://dresden-elektronik.github.io/deconz-rest-doc/getting_started/ first though... right?

barsk, 2025-03-09:
I want to point out that the link to the remote type above: [https://www.zigbee2mqtt.io/devices/TS0044_1.html](https://www.zigbee2mqtt.io/devices/TS0044_1.html) is not the exakt same as I have. It looks mostly identical, but the battery is two standard AAA and the link here says it is a CR2430. Not sure that is of any concern as the mappings seems to work well in deConz and Phoscon.

But, I am having trouble making automations in HA as I mentioned above, as the button events is not visible. All I get is battery status...?

tadeaustria, 2025-03-10:
> I want to point out that the link to the remote type above: https://www.zigbee2mqtt.io/devices/TS0044_1.html is not the exakt same as I have. It looks mostly identical, but the battery is two standard AAA and the link here says it is a CR2430. Not sure that is of any concern as the mappings in the PR seems to work well in deConz and Phoscon.

Yes, you are right. I've also the model using AAA's. But I am not sure if the battery description is correct in the link. To me it seems the whole design of the white lower part is for using AAA's. With a CR2430 it could be way shorter...

> But, I am having trouble making automations in HA as I mentioned above, as the button events is not visible. All I get is battery status...?

I'm sorry, I cannot help you there. I have no experience with Home Assistant.

barsk, 2025-03-10:
Well, the event for long press, i,e 1003, 2003, 3003, 4003 is almost useless. You have to wait 5 long seconds which corresponds to the blue led switching off, and if you wait *just a bit too long* the remote will enter pair mode. This seems to happen on all buttons, not just the first. The pair mode ends after a bit and everything still works, but in the mean time the remote is unresponsive.  

To make this work you have to look at the blue led, wait for it to go off and then release the button. Then you get the long press event. One problem with this is that the led is almost invisible behind that black cover glass. As said, almost useless. I had in mind to use these events for dimming a light, but not really working well for that...

They should have put a dedicated pair button inside the battery compartment, and the long press should trigger in about 2-3 seconds. IMHO...

Smanar, 2025-03-10:
Yeah "long press" are not a good idea on thoses devices ^^, Aqara/Xiaomi have a device with the same problem too.

tadeaustria, 2025-03-11:
> Well, the event for long press, i,e 1003, 2003, 3003, 4003 is almost useless. 

Oh yes. I thought I was just to stupid, but yes this just doesn't work, with the pairing mode. This really sucks...

PR #8134 (2025-03-28): DDF clone for Tuya 4 Gang Remote (_TZ3000_j61x9rxn) and ZemiSmart ZRM4 (_TZ3000_xwuveizv, _TZ3000_wkai4ga5)
Resolves #8131
https://www.zigbee2mqtt.io/devices/TS0044_1.html#tuya-ts0044
closed 2025-03-28 by manup