dresden-elektronik/deconz-rest-plugin #8133: _TZE284_dikb3dp6 Zemismart Tuya Zigbee 3 Phase Electric Energy Meter

nioram, 2025-03-09:
### Is there already an existing issue for this?

- [x] I have searched the existing issues and there is none for my device

### Product name

Zemismart Tuya Zigbee 3 Phase Electric Energy Meter SPM02-D2TZ

### Manufacturer

Zemismart Tuya

### Model identifier

SPM02-D2TZ

### Device type to add

Sensor

### Node info

[image]

### Endpoints and clusters

[image]

### Basic

[image]

### Further relevant clusters

Only Tuya without Data: 

[image]

nioram, 2025-03-09:
I´ve tried to write my own DDF but that didnt report the correct data. 

[image]

Smanar, 2025-03-09:
Hello, your DDF is on "Bronze" status, so not used.
This device is a tuya one, so it don't use classic cluster

It's more like this DDF https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/devices/tuya/_TZE200_byzdayie_din_enrgy_meter.json

Here you have the DP list
```

            tuyaDatapoints: [
                [1, 'energy', tuya.valueConverter.divideBy100],
                [23, 'produced_energy', tuya.valueConverter.divideBy100],
                [29, 'power', tuya.valueConverter.raw],
                [32, 'ac_frequency', tuya.valueConverter.divideBy100],
                [50, 'power_factor', tuya.valueConverter.raw],
                [103, 'voltage_a', tuya.valueConverter.divideBy10],
                [104, 'current_a', tuya.valueConverter.divideBy1000],
                [105, 'power_a', tuya.valueConverter.raw],
                [108, 'power_factor_a', tuya.valueConverter.raw],
                [109, 'energy_a', tuya.valueConverter.divideBy100],
                [110, 'energy_produced_a', tuya.valueConverter.divideBy100],
                [112, 'voltage_b', tuya.valueConverter.divideBy10],
                [113, 'current_b', tuya.valueConverter.divideBy1000],
                [114, 'power_b', tuya.valueConverter.raw],
                [117, 'power_factor_b', tuya.valueConverter.raw],
                [118, 'energy_b', tuya.valueConverter.divideBy100],
                [119, 'energy_produced_b', tuya.valueConverter.divideBy100],
                [121, 'voltage_c', tuya.valueConverter.divideBy10],
                [122, 'current_c', tuya.valueConverter.divideBy1000],
                [123, 'power_c', tuya.valueConverter.raw],
                [126, 'power_factor_c', tuya.valueConverter.raw],
                [127, 'energy_c', tuya.valueConverter.divideBy100],
                [128, 'energy_produced_c', tuya.valueConverter.divideBy100],
            ],
```

nioram, 2025-03-09:
Thanks to your help i was able to get some data working: 

[image] 

but how do i tell it to show all 3 phases? How does a DDF for that look like? (i only need a example the real work i can do myself)

Smanar, 2025-03-09:
If you need only current (no need power and curent, one is the image of the other), there is special field, so you can use them in the same sensor.

```
        {
          "name": "state/current_P1",
          "parse": {
            "fn": "tuya",
            "dpid": 104,
            "eval": "Item.val = Attr.val / 1000;"
          },
          "read": {
            "fn": "none"
          },
          "default": 0
        },
        {
          "name": "state/current_P2",
          "parse": {
            "fn": "tuya",
            "dpid": 113,
            "eval": "Item.val = Attr.val / 1000;"
          },
          "read": {
            "fn": "none"
          },
          "default": 0
        },
        {
          "name": "state/current_P3",
          "parse": {
            "fn": "tuya",
            "dpid": 122,
            "eval": "Item.val = Attr.val / 1000;"
          },
          "read": {
            "fn": "none"
          },
          "default": 0
        }
```

Else you can create "Fake" sensor 

```
    {
      "type": "$TYPE_POWER_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0b04"
      ],
```

Here you have one with 0x01, but can create too 0x02 and 0x03

But it will depend of your third app too, if you are using only phoscon Im not sure it support current_PX

closed 2025-04-08 by github-actions[bot]