dresden-elektronik/deconz-rest-plugin #8138: Third Reality Intelligent Soil Moisture Sensor (3RSM0147Z)

bokysan, 2025-03-18:
### Is there already an existing issue for this?

- [x] I have searched the existing issues and there is none for my device

### Product name

Smart Soil Moisture Sensor

### Manufacturer

Third Reality, Inc

### Model identifier

3RSM0147Z

### Device type to add

Sensor

### Node info

[image]

### Endpoints and clusters

Not quite sure what to put here, but my page "Cluster info" is empty and there's no no "read" button as seen on [the instruction manual](https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Request-Device-Support)

[image]

### Basic

Hi,

I've tried to create a DDF file based on [documentation I found online]() and [Deconz Toolbox](https://deconz-community.github.io/ddf-tools/#/auto-bundler), but I've given up, due to not sure what I am doing.

I've [created a DDF file](https://github.com/user-attachments/files/19314576/soil-moisture-sensor.json), create a DDB and uploaded it to deConz via POST API.

The file _seems_ to be OK, but the device:
- does not show up in Phoscon
- does not show with a name in the deCONZ GUI

I'm attaching the DDF file I created from given data.

### Further relevant clusters

[image]

Smanar, 2025-03-18:
Lol, you have used ChatGPT ?
Take an existing working one and just change the modelID and the Manufacture name.

bokysan, 2025-03-18:
This seems to work. The sensor is showing up in Phoscon and reporting values:

[image]

Thank you so much for quick help!

Smanar, 2025-03-18:
NP, do you want to make the PR to submit the device officialy or want I make it ?

Smanar, 2025-03-19:
Done > https://github.com/dresden-elektronik/deconz-rest-plugin/pull/8140

closed 2025-03-26 by manup