dresden-elektronik/deconz-rest-plugin #8141: Alarm systems and IAS ACE devices

ebaauw, 2025-03-20:
### Does the issue really belong here?

- [x] I definitively want to report a bug within deCONZ or its REST-API

### Is there already an existing issue for this?

- [x] I have searched the existing issues and there is none for the bug at hand

### Describe the bug

Looking into alarm systems (at the request of one of my Homebridge deCONZ users), I found a number of bugs, inconsistencies and shortcomings.  I'll use this issue to document them, hopefully being able to submit one or more PRs to correct them.

1. The API for `/alarmsystems` doesn't report errors correctly.

### Steps to reproduce the behavior

## Ad 1 - wrong API error reporting:
When specifying an invalid attribute:
```
$ deconz put -v /alarmsystems/1/config '{"code": "1234"}'
deconz put: request 1: PUT /alarmsystems/1/config {"code":"1234"}
deconz put: error: request 1: http status 404 Not Found
```
When specifying an invalid value:
```
$ deconz put -v /alarmsystems/1/config '{"armed_away_exit_delay": "aap"}'
deconz put: request 1: PUT /alarmsystems/1/config {"armed_away_exit_delay":"aap"}
deconz put: error: request 1: http status 503 Service Unavailable
```

### Expected behavior

### Ad 1 - wrong API error reporting:
When specifying an invalid attribute, the API should return HTTP status 200 OK with error(s) in the response body, like e.g.:
```
$ deconz put -v /sensors/1/config '{"code": "1234"}'
deconz put: request 1: PUT /sensors/1/config {"code":"1234"}
deconz put: warning: request 1: /sensors/1/config/code: api error 6: parameter, code, not available
[
  {
    "error": {
      "address": "/sensors/1/config/code",
      "description": "parameter, code, not available",
      "type": 6
    }
  }
]
```
When specifying an invalid value, the API should return HTTP status 200 OK with errors in the response body, like e.g.:
```
$ deconz put -v /alarmsystems/1/config '{"code0": "aap"}'
deconz put: request 1: PUT /alarmsystems/1/config {"code0":"aap"}
deconz put: warning: request 1: /alarmsystems/1/config/code0: api error 7: invalid value, aap, for parameter, code0
[
  {
    "error": {
      "address": "/alarmsystems/1/config/code0",
      "description": "invalid value, aap, for parameter, code0",
      "type": 7
    }
  }
]
```

### Screenshots

n/a

### Environment

- Host system: all
- Running method: all
- Firmware version: all
- deCONZ version (not Home assistant Addon version!): 2.29.2 (and others)
- Device: all
- Do you use an USB extension cable: n/a
- Is there any other USB or serial devices connected to the host system? If so: Which? n/a

### deCONZ Logs

n/a

### Additional context

For HomeKit support, see https://github.com/ebaauw/homebridge-deconz/issues/246.

pponce, 2025-04-10:
thanks for adding this improve support for keypads via deconz.
I wanted to know if it was possible to have deconz expose the 4 digit codes that get entered by users.
Similar to how ZHA does it.
"event_type: zha_event
data:
  device_ieee: 00:0d:6f:00:15:74:86:bc
  unique_id: 00:0d:6f:00:15:74:86:bc:1:0x0501
  device_id: dc8c2629e4a29da1cc7af0c10cb262c2
  endpoint_id: 1
  cluster_id: 1281
  command: arm
  args:
    arm_mode: 0
    arm_mode_description: Disarm
    code: "1234"
    zone_id: 0
  params: {}
origin: LOCAL
time_fired: "2024-02-04T17:32:26.071555+00:00"
context:
  id: 01HNTJVD2QESPJXK66XK735CK6
  parent_id: null
  user_id: null"

Right now in Home assistant when I listen for  deconz_alarm_event, I see something like this below for invalid code events.
The actual code entered is not passed though. I don't get any events for successful arming events either.

event_type: deconz_alarm_event
data:
  id: ancillarycontrol_30
  unique_id: 00:0d:6f:00:13:44:55:65
  device_id: dda560983f98da2de1d7b8ce095068fb
  event: invalid_code
origin: LOCAL
time_fired: "2025-04-10T18:36:00.031166+00:00"
context:
  id: 01JRGFSJMZDJ8R3BJBKWKCD8TK
  parent_id: null
  user_id: null

I'd like to ultimately figure out how to use HA to configure codes for different users, one time user codes, etc..
Functionality that this plugin provides for HA when used with ZHA. 
https://github.com/LenirSantiago/HA-Zigbee-Keypad/tree/main

I can create a new issue request if that is the better approach.

pponce, 2026-09-20:
I’d like to add a reproduction related to item 4 and ask for guidance on the supported way to consume keypad code-result events.

I’m migrating garage-door logic from HomeKit/Homebridge scenes and automations into a local controller. The controller coordinates a separate bolt, checks door state, serializes operations, and publishes HomeKit feedback. I want it to consume keypad results directly from deCONZ while retaining deCONZ’s native PIN validation and alarm behavior.

The intended behavior is:

* Correct code + disarm sequence: open through the controller, including when the alarm is already disarmed. If armed, normal keypad disarming should still work.
* Incorrect code + disarm sequence: close through the controller, confirm closure, then extend the bolt.
* Arming remains an explicit user action. Garage movement must not automatically arm the alarm.

I don’t need access to PIN contents—just a reliable indication that a new code submission produced a particular result.

**Environment**

* Xfinity/Universal Electronics keypad: `URC4450BC0-X-R`
* Resource type: `ZHAAncillaryControl`
* deCONZ: `2.33.2`
* API: `1.16.0`
* `websocketnotifyall: true`

**Confirmed observations**

1. Correct entry while armed produced `disarmed` and disarmed the alarm.
2. Correct entry while already disarmed produced `already_disarmed`. Repeating the same correct entry produced another message with the same action and an advanced timestamp.
3. Repeated incorrect entries likewise each produced `invalid_code` with an advanced timestamp.
4. A supervised keypad-open / indoor-button-close cycle, followed by five minutes of observation with the keypad untouched, produced no additional keypad state messages.

We then tested a non-code entry update:

1. Started with the door closed, bolt locked, alarm disarmed, and cached keypad action `already_disarmed`.
2. Connected an observation-only WebSocket listener for the exact keypad resource.
3. Without entering a code or touching the keypad, used the deCONZ GUI to read endpoint `0x01`, IAS Zone server cluster `0x0500`, Zone Status attribute `0x0002`.
4. The Attribute Editor displayed “reading done.”
5. During the observation window, one keypad state message arrived containing `action: already_disarmed` with a newer `lastupdated`. The final REST snapshot also had an advanced timestamp and the same action.

This demonstrates that, in this test, a newer shared timestamp plus an Open-like action did not actually mean a new code entry. The GUI read response was visually confirmed, not independently correlated using a Zigbee packet capture. We have not reproduced an unsolicited supervision-report case or observed unexpected physical movement.  But given it's a door i don't want my script to accidently interpret subsequent messages as a succesfull code entry that triggers my door to open.

Filtering solely on action changes would lose legitimate repeated identical code results. A time-based cooldown would not establish whether a later message came from a new entry.

**Questions**

* Is there an existing supported event, field, rule mechanism, or resource mapping that distinguishes each new code-result event from unrelated IAS Zone updates, while preserving repeated identical results?
* Has the resource separation proposed in item 4 been implemented or become available through a supported DDF change or newer release?
* If a change is still needed, would you prefer separate IAS Zone resources or a code-result-specific event/timestamp/sequence indication?

I’d be happy to test a suggested solution. I’d like to preserve native alarm behavior and avoid exposing PINs.

pponce, 2026-09-20:
Following up: I noticed the merged changes in #8154 and #8157.

From the diffs, #8154 separates the frient KEPZB-110’s IAS Zone status into a separate alarm-sensor resource, removing those items from its ancillary-control resource. #8157 makes the corresponding generic ancillary-control items optional, allowing that separation.

Would the appropriate fix for the Xfinity/Universal Electronics **URC4450BC0-X-R** be an equivalent model-specific DDF update using that infrastructure?

My installed deCONZ 2.33.2 mapping still reproduced a cached `already_disarmed` action with an advanced `lastupdated` after a GUI IAS Zone Status read, without any code entry.

Could this be addressed entirely through the Xfinity DDF, or would additional changes be needed? In particular, I’d like to preserve the existing ancillary-control identity, native PIN validation/disarming, alarm membership and HomeKit integration while separating unrelated IAS Zone updates from code-result events.

Are there migration considerations for an already paired device—for example, existing resources retaining the old items after loading an updated DDF?

I’m happy to test a recommended mapping.

pponce, 2026-09-21:
Update: I tested a local DDF split for the Xfinity/UEI **URC4450BC0-X-R** on **deCONZ 2.33.2**, following the frient approach in #8154 and the optional ancillary items from #8157. I’m attaching the tested DDF.

My goal is to consume genuine keypad code-result events in a local garage controller without interpreting unrelated sensor updates as new code entries.

**Before:** With `websocketnotifyall=true`, a GUI Read of IAS Zone / Zone Status (endpoint `0x01`, cluster `0x0500`, attribute `0x0002`) produced an ancillary message carrying the previous `already_disarmed` action with a newer `lastupdated`, despite no keypad entry. This was a controlled reproduction, not an observed spontaneous garage opening.

**Change:** Preserve the existing temperature and ancillary identities, add a separate `ZHAAlarm` resource at `01-0500`, and move `enrolled`, `pending`, `lowbattery` and `tampered` off ancillary control. I installed a local JSON override with `raw_json` policy, retaining the upstream Gold status. No deCONZ or homebridge-deconz code changes were needed.

**Results:**

* Live mapping verification confirmed the separate Zone resource and removal of those fields from ancillary control.
* Repeating the successful Zone Status read produced **zero ancillary messages and no ancillary timestamp change**.
* Two correct-code entries while disarmed each produced a fresh `already_disarmed` event.
* Two incorrect-code entries each produced a fresh `invalid_code` event.
* Away, Stay/Home and Night arming, followed by correct-code-only disarming, all worked and displayed correctly in HomeKit.

This validates the reproduced case, not every possible report path. Would this separation be appropriate upstream for the Xfinity DDF? 

[xfinity-urc4450-separated.json](https://github.com/user-attachments/files/32447840/xfinity-urc4450-separated.json)