dresden-elektronik/deconz-rest-plugin #8148: TRÅDFRI dimmer doesn't change light

ACrazyConcept, 2025-03-27:
### Does the issue really belong here?

- [x] I definitively want to report a bug within deCONZ or its REST-API

### Is there already an existing issue for this?

- [x] I have searched the existing issues and there is none for the bug at hand

### Describe the bug

I set up a separate deCONZ with Conbee III in docker to see if I can get TRÅDFRI wireless dimmer (the very old round one) working so I could migrate my existing setup.
So far no luck.

I can pair the dimmer just fine. The Phoscon UI react just fine when I turn the dimmer.
But the light in the dimmer group doesn't react.

I have used Hue Essentials app to add the light to the dimmer. Since I could not figure out how to access the old web app anymore. (But previously I did try this when I initially tried to migrate my setup. With same result.)

Also, this particular dimmer and light works fine together in my existing Conbee II setup.

### Steps to reproduce the behavior

1. Add light to dimmer in old web app or hue essentials or with REST PUT request.
2. Try to dim.
3. See no change.

### Expected behavior

The dimmer should dim the light.

### Screenshots

[image]

[image]

### Environment

- Host system: PC with Arch Linux
- Running method: deconz-docker container
- Firmware version: 26530900
- deCONZ version: 2.29.2
- Device: ConBee III
- Do you use an USB extension cable: Yes
- Is there any other USB or serial devices connected to the host system? No

### deCONZ Logs

```
00:37:36:913 DB save zll database items 0x00000001
00:37:36:913 DB saved in 0 ms
00:37:38:888 Websocket 10.0.0.170:62378 send message: {"attr":{"ddf_hash":null,"ddf_policy":"latest_prefer_stable","lastannounced":null,"lastseen":"2025-03-27T23:37Z","manufacturername":"dresden elektronik","modelid":"ConBee III","name":"Configuration tool 1","nwkaddress":null,"swversion":"0x26530900","type":"Configuration tool","uniqueid":"00:21:2e:ff:ff:0e:1e:dc-01"},"e":"changed","id":"1","r":"lights","t":"event","uniqueid":"00:21:2e:ff:ff:0e:1e:dc-01"} (ret = 406)
00:37:40:914 Daylight now: nadir, status: 100, daylight: 0, dark: 1
00:37:44:914 sql exec SELECT conf FROM zbconf ORDER BY rowid desc limit 1
00:37:45:913 APS-DATA.indication srcAddr: 0x7395, srcEp: 0x01 dstAddrMode: 1, profile: 0x0104, cluster: 0x0008, lqi: 195, rssi: -61
00:37:45:913 	asdu: 11460500c3
00:37:45:913 [INFO] - Button 1001 - TRADFRI wireless dimmer, broadcast to: 0x0000, endpoint: 0x01, cluster: LEVEL_CONTROL (0x0008), action: Move up (with on/off), payload: 00C3, zclSeq: 70
00:37:45:913 Websocket 10.0.0.170:62378 send message: {"attr":{"ddf_hash":null,"ddf_policy":"latest_prefer_stable","lastannounced":null,"lastseen":"2025-03-27T23:37Z","manufacturername":"IKEA of Sweden","mode":1,"modelid":"TRADFRI wireless dimmer","name":"TRADFRI wireless dimmer","nwkaddress":29589,"productid":"E1724","swversion":"2.3.028","type":"ZHASwitch","uniqueid":"00:0b:57:ff:fe:24:a3:c0-01-1000"},"e":"changed","id":"2","r":"sensors","t":"event","uniqueid":"00:0b:57:ff:fe:24:a3:c0-01-1000"} (ret = 447)
00:37:45:913 Websocket 10.0.0.170:62378 send message: {"e":"changed","id":"2","r":"sensors","state":{"buttonevent":1001,"lastupdated":"2025-03-27T23:37:45.913"},"t":"event","uniqueid":"00:0b:57:ff:fe:24:a3:c0-01-1000"} (ret = 164)
00:37:46:041 APS-DATA.indication srcAddr: 0x7395, srcEp: 0x01 dstAddrMode: 1, profile: 0x0104, cluster: 0x0008, lqi: 202, rssi: -54
00:37:46:042 	asdu: 114704ff0100
00:37:46:043 [INFO] - Button 1002 - TRADFRI wireless dimmer, broadcast to: 0x0000, endpoint: 0x01, cluster: LEVEL_CONTROL (0x0008), action: Move to level 255 (with on/off), payload: FF0100, zclSeq: 71
00:37:46:043 Websocket 10.0.0.170:62378 send message: {"e":"changed","id":"2","r":"sensors","state":{"buttonevent":1002,"lastupdated":"2025-03-27T23:37:46.043"},"t":"event","uniqueid":"00:0b:57:ff:fe:24:a3:c0-01-1000"} (ret = 164)
00:37:47:514 APS-DATA.indication srcAddr: 0x7395, srcEp: 0x01 dstAddrMode: 1, profile: 0x0104, cluster: 0x0008, lqi: 205, rssi: -51
00:37:47:514 	asdu: 11480101c30000
00:37:47:515 [INFO] - Button 2001 - TRADFRI wireless dimmer, broadcast to: 0x0000, endpoint: 0x01, cluster: LEVEL_CONTROL (0x0008), action: Move down, payload: 01C30000, zclSeq: 72
00:37:47:516 Websocket 10.0.0.170:62378 send message: {"e":"changed","id":"2","r":"sensors","state":{"buttonevent":2001,"lastupdated":"2025-03-27T23:37:47.516"},"t":"event","uniqueid":"00:0b:57:ff:fe:24:a3:c0-01-1000"} (ret = 164)
00:37:47:626 APS-DATA.indication srcAddr: 0x7395, srcEp: 0x01 dstAddrMode: 1, profile: 0x0104, cluster: 0x0008, lqi: 201, rssi: -55
00:37:47:627 	asdu: 114904000100
00:37:47:627 [INFO] - Button 2002 - TRADFRI wireless dimmer, broadcast to: 0x0000, endpoint: 0x01, cluster: LEVEL_CONTROL (0x0008), action: Move to level 0 (with on/off), payload: 000100, zclSeq: 73
00:37:47:628 Websocket 10.0.0.170:62378 send message: {"e":"changed","id":"2","r":"sensors","state":{"buttonevent":2002,"lastupdated":"2025-03-27T23:37:47.628"},"t":"event","uniqueid":"00:0b:57:ff:fe:24:a3:c0-01-1000"} (ret = 164)
00:37:50:914 Daylight now: nadir, status: 100, daylight: 0, dark: 1
00:37:54:165 APS-DATA.request id: 188, addrmode: 0x03, addr: 0x000B57FFFEBCB872, profile: 0x0000, cluster: 0x0031, ep: 0x00 -> 0x00 queue: 0 len: 2 tx.options 0x00
00:37:54:166 	asdu (length: 2): 3700
00:37:54:219 APS-DATA.confirm id: 188, status: 0x00 SUCCESS
00:37:54:284 APS-DATA.indication srcAddr: 0x25F2, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x8031, lqi: 208, rssi: -48
00:37:54:284 	asdu: 3700010001dc1e0effff2e2100dc1e0effff2e21000000040200ca
00:37:54:285 APS-DATA.indication request id: 188 -> finished
00:37:54:285 APS-DATA.request id: 188 erase from queue
00:37:54:286 Websocket 10.0.0.170:62378 send message: {"attr":{"ddf_hash":null,"ddf_policy":"latest_prefer_stable","lastannounced":null,"lastseen":"2025-03-27T23:37Z","manufacturername":"IKEA of Sweden","modelid":"TRADFRI bulb E27 WS opal 980lm","name":"Color temperature light 2","nwkaddress":9714,"productid":"LED1545G12E27EU","swversion":"2.3.095","type":"Color temperature light","uniqueid":"00:0b:57:ff:fe:bc:b8:72-01"},"e":"changed","id":"2","r":"lights","t":"event","uniqueid":"00:0b:57:ff:fe:bc:b8:72-01"} (ret = 459)
00:38:00:915 Daylight now: nadir, status: 100, daylight: 0, dark: 1
```

ebaauw, 2025-03-28:
If memory serves, the round dimmer doesn't (no longer) support(s) group bindings (with its latest firmware).  For direct control, you need to bind the light(s) instead.  And the coordinator, to see the button events in deCONZ.

As the dimmer sends multiple commands in close succession while turning the dial, using group commands would saturate the Zigbee network, so it does make sense to use unicast bindings instead.

ACrazyConcept, 2025-03-28:
Hi @ebaauw 
Thanks for the feedback.
The dimmers are all 2.3.028 which I think is the newest.
It's just curious that this works in my existing Conbee II setup? At least the bindings that has been there for a long time still works after updating along the way. Although I do have some trouble with a few dimmers that have been deleted from deCONZ recently and then repaired. For those I can add the lights to the dimmer groups and the dimming works. But then the group drops the lights again after a short while.
I'm trying the suggestion. Not sure it seems to works as well as the old method though so far.

ebaauw, 2025-03-28:
These are old devices, with very limited resources.  These only hold a few binding table entries, and might ignore the cluster being bound.  E.g. creating a binding from the _Power Configuration_ cluster to the coordinator (for battery reporting) might also send _On/Off_ and _Level Control_ commands to that device, and block any subsequent group bindings for these clusters.  There might be some magic order in which to setup the bindings.

Also, be sure to wake the device by turning it, when sending (un)bind requests.

ACrazyConcept, 2025-03-28:
They're old but I really like them. I use 11 of them around the house.
I'll do some more tinkering!

ACrazyConcept, 2025-04-19:
I just setup a new RPi5 with the Conbee III to see if I could make it work the same as in the docker setup. However no luck so. The dimmer stays unpaired.
I would so much like to get these dimmers working as I really use then a lot.
Would you be willing to do some troubleshooting on this @ebaauw ?

closed 2025-04-27 by github-actions[bot]

ACrazyConcept, 2025-05-04:
Update: I have the dimmers working as I hoped now. I think I had previously not done the immediate "turn" when pairing directly to lights 😣. Thanks for your help @ebaauw !