dresden-elektronik/deconz-rest-plugin #8152: Shelly 1 Mini Gen4

badewanne1234, 2025-03-30:
### Is there already an existing issue for this?

- [x] I have searched the existing issues and there is none for my device

### Product name

Shelly 1 Mini Gen4 

### Manufacturer

Shelly

### Model identifier

Mini1

### Device type to add

Switch

### Node info

[image]

### Endpoints and clusters

[image]

### Basic

[image]

### Further relevant clusters

### Power Configuration

### On/Off

[image]

### Level Control

### Color Control

### Thermostat

### Simple Metering

### Electrical Measurement

### Any other cluster of relevance/interest

Smanar, 2025-03-31:
Hello can nstart with this DDF

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "Shelly",
  "modelid": "Mini1",
  "vendor": "Schelly",
  "product": "Mini Gen4",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_ON_OFF_OUTPUT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/alert",
          "default": "none"
        },
        {
          "name": "state/on"
        },
        {
          "name": "state/reachable"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0006",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x10",
          "min": 1,
          "max": 300
        }
      ]
    }
  ]
}
```

Do you know if the device have special feature ? it have special cluster.

badewanne1234, 2025-03-31:
Thank you I will research how I can use this DFF (I never had to use it previously) and then report back. 
These Shelly Devices are kind of a family and they have the same underlaying ESP32.
In particular, this Shelly 1 Mini Gen4 is only for on/off, however there is for example the Shelly 1PM Mini Gen4 --> the PM version is for PowerMeasurement and I would assume that that cluster would be populated. Not sure though if that would be something that could be used within Zigbee?

That said, the on/off state can be configured either as a switch, or it can be a button.
Switch: every  time you click it, it will toggle between on and off
Button: every time you click it, it will stay on as long as you click, then go back to off when you release the button.

badewanne1234, 2025-04-01:
What these Shellys also are capable of, is single/double/triple click and long press - if that makes a difference?

Btw, tjere is a typo on the above dff under Vendor: Schelly --> Shelly

Smanar, 2025-04-01:
I will correct the typo if the DDF is working.
No more information about this device on Z2MQTT.
And no more on the official site, I m seing only for the input

> User interface
> Inputs
> 
>     One (Control) button
> 
>         Press and hold for 5 seconds to enable Device access point and Bluetooth connection.
> 
>         Press and hold for 10 seconds to factory reset the Device.
> 
>         Press 5 consecutive times to switch the Device from Matter firmware (default) to Zigbee.
> 
>         Press 3 consecutive times to put the Device in Zigbee inclusion mode. The Device stays in this mode for 2 minutes and you can find it in the Home Automation platform through the Zigbee Hub.

Nothing to be used as "zigbee input".

If you need help to use the DDF, tell me your OS.

badewanne1234, 2025-04-01:
Ok great. I'm using The Deconz Addon in HA with the Conbee II - any help would be appreciated :)

mattreim, 2025-04-02:
Here are the DDF and the DDB. The DDB can be loaded directly into Phoscon via device integration. Don't forget to reboot afterward. 

[shelly_mini1_gen4.zip](https://github.com/user-attachments/files/19575339/shelly_mini1_gen4.zip)

badewanne1234, 2025-04-03:
Thank you. Are these methods persistent, I mean will they survive a HA reboot/update and also, a Phoscon update?

mattreim, 2025-04-03:
For me it works even after a restart, although I do it very rarely.

Smanar, 2025-04-03:
Some one want to make the PR to submit the device ?
Or want I make it ?

mattreim, 2025-04-03:
I can do it.

badewanne1234, 2025-04-03:
I will test it in the next days, as soon as I find time to do so. 👍🏻

badewanne1234, 2025-04-06:
@mattreim what would be the process to add the Shelly via the DBB file?
I uploaded it in the Phoscon Addon, then reboot HA - tried to pair it multiple times, but Phoson will not find the device under when trying to add as a "Switch".
In Deconz though, it changed the naming of the device and as I can see, multiple devices connected to it.

[image]

Smanar, 2025-04-06:
The difference will be not visible in the GUI, only in the API.
Your device is a "light" not a "switch", you need to see it in phoscon in "light" part.

badewanne1234, 2025-04-06:
Thanks for the hint - indeed it is listed under _Lights_. Thanks for the files, it works!

I can turn the relais on and off from Deconz, however, how would I now use the integrated switch inside Deconz? When its not listed under _Switches_, I cannot use the shelly switch to turn on/off **other** Zigbee lights.
My goal is to use the shelly as a switch, I dont have light attached to its output...

Smanar, 2025-04-06:
You need to use automation, able to use the output state.
This device is a light, so it don't have input, only output, some Aquara/Xiaomi have a "decoupld" mode, but I haven't see one for this device.

badewanne1234, 2025-04-06:
In the Shelly Config, thats exactly what you can choose.
1) between Button and Switch
2) the mode of that Button or Switch

[image]

I use it like this for other lights via HA, but my entire goal is to use these Shelly independently from HA, only within Zigbee. Why? Its faster AND deconz has an excellent dimming functionality. The latter in contrast is not given for HA...

Smanar, 2025-04-06:
The device probably use cluster 0xFC01 or 0xFC02 but I haven't information about it.

badewanne1234, 2025-04-06:
How could that be done, what do I need to sniff the gateway?
Meanwhile, I also created a support ticket at Shelly to see what their response is.

mattreim, 2025-04-06:
The Shelly website provides the following information about these clusters ([0xFC01](https://shelly-api-docs.shelly.cloud/gen2/Integrations/Zigbee/RPCCluster/)/[0xFC02](https://shelly-api-docs.shelly.cloud/gen2/Integrations/Zigbee/WiFiSetupCluster/)). Perhaps it will help.

badewanne1234, 2025-04-07:
Feedback from Shelly Suppot: "Yes, currently the switch input isn't supported in ZigBee. We cannot confirm anything regarding it as of now."

Is there still a chance that it could be supported with the above documentation that @mattreim has posted or is this now a showstopper?

Smanar, 2025-04-07:
Nice that  Shelly provides support for their stuff, good point.
But you have your answer, "not yet" ^^^

manup, 2025-04-28:
With the device added in v2.29.3-beta the issue can be closed now?

Smanar, 2025-04-29:
For me yes, a new one can be opened (or re-open this one) later when sonoff will pupose more options.

closed 2025-04-29 by manup