dresden-elektronik/deconz-rest-plugin #8160: Aeotec (Samjin) water leak sensor's new FW not working

jlatvala, 2025-04-04:
### Does the issue really belong here?

- [x] I definitively want to report a bug within deCONZ or its REST-API

### Is there already an existing issue for this?

- [x] I have searched the existing issues and there is none for the bug at hand

### Describe the bug

I have multiple Aeotec (Samjin) water leak sensors (IM6001-WLP01).
These devices are added into Phoscon (I am using deCONZ add-on in Home Assistant).

The ones, which I have bought earlier, have the attributes battery, temperature and water (dry or wet) visible in Phoscon and Home Assistant.
(Vendor for all these sensors is Samjin in Phoscon.)

**Problem: The new one has attributes battery and temperature in Phoscon and Home Assistant. The water (dry or wet) attribute is missing.** 
Using this sensor (making it wet) doesn't add the missing water attribute. 
Reseting this sensor doesn't add the missing water attribute.
Changing the battery doesn't add the missing water attribute.

The only difference I can see between the working old sensors and the non-working new sensor is in deCONZ's OTA Update plugin.
In OTA plugin
- old working sensors' version = 0x00000011 (hovering mouse over the value shows 0.0 build 17)
- new non-working sensor's version = 0x00000014 (hovering mouse over the value shows 0.0 build 20)

### Steps to reproduce the behavior

1. Open Phoscon in Home Assistant's deCONZ add-on
2. (factory reset sensor, if not adding it first time)
3. Add a new sensor (with new firmware mentioned in the bug report above) into Phoscon
4. Sensor is added into Phoscon
5. Sensor can be seen also in deCONZ (in HA's deCONZ's plugin)
6. Sensor can be seen in HA
7. Sensor's battery and temperature attributes are visible in Phoscon and in HA
8. Sensor's wet/dry attribute is not shown in Phoscon add-on or HA

### Expected behavior

Wet/dry attribute should be visible same way as it is visible in the old firmwares

### Screenshots

[image]

### Environment

- Host system: Raspberry Pi
- Running method: Home Assistent deCONZ Add-on
- Firmware version: 26720700
- deCONZ version: 2.28.1
- Device: ConBee II
- Do you use an USB extension cable: yes
- Is there any other USB or serial devices connected to the host system? If so: SSD disk

jlatvala, 2025-04-04:
[image]

Note: Manufacturer name and model identifier are shown correctly for both devices in Phoscon (Samjin and water)

Mimiix, 2025-04-04:
> [image]
> 
> Note: Manufacturer name and model identifier are shown correctly for both devices in Phoscon (Samjin and water)

Can you make sure both are populated?

jlatvala, 2025-04-04:
Now I tried again and got more meaningful values (after using read button in Cluster Info tab and waiting little longer).

[image]

Mimiix, 2025-04-05:
@Smanar can we fix this with a ddf?

Smanar, 2025-04-05:
Why not, it's realy old device, managed by legacy core.
Can make a DDF but I don't find this device on old issue, no cluster capture, so I presume it use the IAS cluster ?
```

{
  "schema": "devcap1.schema.json",
  "manufacturername": "Samjin",
  "modelid": "water",
  "vendor": "Samjin, Aeotec",
  "product": "Water alarm with temperature",
  "sleeper": true,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_WATER_LEAK_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0500"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0402",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0001",
          "0x0500"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery"
        },
        {
          "name": "config/enrolled"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/pending"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/lowbattery",
          "awake": true
        },
        {
          "name": "state/tampered",
          "awake": true
        },
        {
          "name": "state/water",
          "awake": true
        }
      ]
    },
   {
      "type": "$TYPE_TEMPERATURE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0402"
      ],
      "items": [
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery"
        },
        {
          "name": "config/offset"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/temperature"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0021",
          "dt": "0x20",
          "min": 60,
          "max": 3600,
          "change": "0x00000001"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0402",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x29",
          "min": 1,
          "max": 300,
          "change": "0x00000014"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0500"
    }
  ]
}

```

jlatvala, 2025-04-05:
These Aeotec devices were earlier Samsung' SmartThings devices. Then Samsung decided to focus only on SmartThings sw and licensed its devices to Aeotec.

Maybe you can find the device as Samsung device?

Smanar, 2025-04-05:
No more, but the DDF can work, have you made a try ?

jlatvala, 2025-04-05:
I have never added any devices with DDF. 
Is there some documentation (especially for HA's deCONZ add-on?

Or is this link the best guide: https://community.home-assistant.io/t/how-to-install-a-new-deconz-ddf-file/390455 ?
(this guide seems to be quite cumbersome :) )

Smanar, 2025-04-06:
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/8152#issuecomment-2770276653

jlatvala, 2025-04-09:
-I removed the non-working device from Phoscon
-copied the DDF above to the clipboard
-pasted it into https://deconz-community.github.io/ddf-tools/#/auto-bundler 
-saved the generated DDB file to own laptop
-used Phoscon's Device integration page to upload the DDB file to Phoscon
-added sensor back to Phoscon

The sensor works partially.
-Temperature works
-Wet/dry state works
-Battery state does **not** work (not visible in Phoscon and 0% in HA)

Entities in HA (three lower ones doesn't work):

[image]

(Old working sensors have only one Battery entity, see the screenshot in the original bug report)

Smanar, 2025-04-09:
Ok so it seem your device don't support state/lowbattery and state/tampered, but the config/battey need to work.
Can you try this DDF ? (Can you remove it from HA, because some fields will be removed on the future DDF, not sure HA will update it)

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "Samjin",
  "modelid": "water",
  "vendor": "Samjin, Aeotec",
  "product": "Water alarm with temperature",
  "sleeper": true,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_WATER_LEAK_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0500"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0402",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0001",
          "0x0500"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "awake": true,
          "refresh.interval": 86400,
          "read": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "fn": "zcl:attr"
          },
          "parse": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "fn": "zcl:attr",
            "eval": "Item.val = Attr.val / 2"
          },
          "default": 0
        },
        {
          "name": "config/enrolled"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/pending"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/water",
          "awake": true
        }
      ]
    },
   {
      "type": "$TYPE_TEMPERATURE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0402"
      ],
      "items": [
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "awake": true,
          "refresh.interval": 86400,
          "read": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "fn": "zcl:attr"
          },
          "parse": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "fn": "zcl:attr",
            "eval": "Item.val = Attr.val / 2"
          },
          "default": 0
        },
        {
          "name": "config/offset"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/temperature"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0021",
          "dt": "0x20",
          "min": 60,
          "max": 3600,
          "change": "0x00000001"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0402",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x29",
          "min": 1,
          "max": 300,
          "change": "0x00000014"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0500"
    }
  ]
}
```

If the reporting don't work for battery, deconz will try a poll, but need to wait 24h.

jlatvala, 2025-04-10:
I used the same process as described above to generate DDB and take it into use in phoscon. 
First the sensor showed only battery and temperature, but also the wet/dry state became visible (only in phoscon) when I dropped some water on the sensor. HA reboot and then the wet/dry state was visible also in HA.
So this DDF seems to work

Smanar, 2025-04-11:
Fine, thx, do you want to make the PR to submit it or want I make it ?

jlatvala, 2025-04-11:
> Fine, thx, do you want to make the PR to submit it or want I make it ?

Maybe you can make it, thanks!

Smanar, 2025-04-12:
Done > https://github.com/dresden-elektronik/deconz-rest-plugin/pull/8167

Smanar, 2025-04-17:
PR updated, but I have a question.
You are only using the "read" part for one entry, logic, only 1 poll can update all "parse", but it mean if we use the defaut 

```
        {
          "name": "attr/swversion"
        }
```

On a sensor with 4 or 5 entry (on a multi sensor), deconz make 4/5 poll for the same value ?

Smanar, 2025-04-17:
Added to PR.

manup, 2025-04-17:
Jup normally it should configure right on initial pairing, this only adds a bit of extra robustness so that the IAS state machine can repair / continue setup if something goes wrong initially.

PR looks fine now thanks 👍

closed 2025-04-19 by manup