dresden-elektronik/deconz-rest-plugin #8164: Aqara LED Bulb T2 GU10, RGB CCT missing CCT controls

DarkFox, 2025-04-08:
### Does the issue really belong here?

- [x] I definitively want to report a bug within deCONZ or its REST-API

### Is there already an existing issue for this?

- [x] I have searched the existing issues and there is none for the bug at hand

### Describe the bug

Adding a Aqara LED Bulb of the model "T2 GU10, RGB CCT", only adds colour and level controls, but not CCT.

The bulb has white LEDs that are usable through the Aqara app, but is not used when the bulb is added to Deconz. When trying to tune it to white, only the RGB LEDs are used, resulting in low light output and poor white rendering.

### Steps to reproduce the behavior

1. Add Aqara LED Bulb T2 GU10, RGB CCT to Deconz
2. Add the bulb to a group in Phoscon
3. Try to click "Colour temperature"
4. Observe that nothing happens
5. Try to adjust CCT in Home Assistant.
6. Observe that there is no CCT mode in the dialog for the light.
7. Inspect the device in deconz
8. Observe that it only has level and colour clusters, no colour temperature

### Expected behavior

The colour temperature controls should be available and the white LEDs used.

### Screenshots

[image]

[image]

[image]

[image]

### Environment

- Host system: Intel NUC
- Running method: Home Assistant deCONZ Add-on
- Firmware version: 26400500
- deCONZ version (not Home assistant Addon version!): 2.28.1
- Device: ConBee I
- Do you use an USB extension cable: yes
- Is there any other USB or serial devices connected to the host system? If so: Aeotec Z‐Stick Gen5 USB Controller, RFLink.

DarkFox, 2025-04-08:
Let me know if you need logs, and I'll try and figure out how to get them out from the HA add-on...

ebaauw, 2025-04-08:
Looks like the device is exposed through legacy code?  That requires whitelisting of ZHA lights, to map the ZHA _Color Dimmable Light_ to the ZCL/ZB3 _Color Temperature Light_ and expose `state/ct` and corresponding capabilities.  Rather than updating the legacy code, best create a DDF for this device.

DarkFox, 2025-04-30:
I've switched to a ZBT-1 and binned my conbee. You can go ahead and close this.