dresden-elektronik/deconz-rest-plugin #8166: TS011F TZ3210_amdymr7l power metering not working

mantaalex, 2025-04-10:
### Is there already an existing issue for this?

- [x] I have searched the existing issues and there is none for my device

### Product name

BlitzWolf BW-SHP-13

### Manufacturer

BlitzWolf

### Model identifier

BW-SHP-13 TS011F TZ3210_amdymr7l

### Device type to add

Switch

### Node info

[image]

### Endpoints and clusters

[image]

### Basic

[image]

### Further relevant clusters

[image]

### Power Configuration

### On/Off

[image]

### Level Control

### Color Control

### Thermostat

### Simple Metering

[image]

### Electrical Measurement

[image]

### Any other cluster of relevance/interest

[image]

```
        },
        "etag": "1c8d7098226056dfeed094cf0848685e",
        "hascolor": false,
        "lastannounced": null,
        "lastseen": "2025-04-10T17:39Z",
        "manufacturername": "_TZ3210_amdymr7l",
        "modelid": "TS011F",
        "name": "Frigogarage1",
        "state": {
          "alert": "none",
          "on": true,
          "reachable": true
        },
        "swversion": null,
        "type": "On/Off plug-in unit",
        "uniqueid": "a4:c1:38:79:8f:25:a8:f4-01"
      },
      "4": {
        "capabilities": {
          "alerts": [
            "none",
            "select",
            "lselect"
          ]
        },
        "config": {
          "groups": [
            "0",
            "2"
          ]
        },
```

mantaalex, 2025-04-10:
i'm willing to provide extra info, i justed buyed 12 plugs to use power metering and its not supported pf :)

Mimiix, 2025-04-10:
@Smanar can you help out?

Smanar, 2025-04-11:
There is already  a DDF for SHP13 and model ID TS011F (and aleady one for the _TZ3000_amdymr7l, but not the same model id)  > https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/devices/blitzwolf/bw_shp13_smart_plug.json

Probably on more clone, can you try to add your model to the list ?

```
  "manufacturername": [
    "_TZ3000_3ooaz3ng",
    "_TZ3000_g5xawfcq",
    "_TZ3000_amdymr7l",
    "_TZ3210_amdymr7l"
  ],
  "modelid": [
    "TS0121",
    "TS0121",
    "TS011F",
    "TS011F
  ],
```

mantaalex, 2025-04-13:
so i guess i have to edit in deconz the json of the device ?

mantaalex, 2025-04-13:
or do i have to do something else?

Smanar, 2025-04-13:
In theory editing the DDF can work, but you can have issue.
The better way is editing the DDF with a text editor, but harder if you have HA or other dockers OS.

Now on your picture, you have an autogenerated DDF, not the official one (you can see the path/title on the editor titlebar).
So if you start from this one, you have all the work to done.

On docker OS it's easier to create a new DDF on the persistent folder.

what is your OS ?

mantaalex, 2025-04-13:
Hm ok, I run deconz addon in home assistant os

mantaalex, 2025-04-13:
I can upload a new ddf trough phoscon
But then its for all my plugs

Smanar, 2025-04-14:
Ha yes if you know how to use this feature.
So create a complete new DDF

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": [
    "_TZ3000_3ooaz3ng",
    "_TZ3000_g5xawfcq",
    "_TZ3000_amdymr7l",
    "_TZ3210_amdymr7l"
  ],
  "modelid": [
    "TS0121",
    "TS0121",
    "TS011F",
    "TS011F
  ],
  "vendor": "BlitzWolf",
  "product": "Electricity metering 16A EU plug (BW-SHP13)",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_SMART_PLUG",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "parse": {
            "fn": "zcl:attr",
            "ep": 1,
            "cl": "0x0000",
            "at": "0x0001",
            "script": "../tuya/tuya_swversion.js"
          },
          "read": {
            "fn": "zcl:attr",
            "ep": 1,
            "cl": "0x0000",
            "at": "0x0001"
          }
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/alert",
          "default": "none"
        },
        {
          "name": "state/on",
          "refresh.interval": 5
        },
        {
          "name": "state/reachable"
        }
      ]
    },
    {
      "type": "$TYPE_POWER_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0b04"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0051",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0B04"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "parse": {
            "fn": "zcl:attr",
            "ep": 1,
            "cl": "0x0000",
            "at": "0x0001",
            "script": "../tuya/tuya_swversion.js"
          }
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/current",
          "refresh.interval": 300
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/power",
          "refresh.interval": 300
        },
        {
          "name": "state/voltage",
          "refresh.interval": 300
        }
      ]
    },
    {
      "type": "$TYPE_CONSUMPTION_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0702"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0051",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0702"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "parse": {
            "fn": "zcl:attr",
            "ep": 1,
            "cl": "0x0000",
            "at": "0x0001",
            "script": "../tuya/tuya_swversion.js"
          }
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/consumption",
          "read": {
            "at": "0x0000",
            "cl": "0x0702",
            "ep": 0,
            "fn": "zcl:attr"
          },
          "parse": {
            "at": "0x0000",
            "cl": "0x0702",
            "ep": 0,
            "eval": "Item.val = Attr.val * 10"
          },
          "refresh.interval": 300
        },
        {
          "name": "state/lastupdated"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0006",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x10",
          "min": 1,
          "max": 300
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0702",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x25",
          "min": 1,
          "max": 300,
          "change": "0x0000000A"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0B04",
      "report": [
        {
          "at": "0x0505",
          "dt": "0x21",
          "min": 1,
          "max": 300,
          "change": "0x00000001"
        },
        {
          "at": "0x0508",
          "dt": "0x21",
          "min": 1,
          "max": 300,
          "change": "0x00000064"
        },
        {
          "at": "0x050B",
          "dt": "0x29",
          "min": 1,
          "max": 300,
          "change": "0x00000001"
        }
      ]
    }
  ]
}
```

And no, it will impact only the _TZ3210_amdymr7l, for other device the DDF will be the same.

mantaalex, 2025-04-15:
in the code:
´´´
  "modelid": [
    "TS0121",
    "TS0121",
    "TS011F",
    "TS011F
  ],
´´´
i see like a missing " ?
How do i make a DDF of your yson?
Could you make a ddf for me that i can try to upload trough phoscon
Would be great, i'm not an expert in this things here :)
Thanks for the effords

mantaalex, 2025-04-15:
Ok thanks and this part:
i see like a missing ", at the end ?

Is this a error or i do i have to do it like this

Smanar, 2025-04-16:
Arf, sorry, you are right, I missed a quote, the json is fine now, jave tested on https://jsonlint.com/

mantaalex, 2025-04-19:
thanks it works
i went to: https://deconz-community.github.io/ddf-tools/#/auto-bundler
and first installed deconz on my windows pc
and without moving the stick i put your json inside there where the tuya file is
then i made the new DDB and it works
so you can add it in the next beta/stable i don't know how to apply for this fix :)

Smanar, 2025-04-19:
Np, thx to you for test
Added here > https://github.com/dresden-elektronik/deconz-rest-plugin/pull/8181

closed 2025-04-21 by manup