dresden-elektronik/deconz-rest-plugin #8185: Switch with Temp device

enboer, 2025-04-23:
### Is there already an existing issue for this?

- [x] I have searched the existing issues and there is none for my device

### Product name

TYZGTH1CH-D1RF

### Manufacturer

Mumibiz

### Model identifier

TYZGTH1CH-D1RF

### Device type to add

None

### Node info

I have this device running in Deconz. Hower it sometimes reports a temp of 85C. is there a way to ignore this high value? It happens 85C, but eg 10-20secs later it is correct again. But I see that sometimes Domoticz picks up these values so I would like to ignore these high values.

### Endpoints and clusters

temp

### Basic

switch and temp

### Further relevant clusters

### Power Configuration

### On/Off

### Level Control

### Color Control

### Thermostat

### Simple Metering

### Electrical Measurement

### Any other cluster of relevance/interest

Smanar, 2025-04-23:
Hello, it happen sudentlty or all the time ?
Be sure for battery, 70% battery level mean battery empty for lot of devices.

Do you know how to edit the DDF ?
The file is `devices\tuya\_TZ3218_ya5d6wth_relay_temperature.json` can edit it to have

```
        {
          "name": "state/temperature",
          "read": {
            "fn": "none"
          },
          "parse": {
            "dpid": 102,
            "eval": "if (Attr.val < 600) { Item.val = 10 * Attr.val + R.item('config/offset').val; }",
            "fn": "tuya"
          },
          "default": 0
        }
```

It's untested, but if I m right it will ignore all value > 600, so 6000, so 60 degrés.

And BTW you know how to remove a bad value on the graph on Domoticz ?

enboer, 2025-04-24:
Thanks, I implemented this and will see if it helps.
Is there a way to for that eg at least every 2 minutes this temperature value is send?

Yes I know that with Shift-Click you can delete data from the Domoticz dbase. However for this device it does not work. As this devices has just been installed I will remove and add it again in Domoticz as the history is not that important.

Smanar, 2025-04-24:
No, on tuya there is no bind/report we can set, the device use the tuya cluster and his own rules.

enboer, 2025-04-25:
Thanks I do not get the high temp values anymore.

closed 2025-04-25 by enboer

Smanar, 2025-04-25:
Yo know how work DDF folders ?
1 Folder is reseted at every deconz update, the other one is fixed.
Honnestly as you are the only one with the issue IDK if it's a good idea to add it officialy to the DDF.

So waiting for something definitive better to use the second folder.