dresden-elektronik/deconz-rest-plugin #8187: Conbee 3 - Tradfri bulbs turning off after scene switch since 2.29.5 26550900

realwax, 2025-04-24:
### Does the issue really belong here?

- [x] I definitively want to report a bug within deCONZ or its REST-API

### Is there already an existing issue for this?

- [x] I have searched the existing issues and there is none for the bug at hand

### Describe the bug

Since the update on conbee 3 to FW 26550900 and the version update of deconz to 2.29.5 a strange new behavior appeared. 

E.g a group of 5 bulbs. 2 scenes. 
Bulb 1 2 3 - on - 4 5 off in scene 1 
Bulb 1 2 3 4 5 on in scene 2.
When you recall scene 1  and recall scene 2 afterwards. bulb 4 5 get turned on all bulbs change accordingly to the scenes but at a random of 1 to 15 seconds bulbs 4 5 turn off again as if they still follow the old scene. If scene 2 will be recalled again they stay. 

This is a complete new behavior which was not there before and appeared with the last update. 
Please help/fix.

Thank you & kind regards
Wolfgang

### Steps to reproduce the behavior

like argued - latest conbee 3 ith fw and last release. 

### Expected behavior

no random turning off of bulbs that's should stay on. 

### Environment

- Host system: (Raspberry Pi / PC / NAS / MAC)
- Running method: (Raspbian / Ubuntu / Home Assistent deCONZ Add-on / deconz-docker container / Windows / Virtual Machine)
- Firmware version: (26xxyy00)
- deCONZ version (not Home assistant Addon version!): (2.xx.yy)
- Device: (ConBee I / ConBee II / ConBee III / RaspBee I / RaspBee II)
- Do you use an USB extension cable: (yes / no) -- only relevant for ConBee I/II/III
- Is there any other USB or serial devices connected to the host system? If so: Which?

doodahdayz, 2025-04-24:
I have seen the same issue with a Signify ( Hue ) bulb.   Also on Pi running latest Conbee III firmware and 2.29.5 version of deConz running in headless mode.  Turning on the light using a "group" command and shortly thereafter the light will turn off.  Another light in the same group does not.    Not sure if its just anecdotal, but the light turning off is a higher ID number than the one that stays on in that group...    Light with ID of 6 stays on... the light with ID of 20 is the one turning itself off after short time.  It appears in further testing that turning it on without using the "groups" command that this light will in fact remain on.

manup, 2025-04-24:
That's odd, I don't recall changes to groups/scenes in recent release. I'll try to recreate this.
From which deCONZ version did you upgrade?

doodahdayz, 2025-04-24:
I tink this problem existed prior to 2.29.5... I was seeing similar issue before but I thought it was related to a new bulb that Hue had released.  So I had moved the bulb to a Hue bridge from deConz.   I will try and see if I can reload a previous version of deConz dropping back to 2.29.3 and see if it exists there as well...    This is a real problem and I understand what you are saying about the changes to 2.29.5.

doodahdayz, 2025-04-24:
I have reloaded 2.29.4 and 2.29.3 now an this problem exists there as well.  One strange thing... I have a custom scene in my app that will turn on a "group".   It uses the groups API to send the ON state.  This is where I first noticed it would turn off the one light ( the last one added in deConz .. highest ID value for light )   I changed my app "scene" to not use the "group" command but just loop through and turn on each light individually.   When doing it that way as well the light will turn off after short time.    If I then just turn on that light via the direct API to set the state of the light to ON then the light will remain on.  So it's almost like if the lights are turned on either via the group API or in a looping fashion of turning the lights on then for some reason it will turn that last light off after some timeout.

doodahdayz, 2025-04-24:
Turns out I was not rolling back deConz just the zigbee configuration....  so my bad...  sorry for the confusion.  But this is a real problem.  I just need to figure out now how to actually roll deConz back to prior release to determine when this was introduced as an issue.....

realwax, 2025-04-24:
to be fair - it appeared after the update to 2.29.5 but I might have happened already in an earlier version. I can't state that precisely. But I occurs... ;) and renders many automations useless....as light turn. Thank you all for picking it up that quick.

doodahdayz, 2025-04-24:
To help add to this mystery...   This is happening for me when I'm using a "group" API call to turn on the group lights.  As I recently learned that not Phoscon is not always the same as the API's that developers have I tried turning on the "group" from Phoscon.   Using Phoscon to turn the group lights on then the lights stay on.  Using the API to turn on this group then I see the "last" light in the group turned off.  It's almost like someone was debugging and put in a timer or something to turn off the last light in the group after some relatively short period of time.

realwax, 2025-04-24:
I parsed my apt history. It must have happend in between deconz:armhf (2.27.3-beta, 2.27.4-beta)  and deconz:armhf (2.29.4-beta, 2.29.5-beta)
and may be in between 
   4551  180 -rw-r--r--  1 pi   pi    182190 11. Jul 2024  deCONZ_ConBeeIII_0x26530900.bin.GCF
   2985  196 -rw-r--r--  1 pi   pi    198814 14. Apr 10:41 deCONZ_ConBeeIII_0x26550900.bin.GCF

I don't know if it's deconz or FW related as I updated both...

doodahdayz, 2025-04-24:
Ive been chasing this problem all day now... and was confused that I was having a group issue and Wolfgang was having an automation problem.   I have for a long time had an issue with one automation ( out of the two I use ) that whenever I updated deConz I would have to delete the automation and add it back in.  Otherwise it just didn't work.  This is an automatic that would be triggered by two different motion sensors and when they stopped detecting motion would delay for 1 minute and then turn the lights off.  It was set for Active times of only evening and night.   

When turning my groups or individual lights on or off I was also sending a transitiontime parameter of 1.  I removed the transition parameter from the commands to turn groups or light on / off but still had issue with the one light turning itself off.  

Finally I deleted all automations that were set in Phoscon.  Rebooted the Pi that runs the Conbee and then with no automations set I no longer had the issue.   I then created the automatics back and cannot reproduce.  

Something is wrong in the automations that when deConz gets updated that the automations are somehow corrupting something.    I was confused as the light group I was seeing the issue with has no automations set on it at all.. but I'm now convinced that the problem lies somewhere in automations.  I don't think it's a firmware issue on the Conbee as I had this issue with automations prior to updating the firmware.   It's been a ghost in the machine problem for some time now.

realwax, 2025-04-25:
> Ive been chasing this problem all day now... and was confused that I was having a group issue and Wolfgang was having an automation problem. I have for a long time had an issue with one automation ( [...]

It's a little off topic I guess talking about the affected automation. I run my deconz setup since years I upgraded from conbee 1 to 3. After getting the old config working on conbee 3 some device needed to be rejoined. Everything working fine. I have two scenarios that showed the new issue. A group of lights with a motions sensor. First sight one light on... If you stay and move there 10 secs later all lights on. After the update when it went from 1 to all 4 on it went back to 1 light on.... Wtf. The second time it got triggered it worked. The other scenario is a xiamio swith where I cylce certain scenes in my living room. Starting with a few light on on the last scene all of them. When I trigger the full scene in a cycling mode the last lights that went on goes off immediately or within 15 seconds.
That's easy to identify. You have not change a thing just updated and some things are messed up.
I hope this one gets fixed. Thank you all.

manup, 2025-04-25:
In my first tests here I also see Ikea lights being turned off explicitly when a scene specifies off for a light (but since is also reported for hue lights there could be more).
We do have a hack which is doing this:

https://github.com/dresden-elektronik/deconz-rest-plugin/blob/13d628882e5fcf881e574ab25190b5a3d54aa62e/rest_groups.cpp#L2704C1-L2717C2

Simply removing this hack won't turn the lights off anymore, but if I read this right they won't turn off at all (which was the reason to introduce this in first place a while ago).

Unfortunately the scenes cluster implementations for a lot of lights are hopelessly broken beside on off, color and color temperature is a mess. I guess the best bet to fix this is to not use real Zigbee scenes for lights which don't provide correct implementation (which in the past was only Hue lights) and instead emulate this with the commands we know are working for lights. E.g. nowadays DDFs can specify what capabilities lights actually have and are working. With that we can look at a scene given by the API and send respective commands. Being Unicast this is of course way slower than one groupcast.

manup, 2025-04-25:
No by hopelessly broken I mean the firmware **scene cluster implementation** which is running *on the light*, this can only be fixed by the lights manufacturers. What we can do in deCONZ is build workarounds for lights which don't support scenes correctly.

For example the hack above was introduced, since Ikea lights for which a scene with `on: false` was recalled (a scene is stored on the lights scene cluster) didn't turn off the light. The workaround was to check if a scene with Ikea lights with scene state off  was recalled, then to send a unicast off command to the light.

As seen in this issue here is problem somewhere which needs to be fixed. Which is what I'm currently checking.
But also when this works again, there is more work todo to provide proper scenes which work for all lights with color and color temperature and so on – and here I mean we need a fallback to emulate scenes via unicast commands.

doodahdayz, 2025-04-25:
One more question... if Phoscon is doing a unicast for each device then why wasn't the Restful API for sending a group command doing just that under the covers and sending a unicast to each device in the group to act as a workaround?   It seems that if Phoscon is doing that then it would be consistent to provide that functionality via the Restful API.   Am I missing something or is this a viable solution to work around this issue since it appears that Phoscon has a way around this problem?

realwax, 2025-04-25:
To be on the safe side in arguments. Trådfri was working as supposed to since years with some flaws always on the changes side of deconz. So Conbee 1 and 3 were able to handle it accordingly even if some clusters were not right.
So do I need to downgrade fw and sw to get it working again?
Thank you

realwax, 2025-04-28:
Hi,
Today I tested different stable versions - the issue comes with 2.29.5 but I can't state at which beta or initial version it was changed. Fact is that it not there and functional with version 2.27.6 and still good in version 2.28.1. So from my logical thinking there must have been a change that messed a functional behavior up. So only 3 days between those two stables.
Hopefully it helps.... So for now I stay at 2.28.1 in hope it will be fixed in future versions.
Thank you
Wolfgang

[deconz-2.27.6-qt5.deb](https://deconz.dresden-elektronik.de/raspbian/stable/deconz-2.27.6-qt5.deb)  15-Aug-2024 12:13            13357222

[deconz-2.28.1-qt5.deb](https://deconz.dresden-elektronik.de/raspbian/stable/deconz-2.28.1-qt5.deb)  16-Apr-2025 00:25            13364584

[deconz-2.29.5-qt5.deb](https://deconz.dresden-elektronik.de/raspbian/stable/deconz-2.29.5-qt5.deb) 19-Apr-2025 18:14            14368086

doodahdayz, 2025-04-28:
This is interesting.   I kept testing different things and found that I had one light that is a newer Philips Hue bulb ( 100w equivalent RGB bulb ) and once I moved it from the Conbee back to a Philips Hue bridge that suddenly all my problems went away. In fact everything now is working great, including using the Restful API to deConz for groups as well as doing unicast to individual devices.     I was just getting ready to post that Manuel appears to be spot on with the bulb firmware in some units are  an issue.  Now its a mystery if its the bulb at fault or something in deConz isn't processing something...  I may see what happens if I roll back but I'm in the middle of trying to get a release out of my app.    It would be interesting though if rolling back to the version you are talking about makes the problem I have had go away with that bulb on the Conbee.

realwax, 2025-04-28:
Please correct me if I am wrong but in my consideration if I have had a working scenario unchanged since years and suddenly (with 2.29.x) an update messes it up. It's more like the updates / change fault. The only exception I see - yes if the implementation was not correct and now fulfills standards which then are not met by IKEA but - in terms of UX - it does not help. It worked before - it's broken now. It can be reproduced. Never the less - I am happy to share my findings, send deeper debugs... just let me know what's needed. 

Here is the FW info of my Tradfri bulb:

HerstellerIKEA of Sweden
ModellTRADFRI bulb E14 WS opal 400lm
Version 2.3.087

Now back on 2.28.1 everything is fine. Not any bulb is getting turned off if in the previous scene it's hasn't been turned on.

manup, 2025-04-29:
I'm currently diving into this as the last fix for the upcoming release.

Some notes for myself to track:

The Ikea turn off workaround was added in 2021 in PR https://github.com/dresden-elektronik/deconz-rest-plugin/pull/4846 to fix https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3135 in v2.11.3-beta.

The initial `TRADFRI bulb E14 WS opal 400lm` DDF was added in 2021 but not active (Bronze status) https://github.com/dresden-elektronik/deconz-rest-plugin/commit/64264c815ff92771698488661210df1f9b1c818a#diff-2cb2a25afd1e50b81e7fa8a7c8fa13e4dc16f25f55f1ba95e8ad31aae26cb6ff

It was activated in v2.29.2-beta https://github.com/dresden-elektronik/deconz-rest-plugin/pull/8111 the DDF it self is fine and has proper configuration.

I'm trying now to recreate such scenes which show improper turning off when the light should be on..

manup, 2025-04-29:
A first manual test via deCONZ Cluster Info Panel to see if the workaround is still needed.
There are two scenes for my *TRADFRI bulb E14 WS opal 400lm* on group 0x0001 (1):

0x01  On with warm color
0x03  Off

To make the off scene the light was turned off and *Store Scene Command* for scene 0x03 and group 0x0001 was send.
This command stores the current state of a light in the specified scene.

Sending a *View Scene Command*:

[image]

Note OnOff and brightness is correct, but no color XY coordinates are in the response. The color temperature oddly isn't part of the scene specification and can only be expressed as xy coordinates. The thing for Ikea is that *View Scene Commands* are broken as they only return half of the truth. The color temperature is in fact part of the scene but can't be verified with scene commands.

Ideally we would use the *Add Scene Command* which would allow to specify all attributes in it. But this is also broken in Ikea lights, so we only use it initially to store the transition time for a scene (which can't be done via *Store Scene*).

Scene cluster attributes look right since the off scene (0x03) is active:

[image]

Next up send a *Recall Scene Command* for scene 0x01 which does recall the scene with 'white and on'.
Looks also good. The scene does correctly turn on the light with warm color:

[image]

-----

Next send a  *Recall Scene Command* for scene 0x03, which **should** turn off the light...

[image]

The light scene attributes say all good 0x03 scene is active.
The light says also the OnOff attribute is **off**.

[image]

But! This ain't true, the light didn't turn off, it is still physically **on**.

That's the bug in the light firmware for which the workaround was added in https://github.com/dresden-elektronik/deconz-rest-plugin/pull/4846 

-----------------------------

Next lets recall the scene 0x03 via REST-API to see if the workaround works:

```
curl -X PUT localhost:8090/api/12345/groups/1/scenes/3/recall | jq
```

In the sniffer we can see the group *Recall Scene Command* but additionally the unicast *Off with Effect Command* which actually turns the light off (the workaround).

[image]

And the light now actually turned physically off.

Then also via REST-API recall scene 0x01 to turn the light on:

```
curl -X PUT localhost:8090/api/12345/groups/1/scenes/1/recall | jq
```

Looks good in the sniffer, and since the scene in the API doesn't have `on=false`there is also no workaround applied.

[image]

The light turns on with warm white and attributes look also fine.

-----------------------

Note this is done with my development deCONZ version which has already other fixes after version v2.29.5 in place, I don't think this has an impact but I'll repeat the tests with the v2.29.5 version to be sure.

I'd also like to check a `zll.db` database from yours for example the one from @realwax to see if the scene in the database has the correct fields, if you could please send a backup or just the `zll.db` file to support@dresden-elektronik.de we can check if the problem sits there.

hanskroner, 2025-04-29:
> A first manual test via deCONZ Cluster Info Panel to see if the workaround is still needed. There are two scenes for my _TRADFRI bulb E14 WS opal 400lm_ on group 0x0001 (1):
> 
> 0x01 On with warm color 0x03 Off
> 
> To make the off scene the light was turned off and _Store Scene Command_ for scene 0x03 and group 0x0001 was send. This command stores the current state of a light in the specified scene.
> 
> [...]
> 
> Ideally we would use the _Add Scene Command_ which would allow to specify all attributes in it. But this is also broken in Ikea lights, so we only use it initially to store the transition time for a scene (which can't be done via _Store Scene_).
> 
> [...]
> 
> But! This ain't true, the light didn't turn off, it is still physically **on**.
> 
> That's the bug in the light firmware for which the workaround was added in [#4846](https://github.com/dresden-elektronik/deconz-rest-plugin/pull/4846)

I'm not having this issue myself, nor do I have any IKEA bulbs to help corroborate - I just happened across this issue and wanted to point something out I remember from a while back: IKEA bulbs have never respected the "off" state when using `scene_store` - but I recall they do if `add_scene` is used instead - which is the only command the IKEA gateway (used to?) use when defining scenes. I guess that furthers @manup 's point that these particular bulbs don't always behave as they're expected to.

ebaauw, 2025-04-29:
(sorry - wrong button)

That's a good point, Hans.  I have seen lights not honouring an off state in scenes, but I don't remember if these were IKEA Trådfri or others.  Usually _Store Scene_ (which stores the current state) works better than _Add Scene_ (to which the state is specied in the command payload).  Especially for _Color Temperature_, which the ZCL standard doesn't include in the scene state.

hanskroner, 2025-04-29:
True, _Color Temperature_ was not part of the Scene Table Extensions for the Color Control cluster for a while. I can't pinpoint exactly when it was added, but it's been there for at least the last 5 years - though the [REST API doesn't (yet?) make use of it](https://github.com/dresden-elektronik/deconz-rest-plugin/blob/cc7bd69d6a4972d08089282c6388d299071d1ac9/zcl_tasks.cpp#L1354).
I'll stop hijacking the issue now - sorry if I derailed you all, but hope the information turns out to be useful!

realwax, 2025-04-30:
@manup Thank you Manuel! I've already sent the mail to the support address containing my zll.db. Furthermore I tested 2.29.1-beta and it was working fine without the issue and the issue appears in 2.29.2-beta. So you are spot on. Thank you!

manup, 2025-04-30:
Cool thanks I'll have a look now 👍 

@hanskroner good catch, I wasn't aware that Color Temperature was added as scene extension. I've checked the R8 ZCL spec and it is there. I'm not sure if it is easy to check if a device supports it. The *Color Control Cluster* has the cluster revision and the latest per specification is `3`, my Ikea lights report revision `1` unfortunately the specification doesn't tell at which revision the Color Temperature was added. But it's worth to test it out anyway.

manup, 2025-04-30:
>  IKEA bulbs have never respected the "off" state when using scene_store - but I recall they do if add_scene is used instead - which is the only command the IKEA gateway (used to?) use when defining scenes. 

Out of interest I made a small test with *Add Scene Command* and `on=0x00` extension field. Unfortunately while the command  is accepted it still doesn't work when recalling the scene the light stays physically off.

ebaauw, 2025-04-30:
> I'm not sure if it is easy to check if a device supports it.

_ColorTemperatureMireds_ as listed last attribute (8) in the scene table extensions for _Color Control_ of ZCL v8 (section 5.2.2.5) and ZCL v7, but not in ZCL v6.  So the check should be: _ZCL Version_ in the _Basic_ cluster has value >= 7.

However ZCL v8 and v7 also contain the following text from v6 and earlier:

> Since there is a direct relation between _ColorTemperatureMireds_ and XY, color temperature, if supported, is
stored as XY in the scenes table.

Sniffing the _View Scene Response_ on my ZCL v8 Hue lights, _ColorTemperatureMireds_ is **not** included in the payload.  However, when recalling the scene, the light switches to _ColorMode_ _Color Temperature_ nonetheless.  Somehow the light has stored the colour temperature in the scene.  If memory serves, this is the same behaviour as under the pre-ZCL-v8 firmware.  Note: the scene was updated using _Store Scene_ to get it to contain the colour temperature.

I haven't tried building an _Add Scene_ command with _ColorTemperatureMireds_ in the payload.  Maybe the Hue bridge sends this, but that's some more work to find out.

hanskroner, 2025-05-01:
> Out of interest I made a small test with Add Scene Command and on=0x00 extension field. Unfortunately while the command is accepted it still doesn't work when recalling the scene the light stays physically off on.

@manup Ah, that's unfortunate. I have a friend with these bulbs in their house and remember them facing this issue. They're using a different bridge though, and [it seems to work there](https://github.com/Koenkk/zigbee2mqtt/issues/5153).

> I haven't tried building an Add Scene command with ColorTemperatureMireds in the payload. Maybe the Hue bridge sends this, but that's some more work to find out.

@ebaauw  If you find yourself with the time and interest, I've gone a bit into that rabbit hole. These days, the Hue bridge uses a manufacturer-specific "variant" of _Add Scene_. You won't be surprised to hear it uses the same frame format as the command in the `FC03` cluster. This means Hue bulbs can store additional attributes, like gradients and effects in their scenes. My guess is that being able to differentiate between "color" and "color temperature" (instead of just cramming the information into XY) simplifies scene definitions for them across their line of white, white ambiance, and white and color ambiance products - there's a single definition with scene information all of their products can use.

I've implemented this in [my fork of deconz-rest-plugin](https://github.com/hanskroner/deconz-rest-plugin/commits/hue-dynamic-scenes/) to add support for the "dynamic scenes" the Hue bridge and mobile app have. It builds on the `FC03` support in #7950, adding the manufacturer-specific _Add Scene_ command and some API endpoints. I decided against contributing back these additions (though I'd be happy to if there's interest) as they required some refactoring to Scenes to allow omitting the scene extension tables of supported clusters (thus allowing, for example, scenes that modify only "color" state without affecting current brightness) and an additional column in the database to store the "dynamic scene" information - so you'll want to back-up your `zll.db` beforehand!

realwax, 2025-05-02:
i did upgrade to deconz-2.30.0 beta. Issue fixed. Thank you very much @manup.

manup, 2025-05-02:
Cool thanks for the feedback. Gonna wait 1-2 more days and if nothing new comes up the stable version will be pushed.

doodahdayz, 2025-05-03:
I have been working with the 2.30.0 beta... and I hope you wait more than 1-2 days.   Some of the changes have broken decoding the JSON as now I'm seeing "etag" attribute set to Null  in some occasions.   It's fine.. i get that you all are trying to tighten things up.  I would have preferred instead of Null that String values were set to "" empty string,  but please give developers a chance to push out updates to their Apps before you release 2.30.0    Thank you!

manup, 2025-05-03:
Good catch I've added a small fix to ensure there is always an Etag in https://github.com/dresden-elektronik/deconz-rest-plugin/pull/8202 the Etag is not supposed to be empty / null in any state.

doodahdayz, 2025-05-03:
Manuel... I can't tell you how much I appreciate this.  I also had to deal with some "last announced" and "last seen" that can show up as null...   but I've managed to deal with those now, as well as the "etag" being null.   I will try and get an update to the App Store this week, hopefully before 2.30 goes into a release state.

manup, 2025-05-03:
Yeah for timestamps in *last announced*, *last seen* and *swversion* I'd not like yet to change to different values yet. These need to be discussed in the next developer meeting first. For the timestamps we could use the creation time of the device if we have no other data yet. That would relax clients to always get a valid timestamp and don't need to bother with `null` or empty string checks. *swversion* could also get a default value like "unknown" but needs to be discussed first.

doodahdayz, 2025-05-03:
I understand completely and I've made the changes needed to handle these being Null.  So hopefully I'm good no mater what is decided.    I should be good and if I can test and get an update out this week before you promote 2.30 to release status that would be great.  Hope you have a wonderful weekend!  Thank you for your attention on these issues.   I sincerely appreciate it.

doodahdayz, 2025-05-03:
FYI.. when adding in some “presence sensors “ I found they are creating a entity that says it’s a switch… and the manufacturer name is null,   So I’ve had to make sure when decoding the JSON that I allow for manufacturers name being set to null as well.    Always Something ™️ 🙄

manup, 2025-05-03:
What's the modelid of the sensor? Normally the basic cluster is queried first and sensors/lights are created after that 🤔

doodahdayz, 2025-05-03:
Model number is TS0601. ….  I am a bit confused…. I have two of these and it looks like they create a device that shows up as a plug.   That is where both manufacturers and model ID are null.

brientim, 2025-05-03:
@doodahdayz @manup was looking for the Manufacture Name and hence the confusion. 

I have just requested issue [8105](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/8105) be re-opened addressing another Tuya clone  "manufacturername"  "_TZE200_n8dljorx"

manup, 2025-05-06:
Closing this issue as completed by https://github.com/dresden-elektronik/deconz-rest-plugin/releases/tag/v2.30.0-beta

The other off topic  bugs are tracked in different issues.

closed 2025-05-06 by manup