dresden-elektronik/deconz-rest-plugin #8190: Another Tuya Door Window Sensor

tobyDoh, 2025-04-27:
### Is there already an existing issue for this?

- [x] I have searched the existing issues and there is none for my device

### Product name

Tuya Door Window Sensor

### Manufacturer

TZ3210_r0xgkft5

### Model identifier

TS0203

### Device type to add

Sensor

### Node info

[image]

[image]

### Endpoints and clusters

[image]

### Basic

[image]

### Further relevant clusters

### Power Configuration

### On/Off

### Level Control

### Color Control

### Thermostat

### Simple Metering

### Electrical Measurement

### Any other cluster of relevance/interest

Smanar, 2025-04-27:
Hello, can you try to add it in the looonnnng DDF https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/devices/tuya/_TZ3000_TS0203_door_sensor.json
Most of them have nothing special.

You need to add 2 lines, the manufacture name AND the model ID.

tobyDoh, 2025-04-27:
i added the two lines:

[image]

restarted deconz. Still get the error.

[image]

what am i doing wrong? Appreciate your support.

Smanar, 2025-04-27:
There is something wrong about your device
TZ3210_r0xgkft5 is a light from lidl, so can't be your device.
On the second capture the fields are "unknow", you need to awake the device before asking for attribute.
https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/How-to-read-Clusters

We need the model id/Manufacture name, will be visible on the cluster "Basic" 0x0000 or the "Node info"

tobyDoh, 2025-04-28:
Hi Smanar,

it took a moment, but now i got the correct manufacture ID 

_TZ3000_bpkijo14

with having this entered in the mentioned file and doing a restart, the sensors are connected probably. 

Thanks for your support and maybe you like to add this to the file in the repo as well.

Regards

closed 2025-04-28 by tobyDoh