dresden-elektronik/deconz-rest-plugin #8200: Shelly 1PM Mini Gen4

SwoopX, 2025-05-02:
### Is there already an existing issue for this?

- [x] I have searched the existing issues and there is none for my device

### Product name

Shelly 1PM Mini Gen4

### Manufacturer

Shelly

### Model identifier

Mini1PM

### Device type to add

Other

### Node info

[image]

### Endpoints and clusters

[image]

### Basic

[image]

### Further relevant clusters

### Power Configuration

### On/Off

[image]

### Level Control

### Color Control

### Thermostat

### Simple Metering

[image]

[image]

### Electrical Measurement

[image]

[image]

### Any other cluster of relevance/interest

manup, 2025-05-03:
Intersting that it doesn't provide any version information nor an OTA cluster (upgrades perhaps are done differently via WiFi/Bluetooth).

SwoopX, 2025-05-03:
Yeah, that's indeed a bit strange. I checked the device via Wifi and it runs rather old firmware. Would've been nice to also have it exposed through basic cluster. As I haven't seen any place to officially download a new FW (yet?), it seems like you need to do that via the integrated GUI, which either would prohibit to run in Zigbee mode only or additionally activate wifi (if that is possible at all).

closed 2025-05-06 by manup