dresden-elektronik/deconz-rest-plugin #8210: Philips Hue Dimmer Switch creating LightGroup that should be marked as hidden.

doodahdayz, 2025-05-04:
### Does the issue really belong here?

- [x] I definitively want to report a bug within deCONZ or its REST-API

### Is there already an existing issue for this?

- [x] I have searched the existing issues and there is none for the bug at hand

### Describe the bug

When adding a Philips Hue Dimmer Switch model id:  RWL020 it also is creating a LightGroup for this switch.  I believe that like some other newly added groups such as the Phoscon_All_Off group that this new LightGroup should be marked as hidden.     

### Steps to reproduce the behavior

Add Hue Dimmer Switch Model RWL020 to the system
It caveats a LightGroup associated with this dimmer but it is not marked as Hidden.

### Expected behavior

I would think the LIghtGroup created for some special purpose would be marked as hidden. If this is incorrect then are we to assume that LightGroups created with an ID value of 20000 or higher is not really a "group" ie Room...  and we should just filter it out in our apps? 

### Environment

- Host system:  Raspberry Pi 
- Running method: (Raspbian ) 
- Firmware version: (26550900)
- deCONZ version (not Home assistant Addon version!): (2.30.0)
- Device: Conbee III
- Do you use an USB extension cable:  yes 
- Is there any other USB or serial devices connected to the host system?  No

brientim, 2025-05-04:
In addition to switches that creates light groups to manage the assigned lights, you may want to also consider the upcoming pending implementation of the new Phoscon Home (dashboard). 

Phoscon Home also creates addition system-generated light groups , such as “Phoscon_room_X_on_off_group”,  as discussed in the [forum](https://forum.phoscon.de/t/phoscon-home-dashboard-beta-test/6215).  

I believe that all system-generated light groups could default to hidden and the dashboard could include an enhancement to display "All" if desired.

brientim, 2025-05-26:
Please keep this open as has the potential impacts on the development path of the new Phoscon desktop, 3rd party applications, and the api by enforcing consistency of system generated groups.

brientim, 2025-06-17:
Bump. Need to check and verify behavior in Phoscon App v2.0.6.

doodahdayz, 2025-06-17:
yes, please see if its possible to address this issue.  It definitely impacts third party apps.

closed 2025-07-16 by github-actions[bot]