dresden-elektronik/deconz-rest-plugin #8211: 2 Button Switch  Model  TS0042, manufacturers:  _TZ3000_fkvaniuu    incorrectly reported as 4 button switch

doodahdayz, 2025-05-04:
### Does the issue really belong here?

- [x] I definitively want to report a bug within deCONZ or its REST-API

### Is there already an existing issue for this?

- [x] I have searched the existing issues and there is none for the bug at hand

### Describe the bug

The two button switch   _TZ3000_fkvaniuu    is being reported as a 4 button switch.  According to the info at https://dresden-elektronik.github.io/deconz-rest-doc/devices/   This switch is a 2 button switch.  It is also reported in the button_maps.json file as being listed with the 4 gang switches.  

The introspect API is also reported this switch as having 4 buttons.

### Steps to reproduce the behavior

If you look at  https://dresden-elektronik.github.io/deconz-rest-doc/devices/   you will see that _TZ3000_fkvaniuu is shown as a 2 button device.   Yet Phoscon and the introspect API to get the switch info are reporting it as a 4 button switch. 

### Expected behavior

Both Phoscon and the introspect API should be indicating this is a 2 button switch. 

### Environment

- Host system:  Raspberry Pi 5
- Running method: Raspbian
- Firmware version: (26550900)
- deCONZ version: 2.30.0
- Device: Conbee III
- Do you use an USB extension cable: (yes / no)  yes 
- Is there any other USB or serial devices connected to the host system?    NO

Smanar, 2025-05-05:
Hello, but where are you seing it's reported as 4 button switch ?
The DDF create only 1 entry, a ZHASwitch and with it you will have event on state/buttonevent with value like 1XXX or 2XXX ?

doodahdayz, 2025-05-06:
An additional observation is that there must be an issue with DDF as well..   sometimes if you add this switch in then the battery levels are reported correctly... and other times it just reports 0%.  So sometimes it seems to get added properly and other times not.  It is also reporting the 0% battery in Phoscon as well.

Smanar, 2025-05-06:
But even with a 0% value at start, the value is updated after 24 H, no ?

doodahdayz, 2025-05-06:
I was hoping that it would get updated but it did not.    If I take it out and add it back there is about a 10% chance that it gets a battery level.   But leaving it for going on 2 days now it's still reporting 0% battery level.   Double checked and Phoscon is showing the same on their API for the switch as well.

doodahdayz, 2025-05-29:
Thank you for the update...  now I need to figure out where is the DDF, how do I edit a DDF...   Not something I have ever done.   Will this be fixed in a beta release down the road or is this just a patch that I will always have to make sure is applied to the DDF ?

Smanar, 2025-05-29:
I prefer someone test it first, because few return like your, and the change can have impact (more requests possibles)
What is your OS ? If it's not HA will be easier.

doodahdayz, 2025-05-29:
running  deConz and Conbee III on Raspberry Pi raspbian OS....   not using HA or any 3rd party...    have my own  iOS and macOS app that just uses the deConz API's.   I run deConz in headless mode...but can turn on the GUI interface....   frankly I have no clue about what all that shows... so I'm basically stupid when it comes to the GUI interface and the DDF's...

Smanar, 2025-05-29:
So no problem, will be fast and without danger.

`All DDF provided directly with deCONZ typically reside in /usr/share/deCONZ/devices/ on a Linux system and are loaded first. However, files residing in the home directory of the user running deCONZ (e.g. /home/<DECONZUSER>/.local/share/dresden-elektronik/deCONZ/devices) will override the pre-packaged files to allow users to amend and keep their own files if desired.`

If you use /usr/share/deCONZ/devices/tuya, you will found inside a file called _TZ3000_TS0042_2gang_remote.json, it's the one used for devices 
```
  "manufacturername": [
    "_TZ3000_5e235jpa",
    "_TZ3000_t8hzpgnd",
    "_TZ3000_tzvbimpq",
    "_TZ3000_i3rjdrwu",
    "_TZ3000_fkvaniuu"
  ],
```

(and yes the change will impact all devices)

Can use a simple text editor to edit the file, can stay in headless mode.

If you use the second folder, the change will be definitive, deconz reset only the first folder. In this folder you need to copy/paste the whole DDF (I can create it if you need) because it will be empty. But if there is a typo in your edition, deconz will use the DDF from the first folder.

Then restart deconz, deconz load DDF only at start. And wait at least 24h.

doodahdayz, 2025-05-29:
I really appreciate your time and explanations.    I will try this later today.  I have some other tests running and need to let those run.   Thank you very much.

doodahdayz, 2025-05-29:
Your instructions were excellent and I put the DDF file in the user directory rather than editing the default DDF.   It has already now reported the battery level and averring seems to be working fine.

Thank you again for your time and teaching me some new things.  Very much appreciate it.
I'm guessing that the issue of it reporting as a 4 button switch is not something that we will see addressed anytime soon.   If that's the case then I would just go ahead and close this.  Hoping that long term they add capabilities to the RESTfjul API returned for a switch.  That would make it easy to know what a switch actually supports for buttons and events.

doodahdayz, 2025-05-31:
So far there has been no change in the battery level reported, but will keep on eye on it as it will take  a few days to verify that it isn't draining it based on the reporting frequency.

Smanar, 2025-05-31:
The reporting is fine 86400s mean 24h.
But the device is a sleeper, so deconz need to wait for the device be awake, make a try, and if not succed, retry again and again and again ....

doodahdayz, 2025-05-31:
This switch gets used a lot...  so I'm guessing that all is good.

Smanar, 2025-06-22:
PR merged

closed 2025-06-22 by Smanar

doodahdayz, 2025-06-22:
I hate to say this.. but I was about to update this as the battery issue for the 2 gang switch is still not right...  In a week the battery went from 58% to 8% and on the off chance it was a cheap battery I replaced it with a brand new name brand battery... and after 24 hours it still is saying 8% battery level.

Smanar, 2025-06-23:
Arf, I will make a PR to remove the last change for now

Smanar, 2025-06-23:
PR for rollback done, but now why you have this issue.
The pool frequency is 24 H .....

doodahdayz, 2025-06-23:
Let me know if there is anything I can do that would help.  Since its not updated the battery level since I replaced the battery I would question whether its draining the battery or is it just not reporting the level correctly.  It really could be either at this point.    I was expecting it to update the battery level after the new battery was installed but it has remained at 8%.  Which I question now if that was ever correct.  All I ever observed was 58% when you gave me the DDF and then noticed that it was now reporting 8%.

I have another switch that at some point I need to open an issue on ...  now it just drains batteries like there is no tomorrow.  It can drain a battery in less than a day.

Smanar, 2025-06-23:
> Since its not updated the battery level since I replaced the battery I would question whether its draining the battery or is it just not reporting the level correctly

You mean with the new DDF you have battery drain, and not the battery level ? (It need to be updated every 24h)

I m checking on other project, and execpt removing the bind (cause a battery drain) nothing special.
Perhaps the tuya unlock sequence ?

Try to add 
```
        {
         "name": "config/tuya_unlock"
        },
```
To the DDF, and re-include the device (the tuya unlock sequence is done at pairing or every 24h), no need to delete it first.

And from that I m reading the device send itself value time to time.

Something you can do after having add the tuya unlock sequence is taking a look in logs if you have the GUI ? With "info" and "info_l2", your device need to be a sleeper, so if you see periodic request from it, there is a problem.

doodahdayz, 2025-06-24:
I added the unlock sequence as you requested and rebooted the system.   After rebooting and using the switch a couple of times I checked the device using Phoscon and my app.  Both are now reporting the battery level at 100% .  That is what I would suspect with a new battery just replace a few days ago.  Hopefully this is the garlic that needed to be shaken over this vampire issue.  Thank you again for your time and assistance on this.    I have never used the GUI but one time and found myself lost and confused by it.   I need to read up more and understand the system better at that level.

Smanar, 2025-06-24:
We will wait a little this time, before making a new PR ^^.
Tell me if this time it's working.

doodahdayz, 2025-07-07:
It's been two weeks now and the battery level is now reported as 99%.  I have a lot of switches that have been installed for a long time that still report 100% but this is much better so far than it was for battery drain.    I have a couple of Tuya clone switches that I should open a PR on as they will drain a battery in less than 24 hours.   Hopefully if I open up on those they can be addressed by something similar to the fix for this 2 gang switch.  Thank you again for your time on this.  Much appreciated.

Smanar, 2025-07-07:
>Hopefully if I open up on those they can be addressed by something similar to the fix for this 2 gang switch

Honnestly I don't see what other problem it can be, there is few action possibles on them.
For the moment the PR was rollback, if you haven't issue I will remake a PR with the tuya unlock sequence.

Wich ones others device other the same issues ? You have corrected the 4 gangs, others are 1,2 or 3 gangs ?

doodahdayz, 2025-07-08:
I just looked at the 2 gang switch that we made the changes to the DDF ...  yesterday it was reporting 99% battery level and today it is down to 97%.    I'm starting to think that this switch is just power hungry or something.   With it set to only update every 24 hours I would have thought it would be using less power than this.  It's a brand new, brand name battery, not that it means anything these days.  Other devices in my system stay at 100% reporting levels for what sees like forever.  I wish someone would make a quality switch instead of all these Tuya clones.  I'd be happy to pay a lot more for something that doesn't feel like junk when you press it.  And even worse look like something from the bargain bin at "We Print Stuff, Just Not Good Stuff" Store.