dresden-elektronik/deconz-rest-plugin #8212: eWeLink bulb CK_BL702-AL-01 not completely controllable

FrankH1, 2025-05-05:
### Does the issue really belong here?

- [x] I definitively want to report a bug within deCONZ or its REST-API

### Is there already an existing issue for this?

- [x] I have searched the existing issues and there is none for the bug at hand

### Describe the bug

Bought a pair of Zigbee bulbs. Pairing works, they are reported as 
Hersteller eWeLink
ModellCK-BL702-AL-01(7009_Z102LG03-1)
Version 0x00001203
Controlling in Phoscon App (as well in OpenHab) works only limited.
I'm not able to set white color. 
In Phoscon app the "Color temperature" doesnt show a slider control - so I can't set. In the "Color" interface going to the "red end" for color, and zero saturation gives a pale cold white but the brightness slider effects only somewhere between 0 and 50%. More brightness isnt delivered, so I stay with rather dark and bluish white.
Obviously the same from Openhab - I'm expecting to set HSBvalues to "0,0,100" or "360,0,100" but it doesnt work either.

### Steps to reproduce the behavior

Behavior is reproducable and stays from the very beginning. Before pairing the bulbs where bright white, but once they are paired I cant obbviously get to that brightness.

### Expected behavior

Control as expected

### Environment

- Host system: Raspberry Pi
- Running method: Raspbian
- Firmware version: 26780700)
- deCONZ version : 2.28.1
- Device: ConBee II
- Do you use an USB extension cable: yes
- Is there any other USB or serial devices connected to the host system? yes - SD card reader

FrankH1, 2025-05-09:
Here is the ddf file as in /usr/share/deCONZ/devices/

[ck-bl702-swp-01(7020)_plug.json](https://github.com/user-attachments/files/20127787/ck-bl702-swp-01.7020._plug.json)

FrankH1, 2025-05-09:
> Hello can you show the node picture from the GUI, to see missing possible missing clusters, because I do'nt see why deconz badly reconize it.
> And if you can share the autogenerated DDF from the "preview" tab from the DDF editor, easier for me to complete it, than remake it from scratch.

Hi,

cant give something from gui as I’m running headless – but the other devices around work, so I expect no missing cluster.

Uploaded the ddf.

Cheers Frank

Smanar, 2025-05-10:
Hello, you need the GUI to have the "autogenerated DDF"
But on my side, I give up, this device need again a Tuya hack, I don't found other devices with a similar one, and I don't have time for this kind of device.

closed 2025-06-09 by github-actions[bot]