dresden-elektronik/deconz-rest-plugin #8224: _TZE284_rccxox8p smoke detector support

phwagner82, 2025-05-08:
### Is there already an existing issue for this?

- [x] I have searched the existing issues and there is none for my device

### Product name

Tuya Smoke Fire Alarm Detector

### Manufacturer

_TZE283_rccxox8p

### Model identifier

TS0601

### Device type to add

Sensor

### Node info

[image]

### Endpoints and clusters

[image]

### Basic

[image]

### Further relevant clusters

### Power Configuration

### On/Off

### Level Control

### Color Control

### Thermostat

### Simple Metering

### Electrical Measurement

### Any other cluster of relevance/interest

Smanar, 2025-05-09:
All thoses devices are clones
`fingerprint: tuya.fingerprint('TS0601', ['_TZE200_m9skfctm', '_TZE200_rccxox8p', '_TZE284_rccxox8p']),`

Can try this DDF `_TZE200_TS0601_smoke_detector.json`

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": [
    "_TZE200_m9skfctm",
    "_TZE200_vzekyi4c",
    "_TZE200_rccxox8p",
    "_TZE284_rccxox8p"
  ],
  "modelid": [
    "TS0601",
    "TS0601",
    "TS0601",
    "TS0601"
  ],
  "vendor": "Tuya",
  "product": "Photoelectric smoke sensor (TS0601)",
  "sleeper": true,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_FIRE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0xef00"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "parse": {
            "fn": "zcl:attr",
            "ep": 1,
            "cl": "0x0000",
            "at": "0x0001",
            "script": "tuya_swversion.js"
          },
          "read": {
            "fn": "zcl:attr",
            "ep": 1,
            "cl": "0x0000",
            "at": "0x0001"
          }
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "awake": true,
          "refresh.interval": 3600,
          "read": {
            "fn": "none"
          },
          "parse": {
            "dpid": 15,
            "eval": "Item.val = Attr.val;",
            "fn": "tuya"
          },
          "default": 0
        },
        {
          "name": "config/enrolled",
          "public": false
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/pending"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/errorcode",
          "read": {
            "fn": "none"
          },
          "parse": {
            "dpid": 11,
            "eval": "Item.val = String(Attr.val);",
            "fn": "tuya"
          },
          "default": "0"
        },
        {
          "name": "state/fire",
          "awake": true,
          "read": {
            "fn": "none"
          },
          "parse": {
            "dpid": 1,
            "eval": "Item.val = (Attr.val == 0 ? true : false);",
            "fn": "tuya"
          },
          "default": false
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/test",
          "read": {
            "fn": "none"
          },
          "parse": {
            "dpid": 101,
            "eval": "Item.val = Attr.val;",
            "fn": "tuya"
          },
          "default": 0
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0xEF00"
    }
  ]
}
```

Have just added 2 clones.

phwagner82, 2025-05-10:
Thanks @Smanar !

It is working!

Smanar, 2025-05-10:
Nice, do you want to make a PR to submit it or want I make it ?
Just need to edit this file https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/devices/tuya/_TZE200_TS0601_smoke_detector.json

No need a new one.

mattreim, 2025-05-12:
Here is a little help

closed 2025-05-17 by manup