dresden-elektronik/deconz-rest-plugin #8233: Rquest for SMOKE SENSOR MultiR MIR-SM100-E

jortdejong, 2025-05-16:
### Is there already an existing issue for this?

- [x] I have searched the existing issues and there is none for my device

### Product name

SmartWise Zigbee Smoke Sensor

### Manufacturer

MultIR

### Model identifier

MIR-SM100-E

### Device type to add

Sensor

### Node info

[image]

### Endpoints and clusters

[image]

### Basic

[image]

### Further relevant clusters

[image]

### On/Off

### Level Control

### Color Control

### Thermostat

### Simple Metering

### Electrical Measurement

### Any other cluster of relevance/interest

Smanar, 2025-06-12:
It have worked ?

jortdejong, 2025-06-12:
Cab you please do?