dresden-elektronik/deconz-rest-plugin #8239: Smart Sensor C3007 TZ-TR_F   Modell ZG-303Z  ZIGBEE 3.0

bereberlin, 2025-05-23:
Hallo, 

ich habe neue Feuchtigkeitssensoren für Blumentöpfe. 
Laut Antwort von dresdenElektronik:

 "Hallo Herr Madaus,
wir haben es also hier mit folgendem Gerät zu tun:
_TZE200_wqashyqo

TS0601
Im verlinkten DDF sind aber nur diese enthalten:

    "_TZE200_nnrfa68v",    "_TZE200_znbl8dj5",    "_TZE200_qoy0ekbd",    "_TZE200_a8sdabtg",    "_TZE200_whkgqxse",
    "_TZE200_yjjdcqsq",    "_TZE200_utkemkbs",    "_TZE200_locansqn",    "_TZE204_yjjdcqsq",    "_TZE200_vs0skpuc",    "_TZE284_wtikaxzs"

Das von Ihnen eingesetzte Gerät wird aktuell weder in den Kompatibilitätslisten geführt, noch findet es sich „in Arbeit“. Hier muss also ein neuer Request angelegt werden. Das geht wie hier beschrieben:[ https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Request-Device-Support](https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Request-Device-Support)

Die Entwickler-Community dürfte schnell ein passendes DDF erstellen können, wenn es den Geräten aus dem verlinkten DDF sehr ähnlich ist. Am besten im Issue den Link zum bestehenden DDF erwähnen...."
Bisher habe ich folgende Screenshots

[image]

[image]

[image]
Falls weitere Angabe erforderlich sind, kann ich damit dienen.

Mit freundlichen Grüßen

Bernd Madaus

brientim, 2025-06-06:
Hallo Smanar

This issue is for a Soil Moisture Sensor 

https://www.zigbee2mqtt.io/devices/ZG-303Z.html

Smanar, 2025-06-07:
Hello, this device don't use same dpid, can you try this DDF to start

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": [
    "_TZE200_wqashyqo"
  ],
  "modelid": [
    "TS0601"
  ],
  "vendor": "Tuya",
  "product": "Tuya Soil Sensor (TS0601)",
  "sleeper": true,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_TEMPERATURE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0402"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "config/tuya_unlock"
        },
        {
          "name": "attr/swversion",
          "parse": {
            "fn": "zcl:attr",
            "ep": 1,
            "cl": "0x0000",
            "at": "0x0001",
            "script": "tuya_swversion.js"
          },
          "read": {
            "fn": "zcl:attr",
            "ep": 1,
            "cl": "0x0000",
            "at": "0x0001"
          }
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "read": {
            "fn": "none"
          },
          "parse": {
            "dpid": 108,
            "eval": "Item.val = Attr.val;",
            "fn": "tuya"
          },
          "default": 0
        },
        {
          "name": "config/offset"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/temperature",
          "read": {
            "fn": "none"
          },
          "parse": {
            "dpid": 101,
            "eval": "Item.val = Attr.val * 10;",
            "fn": "tuya"
          },
          "default": 0,
          "awake": true
        }
      ]
    },
    {
      "type": "$TYPE_MOISTURE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0408"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "parse": {
            "fn": "zcl:attr",
            "ep": 1,
            "cl": "0x0000",
            "at": "0x0001",
            "script": "tuya_swversion.js"
          },
          "read": {
            "fn": "zcl:attr",
            "ep": 1,
            "cl": "0x0000",
            "at": "0x0001"
          }
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "read": {
            "fn": "none"
          },
          "parse": {
            "dpid": 108,
            "eval": "Item.val = Attr.val;",
            "fn": "tuya"
          },
          "default": 0
        },
        {
          "name": "config/offset"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/moisture",
          "read": {
            "fn": "none"
          },
          "parse": {
            "dpid": 107,
            "eval": "Item.val = Attr.val * 100;",
            "fn": "tuya"
          },
          "default": 0,
          "awake": true
        },
        {
          "name": "state/lastupdated"
        }
      ]
    }
  ]
}
```

pedolsky, 2025-06-17:
Maybe sitting in the same boat. Today I received a ZG-303Z „Smart Sensor Model: C3007“. Purely. visual, it is this one: https://www.zigbee2mqtt.io/devices/CS-201Z.html

Since the Phoscon App doesn’t let m upload a .json file, I glanced over the files in /usr/share/deCONZ/devices/ . 
@Smanar, I would like to test the DDF file above, should I follow a naming convention?

pedolsky, 2025-06-19:
> As DeCONZ creates the database records, even though the device is not visible, a ghost, in Phoscon, it should still the API can query the records and I have done this many times.
> 
> Therefore through Help > API (or others like Postman) the user should be able to see information.

That was my first thought too, but there's nothing to be seen there.

Smanar, 2025-07-04:
Yeah sensor are "lazy" device.
You need to awake it when you make the "read attribute" request 

https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/How-to-read-Clusters

Smanar, 2025-07-04:
Yes and no, after searching your device on google, it seem it'"s another shit from tuya.
You can forget cluster 0x0402 and 0x0405, your device work with tuya cluster the 0xEF00

It's s like this one https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/devices/fantem/zb003-x_multi_sensor.json
But with other dpid.

I will make a try later.

Smanar, 2025-07-04:
Can try this DDF

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZE200_npj9bug3",
  "modelid": "TS0601",
  "vendor": "Tuya",
  "product": "ZG-303Z",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_TEMPERATURE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0402"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "refresh.interval": 86400,
          "parse": {
            "fn": "zcl:attr",
            "ep": 1,
            "cl": "0x0000",
            "at": "0x0001",
            "script": "../tuya/tuya_swversion.js"
          },
          "read": {
            "fn": "zcl:attr",
            "ep": 1,
            "cl": "0x0000",
            "at": "0x0001"
          }
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "parse": {
            "fn": "tuya",
            "dpid": 15,
            "eval": "Item.val = Attr.val;"
          },
          "read": {
            "fn": "none"
          },
          "default": 0
        },
        {
          "name": "config/offset",
          "default": 0
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/temperature",
          "read": {
            "fn": "none"
          },
          "parse": {
            "dpid": 5,
            "eval": "Item.val = (10 * Attr.val) + R.item('config/offset').val;",
            "fn": "tuya"
          },
          "default": 0
        }
      ]
    },
    {
      "type": "$TYPE_HUMIDITY_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0405"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "refresh.interval": 86400,
          "parse": {
            "fn": "zcl:attr",
            "ep": 1,
            "cl": "0x0000",
            "at": "0x0001",
            "script": "../tuya/tuya_swversion.js"
          },
          "read": {
            "fn": "zcl:attr",
            "ep": 1,
            "cl": "0x0000",
            "at": "0x0001"
          }
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "parse": {
            "fn": "tuya",
            "dpid": 15,
            "eval": "Item.val = Attr.val;"
          },
          "read": {
            "fn": "none"
          },
          "default": 0
        },
        {
          "name": "config/offset",
          "default": 0
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/humidity",
          "read": {
            "fn": "none"
          },
          "parse": {
            "dpid": 109,
            "eval": "Item.val = (100 * Attr.val) + R.item('config/offset').val;",
            "fn": "tuya"
          },
          "default": 0
        },
        {
          "name": "state/lastupdated"
        }
      ]
    }
  ],
  "bindings": []
}
```

petersjostrom, 2025-07-08:
> You can't use cluster 0x0402 and 0x0405 for this device.
> 

What do you mean?

Smanar, 2025-07-09:
Just you can't use thoses clusters, this device don't use them but use instead the cluster 0xEF00

All you see on them is useless, take a look on the DDF, it don't use them.

closed 2025-08-07 by github-actions[bot]