dresden-elektronik/deconz-rest-plugin #8241: Add _TZE204_c2fmom5z  (TS0601)Air Box Smart Air Detector

brientim, 2025-05-27:
### Is there already an existing issue for this?

- [x] I have searched the existing issues and there is none for my device

### Product name

Tuya Air Box Smart Air Detector (Tuya clone)

### Manufacturer

TZE204_c2fmom5z 

### Model identifier

TS0601

### Device type to add

Sensor

### Node info

The TZE204_c2fmom5z Air Box Smart Air Detector (AQI) is another Tuya clone and .
It has the same capabilities and functions as  the Tuya _TZE200_dwcarsat Air Sensor 

This request is to append to _TZE200_dwcarsat_air_sensor.json
"manufacturername":  _TZE204_c2fmom5z
"modelid":  TS0601

[_TZE200_dwcarsat_air_sensor.json](https://github.com/Smanar/deconz-rest-plugin/blob/b865e0f35ef204e0f25a079206f6067795efd9dc/devices/tuya/_TZE200_dwcarsat_air_sensor.json#L4)

[image]

### Endpoints and clusters

[image]

### Basic

[image]

### Further relevant clusters

### Power Configuration

### On/Off

### Level Control

### Color Control

### Thermostat

### Simple Metering

### Electrical Measurement

### Any other cluster of relevance/interest

brientim, 2025-06-06:
Hi Smanar,

Thank you and that is fine, and always appreciate the support.  Can you please raise the PR.

I have commented on the other post and not related 
same as #8239, Soil Moisture Sensor

Smanar, 2025-06-07:
Thx, have created the PR > https://github.com/dresden-elektronik/deconz-rest-plugin/pull/8244

And have added some more clone I have found in same time.

closed 2025-06-23 by manup