dresden-elektronik/deconz-rest-plugin #8253: Aquara MCCGQ11LM, temperature info lost some minutes after pairing

openschwall, 2025-06-22:
### Does the issue really belong here?

- [x] I definitively want to report a bug within deCONZ or its REST-API

### Is there already an existing issue for this?

- [x] I have searched the existing issues and there is none for the bug at hand

### Describe the bug

Aquara window/door sensors MCCGQ11LM (https://dresden-elektronik.github.io/deconz-rest-doc/devices/xiaomi/lumi_sensor_magnet_aq2/#aqara-door-and-window-sensor-mccgq11lm) deliver temperature information in my home for years. Since some months, temperature information is no longer delivered. I rejoined sensors, delivering temperature for some minutes or hours only. Thenafter, no more temperature information.
I bought a new one with the same behaviour. 

Looking at REST API via http://192.168.0.163:8007/api/..../sensors/ I can see in the config field that it contains attributes battery, on, reachable  but no longer temperature as would be necessary from documentation.

Therefore, I assume that deconz "forgets" temperature some minutes after joining.

Any help greatly appreciated. 

### Steps to reproduce the behavior

Join this sensor and wait.

### Expected behavior

Temperature is delivered not only for some minutes/hours after joining these sensors.

### Environment

- Host system: Raspberry Pi 3b 
- Running method: Raspbian/Openhabian
- Firmware version: 26780700
- deCONZ version: 2.30.2
- Device: ConBee II
- Do you use an USB extension cable: yes 
- Is there any other USB or serial devices connected to the host system? If so: No

MSchmitt67, 2025-06-22:
Hallo,

its removed since [v2.29.0-beta](https://github.com/dresden-elektronik/deconz-rest-plugin/releases/tag/v2.29.0-beta)

**- Removal of config/temperature for various Xiaomi devices** https://github.com/dresden-elektronik/deconz-rest-plugin/pull/7827

Mimiix, 2025-06-22:
As stated above, not a bug.

Closing

closed 2025-06-22 by Mimiix