dresden-elektronik/deconz-rest-plugin #8266: Add DDF to support HOBEIAN ZG-204Z Motion Sensor

MarkRotNF, 2025-07-02:
### Is there already an existing issue for this?

- [x] I have searched the existing issues and there is none for my device

### Product name

Zigbee Motion Sensor Model ZM-P1

### Manufacturer

HOBEIAN

### Model identifier

ZG-204Z

### Device type to add

Sensor

### Node info

[image]

### Endpoints and clusters

[image]

### Basic

[image]
[image]

### Further relevant clusters

I got two ZM-P1 motion detectors (from different sellers) on Aliexpress, both from HOBEIAN. I was able to get them working in phoscon using the following DDF.

[zg-204z-motion-sensor.json](https://github.com/user-attachments/files/21024452/zg-204z-motion-sensor.json)

MarkRotNF, 2025-07-03:
Mine doesn't have a lux sensor. The DDF also doesn't seem to be working properly. It's displayed, but motion isn't reported (in Phoscon). (The battery level entry is also missing :D.)

[image]

I have another ZG-204Z that I haven't been able to get to work yet. Unfortunately, your DDF doesn't work with that one either.

MarkRotNF, 2025-07-08:
I'm still a little confused. I have a second one with the same label, but it doesn't seem to be identical. It doesn't work.

[image]

[image]

[image]

[image]

[Untitled Document 5-1.txt](https://github.com/user-attachments/files/21126954/Untitled.Document.5-1.txt)

Smanar, 2025-07-09:
I don't see tuya request on your logs, you have enabled both "info" flag ?
They are like that

```
17:13:10:234 writeTuyaData, dpid: 0x01, type: 0x10, expr: Item.val == 1 ? 1 : 0;
17:13:10:235 Tuya write expression: Item.val == 1 ? 1 : 0; --> 1
```

What is the difference beetween your device ? both are HOBEIAN/ZG-204Z ? if yes the DDF need to work same.
The enrollement can have failed, if after 24 you still have the issue, re-include it.

MarkRotNF, 2025-07-16:
[image]
Looks good. It detects movement and displays information like battery level.

Smanar, 2025-07-17:
Thoses device are generaly more presence sensor than motion sensor but I don't understand

> If you move and stay in front of the sensor until it no longer detects movement

But if you are still moving state/presence" stay at true ?

MarkRotNF, 2025-07-29:
Two scenarios:
1. I stand in front of the sensor -> it triggers. I stay in front of the sensor and move until it no longer detects movement. I continue moving. However, it no longer detects movement and doesn't trigger again.

2. I walk past the sensor -> it triggers. I wait until it no longer detects movement and then walk past the sensor (about 1 second after it no longer detects movement). It triggers again immediately.

So it seems that if it doesn't trigger again if you stand "continuously" in front of the sensor.

Smanar, 2025-07-29:
Ok so we probably need to remove config/duration.
If you are still moving, the device need to stay on presence  = true.
Some device trigger at every detection but stay on presence = true, you just have a new notification, but you will never have presence= false.

Perhaps it's config/delay ? I m not able to remember the trick with thoses settings ....