dresden-elektronik/deconz-rest-plugin #8269: Add DDF to support Tuya 6-Gang Switch

Maik1, 2025-07-11:
### Is there already an existing issue for this?

- [x] I have searched the existing issues and there is none for my device

### Product name

Tuya 6-Gang Switch 

### Manufacturer

_TZ3000_nrfkrgf4

### Model identifier

TS0046

### Device type to add

Switch

### Node info

[image]

### Endpoints and clusters

[image]

### Basic

[image]

### Further relevant clusters

### Power Configuration
[image]

### On/Off

[image]

[image]

[image]

[image]

[image]

### Level Control

### Color Control

### Thermostat

### Simple Metering

### Electrical Measurement

### Any other cluster of relevance/interest

https://de.aliexpress.com/item/1005007922384796.html?spm=a2g0o.order_list.order_list_main.5.10a55c5fI5YBGs&gatewayAdapt=glo2deu

Smanar, 2025-07-12:
6 gangs .... with only 4 endpoint ?
Can start with that

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZ3000_nrfkrgf4",
  "modelid": "TS0046",
  "vendor": "Tuya",
  "product": "6-gang remote (TS0046)",
  "comment": "Locked by Tuya",
  "sleeper": true,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_SWITCH",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0006"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "parse": {
            "fn": "zcl:attr",
            "ep": 1,
            "cl": "0x0000",
            "at": "0x0001",
            "script": "tuya_swversion.js"
          },
          "read": {
            "fn": "zcl:attr",
            "ep": 1,
            "cl": "0x0000",
            "at": "0x0001"
          }
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "refresh.interval": 86400,
          "read": {
            "fn": "zcl:attr",
            "ep": 1,
            "cl": "0x0001",
            "at": "0x0021"
          },
          "parse": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "eval": "Item.val = Attr.val / 2;",
            "fn": "zcl:attr"
          }
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/tuya_unlock"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/buttonevent"
        },
        {
          "name": "state/lastupdated"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0006"
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0021",
          "dt": "0x20",
          "min": 64500,
          "max": 65500,
          "change": "0x00000002"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 2,
      "dst.ep": 1,
      "cl": "0x0006"
    },
    {
      "bind": "unicast",
      "src.ep": 3,
      "dst.ep": 1,
      "cl": "0x0006"
    },
    {
      "bind": "unicast",
      "src.ep": 4,
      "dst.ep": 1,
      "cl": "0x0006"
    }
  ]
}
```

You will miss event for sure, need to edit the buttonmap.json file but idk with what.
So enable logging on deconz/help/debug view with flag "info" and "info_l2", and press not working button.
You will see an error message about missing buttonmap, can you share them pls.

Maik1, 2025-07-12:
Awesome, that worked really well.

My ddf file: [_TZ3000_TS0046_6gang_remote.json](https://github.com/user-attachments/files/21195248/_TZ3000_TS0046_6gang_remote.json)

I added the following to the button-maps.json to hafe short, double and long press events.
```
"Tuya6gangMap": {
  "vendor": "Tuya",
  "doc": "TS0046 6-gang remote",
  "modelids": ["_TZ3000_nrfkrgf4"],
  "map": [
    [1, "0x01", "ONOFF", "0xfd", "0", "S_BUTTON_1", "S_BUTTON_ACTION_SHORT_RELEASED", "B1 short"],
    [1, "0x01", "ONOFF", "0xfd", "1", "S_BUTTON_1", "S_BUTTON_ACTION_DOUBLE_PRESS", "B1 double"],
    [1, "0x01", "ONOFF", "0xfd", "2", "S_BUTTON_1", "S_BUTTON_ACTION_LONG_RELEASED", "B1 long"],

    [1, "0x02", "ONOFF", "0xfd", "0", "S_BUTTON_2", "S_BUTTON_ACTION_SHORT_RELEASED", "B2 short"],
    [1, "0x02", "ONOFF", "0xfd", "1", "S_BUTTON_2", "S_BUTTON_ACTION_DOUBLE_PRESS", "B2 double"],
    [1, "0x02", "ONOFF", "0xfd", "2", "S_BUTTON_2", "S_BUTTON_ACTION_LONG_RELEASED", "B2 long"],

    [1, "0x03", "ONOFF", "0xfd", "0", "S_BUTTON_3", "S_BUTTON_ACTION_SHORT_RELEASED", "B3 short"],
    [1, "0x03", "ONOFF", "0xfd", "1", "S_BUTTON_3", "S_BUTTON_ACTION_DOUBLE_PRESS", "B3 double"],
    [1, "0x03", "ONOFF", "0xfd", "2", "S_BUTTON_3", "S_BUTTON_ACTION_LONG_RELEASED", "B3 long"],

    [1, "0x04", "ONOFF", "0xfd", "0", "S_BUTTON_4", "S_BUTTON_ACTION_SHORT_RELEASED", "B4 short"],
    [1, "0x04", "ONOFF", "0xfd", "1", "S_BUTTON_4", "S_BUTTON_ACTION_DOUBLE_PRESS", "B4 double"],
    [1, "0x04", "ONOFF", "0xfd", "2", "S_BUTTON_4", "S_BUTTON_ACTION_LONG_RELEASED", "B4 long"],

    [1, "0x05", "ONOFF", "0xfd", "0", "S_BUTTON_5", "S_BUTTON_ACTION_SHORT_RELEASED", "B5 short"],
    [1, "0x05", "ONOFF", "0xfd", "1", "S_BUTTON_5", "S_BUTTON_ACTION_DOUBLE_PRESS", "B5 double"],
    [1, "0x05", "ONOFF", "0xfd", "2", "S_BUTTON_5", "S_BUTTON_ACTION_LONG_RELEASED", "B5 long"],

    [1, "0x06", "ONOFF", "0xfd", "0", "S_BUTTON_6", "S_BUTTON_ACTION_SHORT_RELEASED", "B6 short"],
    [1, "0x06", "ONOFF", "0xfd", "1", "S_BUTTON_6", "S_BUTTON_ACTION_DOUBLE_PRESS", "B6 double"],
    [1, "0x06", "ONOFF", "0xfd", "2", "S_BUTTON_6", "S_BUTTON_ACTION_LONG_RELEASED", "B6 long"]
  ]
},

```

Smanar, 2025-07-12:
And all is working ?
So it mean the GUI don't display the endpoint 0x06 and 0x05 ?

Do you want to make a PR to submit the device offcialy ?

Smanar, 2025-07-12:
For phoscon I can understand, but for deconz for me there is a problem, but can be fom the device.
BTW thx for PR ^^

closed 2025-07-31 by manup

Smanar, 2025-09-16:
Never tried on my side, but this can help you https://dresden-elektronik.github.io/deconz-rest-doc/endpoints/ddf/#automatically-with-the-github-action

HunZ83, 2025-09-17:
Thank you, it worked fine with the ddb-file! Now, I can add the new switch to deconz and also find the battery state (surprisingly, I only have one button instead of four, like Mike1).

[image]

Though, I´m still missing the events in HA. I started listening to "zha_event", but nothing swos up, when I press the buttons on the switch. 
Do I also need to add the information above to the button-maps.json? I´m sorry to ask, but I have no clue, how to do that via Home assistant, since I have no access to that file (I know where it is, but not, how to open/edit it).

Smanar, 2025-09-18:
But the ddb file don't contain the DDF AND the change on button-maps.json file ?

HunZ83, 2025-10-06:
I don´t think so or I just don´t know, how to do it. 
I guess, I´ll have to wait  until the next stable version is available for HomeAssistant. 
I understood that the device should be fully supported in it, right?
Thank you very much for the support, anyway!

HunZ83, 2025-10-29:
> I don´t think so or I just don´t know, how to do it. I guess, I´ll have to wait until the next stable version is available for HomeAssistant. I understood that the device should be fully supported in it, right? Thank you very much for the support, anyway!

Update: With the release of the deconz-Update in HomeAssistant, everything works fine with the mentioned switch, now. All events are visible in HomeAssistant and I can use them in different automations.

PR #8270 (2025-07-31): DDF for Tuya TS0046 6-gang switch
Add full support for Tuya TS0046 6-gang switch with model `_TZ3000_nrfkrgf4`.

Closes #8269