dresden-elektronik/deconz-rest-plugin #8278: Yooksmart D10110 Displayed as a Battery vs Window Covering

jcarusone1650, 2025-07-21:
### Does the issue really belong here?

- [x] I definitively want to report a bug within deCONZ or its REST-API

### Is there already an existing issue for this?

- [x] I have searched the existing issues and there is none for the bug at hand

### Describe the bug

Looks like the Yooksmart D10110 window covering is now being detected as a battery vs a window covering as it was in prior versions. 

### Steps to reproduce the behavior

1- Pair devices in Phoscon
2- Devices show as "Battery" in Deconz

### Expected behavior

This are window coverings should be reflected as such.

### Screenshots

[image]

### Environment

- Host system:Mac
- Running method: DMG
- Firmware version: (26xxyy00)
- deCONZ version 2.30.2
- Device: ConBee III 
- Do you use an USB extension cable: No
- Is there any other USB or serial devices connected to the host system? If so: No

jcarusone1650, 2025-07-28:
@Smanar I attempted to produce a DDF for this one but I am unable to open in Deconz so I suspect I messed something up. Happy to test/open PR if you have any ideas. I worked off the DDF you created several years ago for this device.

[Yooksmart.json](https://github.com/user-attachments/files/21458993/Yooksmart.json)

Smanar, 2025-07-29:
>however "open" reports as closed and closed reports as "closed"

2 time "closed" ?
There is 2 possiblities, a reverse command, or reverse it in the DDF.
If ONLY the return is reversed and not the movement, probalby a problem in the DDF

Can replace
```

        {
          "name": "state/open"
        },
```
by

```
        {
          "name": "state/open",
          "parse": {
            "at": "0x0008",
            "cl": "0x0102",
            "ep": 1,
            "eval": "Item.val = Attr.val < 100",
            "fn": "zcl:attr"
          }
        },
```

or the reverse

```
        {
          "name": "state/open",
          "parse": {
            "at": "0x0008",
            "cl": "0x0102",
            "ep": 1,
            "eval": "Item.val = Attr.val > 0",
            "fn": "zcl:attr"
          }
        },
```

jcarusone1650, 2025-07-29:
Thanks. I tried both and the open/closed status still reports incorrectly which seems odd. Anything I can provide from Deconz UI that would be helpful? If i use 99% and 1% vs open and closed everything works correctly.

Smanar, 2025-07-29:
But what are you calling open/closed ?
You are speaking about state/open ? and you need to have state/open = true or false.

Or you mean when you make the order with open = false, the device work but in wrong side ?

Smanar, 2025-07-30:
Ha right, can be a problem.

closed 2025-08-29 by github-actions[bot]