dresden-elektronik/deconz-rest-plugin #8282: HZC T11_ZG / Namron 4512783 Thermostat

phuongpham, 2025-07-27:
### Does the issue really belong here?

- [x] I definitively want to report a bug within deCONZ or its REST-API

### Is there already an existing issue for this?

- [x] I have searched the existing issues and there is none for the bug at hand

### Describe the bug

I'm currently try to build DDF file for this thermostat.
I have managed to get most of the information.

But I'm missing 1 cluster in deconz. Based on the documentation it should also having Private Thermostat cluster. But somehow it's not available. https://www.elektroimportoren.no/namron//4512783-84-Zigbee-User-Guide.pdf

[image]

The solutions at: https://github.com/Koenkk/zigbee2mqtt/issues/24047

My current ddf file: 

[4512783_thermostat.json](https://github.com/user-attachments/files/21455933/4512783_thermostat.json)

### Steps to reproduce the behavior

Open deconz gui and expand the device.

### Expected behavior

The Private Thermostat cluster is available.

### Environment

- Host system: (Raspberry Pi)
- Running method: (Raspbian)
- Firmware version: (26780700)
- deCONZ version (not Home assistant Addon version!): (2.30.2)
- Device: (ConBee II)
- Do you use an USB extension cable: (yes)-- only relevant for ConBee I/II/III

mattreim, 2025-07-27:
With this DDF it should work for now.

[4512783_thermostat.json](https://github.com/user-attachments/files/21456800/4512783_thermostat.json)

phuongpham, 2025-07-29:
> No, not the cluster 0x0204, but the 0x0201. On Z2M "hvacThermostat" is the classic thermostat Cluster, the 0x0201

From 0201 cluster:
https://github.com/user-attachments/assets/499f9ef7-77a3-461f-a350-29e07582f7d2

phuongpham, 2025-07-29:
ok I have to look into this tommrow. Thanks for your help so far
the mfcode is: 0x126a. 
Managed to get chatGPT to generate all attributes based on the pdf file :)

closed 2025-08-28 by github-actions[bot]