dresden-elektronik/deconz-rest-plugin #8298: eWeLink CK-TLSR8656-SS5-01(7014) temperature humidity sensor

heyajohnny, 2025-08-04:
### Is there already an existing issue for this?

- [x] I have searched the existing issues and there is none for my device

### Product name

Tuya Wifi/Zigbee Temperature And Humidity Sensor Indoor Thermometer For Home Work With Alexa Google Home Assistant gateway

### Manufacturer

eWeLink

### Model identifier

CK-TLSR8656-SS5-01(7014)

### Device type to add

Sensor

### Node info

[image]

### Endpoints and clusters

[image]

### Basic

[image]

### Further relevant clusters

### Power Configuration

### On/Off

### Level Control

### Color Control

### Thermostat

### Simple Metering

### Electrical Measurement

### Any other cluster of relevance/interest

[Aliexpress product link](https://nl.aliexpress.com/item/1005008795838359.html?spm=a2g0o.order_list.order_list_main.26.5c6f79d2psNeiu&gatewayAdapt=glo2nld)

[image]

heyajohnny, 2025-08-05:
Adding the manufacturername and modelid to[ snzb-02-multisensor.json](https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/devices/sonoff/snzb-02-multisensor.json) seems to work, but it feels a bit off because of the modelid that isn't something like 'SNZB...', so i don't know if it it's the right ddf file to add it to.

[image]

[image]

mattreim, 2025-08-06:
I have now added the sensor.

heyajohnny, 2025-08-06:
@mattreim Thnx!