dresden-elektronik/deconz-rest-plugin #8301: SNZB-03 PIR Motion sensor detected as Water Leak Sensor

saxn-paule, 2025-08-06:
### Does the issue really belong here?

- [x] I definitively want to report a bug within deCONZ or its REST-API

### Is there already an existing issue for this?

- [x] I have searched the existing issues and there is none for the bug at hand

### Describe the bug

I paired a new eWeLink motion sensor. It is detected as eWeLink Water Leak Sensor SNZB-03 and doesn't react on motions.

### Steps to reproduce the behavior

Pair sensor

### Expected behavior

Detected as motion sensor

### Screenshots

[image]

### Environment

- Host system: NUC6 with Debian
- Running method: deconz-docker container
- Firmware version: 26550900
- deCONZ version: 2.30.2
- Device: ConBee III
- Do you use an USB extension cable: yes
- Is there any other USB or serial devices connected to the host system?: No

mattreim, 2025-08-06:
Please attach some deconz screenshots (e.g. tlike [this](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/8283))

saxn-paule, 2025-08-06:
My NUC is running in headless mode with no X-Server installed. How could I get the requested screenshots?

mattreim, 2025-08-06:
You can either enable VNC in docker or use Phoscon backup on a Windows PC.

saxn-paule, 2025-08-06:
VNC is already enabled in docker-compose.yml and the port also exposed to the host. But no connection possible.
No direkt connection possible to docker host on port 5959 and also no connection possible when tunneling the port through a SSH session and connection to localhost.

[image]

saxn-paule, 2025-08-06:
[image]

[image]

[image]

[image]

mattreim, 2025-08-06:
@manup @SwoopX @Smanar @ebaauw Hello everyone, there seem to be two devices with the same manufacturer and the same model ID. Perhaps one of you has an idea where the problem is.

- Motion sensor: [eWelink/SNZB-03](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/8301)
- Water sensor: [eWelink/SNZB-03](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/7911)

Smanar, 2025-08-07:
Yeah, they are doing ****.

Same Manufacture name, same model id, same endpoint. I think even them are not able to differentiate their devices.

saxn-paule, 2025-08-07:
[image]

ebaauw, 2025-08-08:
Indeed, attribute 0x0001 shows _Motiin Sensor_.  I suspect the water sensor will show a different value.  You might want to double check supervision reports and restore reports, once the device has enrolled.  If it doesn’t support supervision reports, you might try attribute reporting instead.  If it doesn’t support restore reports, you need to use `config/duration` to let the REST API plugin reset `state/presence`.

tpzltn, 2025-08-15:
Just got the same motion sensor today and identifies as water sensor as reported by OP with ConbeeII. Is there any fix available? Thanks.

saxn-paule, 2025-10-05:
Is there any update on this? Could you please reopen the issue?

saxn-paule, 2025-10-30:
@ebaauw are you able to provide a corresponding DDF with your findings?

saxn-paule, 2025-11-25:
Up

Smanar, 2025-11-25:
@saxn-paule you have the motion sensor ?

I m seing the device is already in tuya\ts0202_presence_sensor.json so the problem is the water sensor
So edit sonoff\snzb-03_water_leak_sensor.json to have

```
  "modelid": "SNZB-03",
  "vendor": "Sonoff/eWeLink",
  "product": "water sensor (SNZB-03)",
  "matchexpr": "R.item('attr/zonetype').val === 0x002a",
  "sleeper": true,
```

With this code the DDF will be used ONLY if the device type is 

`<value name="Water sensor" value="0x002a"></value>`

If it break the water sensor DDF, we just need to wait for someone share the good type for it.

But to make something perfect it's better to remove the snzb-03 in the tuya DDF ts0202_presence_sensor.json and make a specific DDF for him

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "eWeLink",
  "modelid": "SNZB-03",
  "vendor": "Sonoff/eWeLink",
  "product": "Motion sensor (SNZB-03)",
  "matchexpr": "R.item('attr/zonetype').val === 0x000d",
  "sleeper": true,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_PRESENCE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0500"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0402",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0001",
          "0x0500"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "parse": {
            "fn": "zcl:attr",
            "ep": 1,
            "cl": "0x0000",
            "at": "0x0001",
            "script": "tuya_swversion.js"
          },
          "read": {
            "fn": "zcl:attr",
            "ep": 1,
            "cl": "0x0000",
            "at": "0x0001"
          }
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "refresh.interval": 86400,
          "read": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "fn": "zcl:attr"
          },
          "parse": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "eval": "Item.val = Attr.val/2"
          },
          "default": 0
        },
        {
          "name": "config/duration"
        },
        {
          "name": "config/enrolled",
          "public": false
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/pending"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/lowbattery"
        },
        {
          "name": "state/presence",
          "awake": true,
          "parse": {
            "fn": "ias:zonestatus",
            "mask": "alarm1"
          }
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0500"
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0021",
          "dt": "0x20",
          "min": 3600,
          "max": 14400,
          "change": "0x00000001"
        }
      ]
    }
  ]
}
```

This DDF will ONLY used by the motion sensor.

saxn-paule, 2025-11-25:
I tried the DDF but it doesn't work as expected.

I deleted the device and repaired it.

Via VNC it shows a motion sensor

[image]

But the API, Phoscon and Home-Assistant show a Water sensor:

[image]

mattreim, 2025-11-30:
Hi everyone, since there's currently no other solution for this problem, I wanted to ask if I should still do a pull request with the DDFs or if we should wait a bit longer?

mattreim, 2025-12-04:
@saxn-paule The issue has been fixed in v2.32.2-beta and can now be closed.

closed 2025-12-04 by saxn-paule