dresden-elektronik/deconz-rest-plugin #8304: Tuya Motion Sensor TS0601

saxn-paule, 2025-08-06:
### Is there already an existing issue for this?

- [x] I have searched the existing issues and there is none for my device

### Product name

Tuya Motion Sensor

### Manufacturer

_TZE200_bh3n6gk8

### Model identifier

TS0601

### Device type to add

Sensor

### Node info

[image]

### Endpoints and clusters

[image]

### Basic

[image]

### Further relevant clusters

### Power Configuration

### On/Off

### Level Control

### Color Control

### Thermostat

### Simple Metering

### Electrical Measurement

### Any other cluster of relevance/interest

mattreim, 2025-08-06:
Here we can only hope that it's a Tuya clone. I've extended a DDF for that.

[pir_motion_detector_and_light_sensor.zip](https://github.com/user-attachments/files/21628968/pir_motion_detector_and_light_sensor.zip)

saxn-paule, 2025-08-06:
Yes, this worked. Thank you very much for your fast support 

[image]

closed 2025-08-06 by saxn-paule

mattreim, 2025-08-06:
@saxn-paule Are you sure? This is important, because otherwise you'd need a separate DDF.