dresden-elektronik/deconz-rest-plugin #8327: Water Leak Sensor - HOBEIAN ZG-222Z

Crayjin, 2025-09-10:
### Is there already an existing issue for this?

- [x] I have searched the existing issues and there is none for my device

### Product name

Zigbee Smart Water Sensor Flood Leak Detector Model: C3007

### Manufacturer

HOBEIAN

### Model identifier

ZG-222Z

### Device type to add

Sensor

### Node info

[image]

### Endpoints and clusters

[image]

### Basic

[image]

### Further relevant clusters

### Power Configuration

[image]

### Any other cluster of relevance/interest

#### IAS Zone

[image]

mattreim, 2025-09-13:
It might be a clone of this `Sonoff Zigbee Water Leak Sensor (SNZB-05P)` device. Please give it a try.

[snzb-05p_water_leak_sensor.json](https://github.com/user-attachments/files/22314916/snzb-05p_water_leak_sensor.json)

mattreim, 2025-10-02:
@Crayjin The PR has now been merged, and the device is now supported. Therefore, this issue can be closed. Enjoy your device.

closed 2025-11-09 by Crayjin