dresden-elektronik/deconz-rest-plugin #8329: BSEED ZigBee

GitMensch, 2025-09-15:
### Is there already an existing issue for this?

- [x] I have searched the existing issues and there is none for my device

### Product name

BSEED ZigBee EU Wall Sockets Type-C With USB With Energy Monitoring

### Manufacturer

_TZ3210_5ct6e7ye

### Model identifier

TS011F

### Device type to add

None

### Node info

[image]

### Endpoints and clusters

[image]

### Basic

[image]

### Further relevant clusters

### Simple Metering
seems to be "empty"
[image]

### Electrical Measurement
[image]

### On/Off
[image]
[image]
side note: all but the last attributes for back light (a small LED next to a manual on/off switch) work
[image]

## Tuya
[image]

### OTAU
[image]

mattreim, 2025-09-18:
It could also be a `BlitzWolf` clone. Please test this DDF.

[bw_shp13_smart_plug.json](https://github.com/user-attachments/files/22412004/bw_shp13_smart_plug.json)

mattreim, 2025-10-02:
@GitMensch The PR has now been merged, and the device is now supported. Therefore, this issue can be closed. Enjoy your device.

GitMensch, 2025-10-03:
The current json file from master works fine!

It is a bit sad that those sockets only reports the momentarily use from some wH - a stand lamp with LED does count to the "overall use since installation", but is not measured otherwise (quite different to the INSPELNING plugs, which I really like - but those aren't sockets and are much less fancy).

GitMensch, 2025-10-26:
closed as supported (not sure if there are any docs to update)