dresden-elektronik/deconz-rest-plugin #8333: Eurotronic Spirit, DDF parse for valve is ignored

LukasA83, 2025-09-21:
### Does the issue really belong here?

- [x] I definitively want to report a bug within deCONZ or its REST-API

### Is there already an existing issue for this?

- [x] I have searched the existing issues and there is none for the bug at hand

### Describe the bug

I created the following DDF, but the valve value is still reported as 0-255 in REST-API, instead of 0-100. When I set deconz to "strict" mode, then it works. I see no error in the logs, the DDF logs even report the correct value, but it seems to be "overruled" or so.

`{
  "schema": "devcap1.schema.json",
  "manufacturername": "Eurotronic",
  "modelid": "SPZB0001",
  "vendor": "Eurotronic",
  "product": "Thermostat Spirit",  
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_THERMOSTAT",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0201"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0301",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0001",
          "0x0201"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "parse": {
            "at": "0x0021",
            "cl": "0x0001",
            "cppsrc": "power_configuration.cpp:196",
            "ep": 1,
            "eval": "Item.val = Attr.val / 2"
          },
          "default": 0
        },
        {
          "name": "config/displayflipped",
          "default": false
        },
        {
          "name": "config/heatsetpoint",
          "description": "Target temperature of a thermostat.",
          "read": {
            "at": "0x0012",
            "cl": "0x0201",
            "ep": 0,
            "fn": "zcl:attr"
          },
          "write": {
            "at": "0x0012",
            "cl": "0x0201",
            "dt": "0x29",
            "ep": 0,
            "eval": "Item.val;",
            "fn": "zcl:attr"
          },
          "parse": {
            "at": "0x0012",
            "cl": "0x0201",
            "ep": 0,
            "eval": "Item.val = Attr.val;",
            "fn": "zcl:attr"
          },
          "default": 0
        },
        {
          "name": "config/hostflags",
          "public": false
        },
        {
          "name": "config/locked",
          "description": "Locks or unlocks the device. This item has to be understood as child-lock.",
          "read": {
            "at": "0x0001",
            "cl": "0x0204",
            "ep": 0,
            "fn": "zcl:attr"
          },
          "write": {
            "at": "0x0001",
            "cl": "0x0204",
            "dt": "0x30",
            "ep": 0,
            "eval": "Item.val",
            "fn": "zcl:attr"
          },
          "parse": {
            "at": "0x0001",
            "cl": "0x0204",
            "ep": 0,
            "eval": "Item.val = Attr.val;",
            "fn": "zcl:attr"
          },
          "default": false
        },
        {
          "name": "config/mode"
        },
        {
          "name": "config/offset",
          "description": "Relative offset to the main measured value.",
          "default": 0
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/on",
          "description": "True when device is on; false when off.",
          "refresh.interval": 5,
          "read": {
            "at": "0x0000",
            "cl": "0x0006",
            "ep": 0,
            "fn": "zcl:attr"
          },
          "parse": {
            "at": "0x0000",
            "cl": "0x0006",
            "ep": 0,
            "eval": "Item.val = Attr.val",
            "fn": "zcl:attr"
          },
          "default": false
        },
        {
          "name": "state/temperature",
          "parse": {
            "at": "0x0000",
            "cl": "0x0402",
            "ep": 0,
            "eval": "Item.val = Attr.val + R.item('config/offset').val",
            "fn": "zcl:attr"
          },
          "default": 0
        },
        {
          "name": "state/valve",
          "read": {
            "fn": "none"
          },
          "parse": {
            "at": "0x0008",
            "cl": "0x0201",
            "fn": "zcl:attr",
            "eval": "Item.val = Attr.val / 255 *100"
          },
          "default": 0
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0021",
          "dt": "0x20",
          "min": 3600,
          "max": 43200,
          "change": "0x00000002"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0201",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x29",
          "min": 1,
          "max": 600,
          "change": "0x00000014"
        },
        {
          "at": "0x0008",
          "dt": "0x20",
          "min": 1,
          "max": 600,
          "change": "0x00000001"
        }
      ]
    }
  ]
}
`

### Steps to reproduce the behavior

1. Connect Eurotronic Spirit
2. Add DDF, see above
3. Set thermostat temperature to a high value, e.g. 30°C
4. Check reported value of "state/valve" in REST API

### Expected behavior

Use DDF and report correct value in REST-API.

### Environment

- Host system: Raspberry Pi - docker
- Running method: deconz-docker container
- Firmware version: 26780700
- deCONZ version 2.31.1
- Device:  RaspBee II

ebaauw, 2025-09-21:
If memory serves, the Spirit (also?) used a manufacturer-specific attribute to report the value position.  My bet would be that the legacy code still populates `state/valve` from that attribute.

LukasA83, 2025-09-21:
If I put deconz to "strict" mode it works. On VNC you can see the attribute in cluster 0201 under attribute 0008.

I am not familiar with what you mentioned as _bis item, can you please explain?

Smanar, 2025-09-22:
I don't found the PDF, the previous link is dead but from the XML (you can see them on the GUI too)

```
        <!-- Eurotronic manufacturer specific -->
        <attribute-set id="0x4000" description="Eurotronic" mfcode="0x1037">
          <attribute id="0x4000" name="TRV Mode" type="enum8" default="0x02" access="rw" required="m" mfcode="0x1037">
            <value name="Unknown 1" value="0"></value>
            <value name="Unknown 2" value="1"></value>
            <value name="manual" value="2"></value>
          </attribute>
          <attribute id="0x4001" name="Set Valve Position" type="u8" default="0x00" access="rw" required="m" mfcode="0x1037"></attribute>
          <attribute id="0x4002" name="Errors" type="u8" default="0x00" access="r" required="m" mfcode="0x1037">
            <value name="Reserved" value="0"></value>
            <value name="Reserved" value="1"></value>
            <value name="Reserved" value="2"></value>
            <value name="Valve Adaption Failed (E1)" value="3"></value>
            <value name="Valve Movement too slow (E2)" value="4"></value>
            <value name="Valve not Moving (E3)" value="5"></value>
            <value name="Reserved" value="6"></value>
            <value name="Reserved" value="7"></value>
          </attribute>
          <attribute id="0x4003" name="Current Temperature Setpoint" type="s16" default="0x07d0" access="rw" required="m" mfcode="0x1037"></attribute>
          <attribute id="0x4008" name="Host Flags" type="u24" default="0x000000" showas="hex" access="rw" required="m" mfcode="0x1037"></attribute>
        </attribute-set>
```

ebaauw, 2025-09-22:
Stil have them, in German.  Not sure which one is the latest version.

[Spirit_ZigBee_BAL_web_DE_view_V9.pdf](https://github.com/user-attachments/files/22475020/Spirit_ZigBee_BAL_web_DE_view_V9.pdf)

[Spirit_ZigBee_BAL_web_DE_Okt.-2019.pdf](https://github.com/user-attachments/files/22475026/Spirit_ZigBee_BAL_web_DE_Okt.-2019.pdf)

LukasA83, 2025-09-26:
Thanks, that did the trick. Here is a complete DDF including locking the devices and display flip. Only missing thing is where to map the valve setting in the rest api? For testing I used "config/externalsensortemp" but that is obviously wrong. Please advise, so we can finalize and get the legacy code removed :)

`{
  "schema": "devcap1.schema.json",
  "manufacturername": "Eurotronic",
  "modelid": "SPZB0001",
  "vendor": "Eurotronic",
  "product": "Thermostat Spirit",
  "sleeper": false,
  "status": "Gold",
  "path": "/devices/spzb0001.json",
  "subdevices": [
    {
      "type": "$TYPE_THERMOSTAT",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0201"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0301",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0001",
          "0x0201"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "parse": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "eval": "Item.val = Attr.val / 2",
            "fn": "zcl:attr"
          },
          "default": 0
        },
        {
[... 311 lines cut ...]
          "at": "0x4000",
          "dt": "0x30",
          "min": 1,
          "max": 600,
          "change": "0x00000001"
        },
        {
          "at": "0x4001",
          "dt": "0x20",
          "mf": "0x1037",
          "min": 1,
          "max": 600,
          "change": "0x00000001"
        },
        {
          "at": "0x4008",
          "dt": "0x22",
          "mf": "0x1037",
          "min": 1,
          "max": 600
        }
      ]
    }
  ]
}`

SwoopX, 2025-09-27:
I've created a DDF for this device ages ago, but couldn't really tedt it as I don't use it anymore.

Happy to see if I can find it agan and share it. It includes all the bitmap magic and device specific functionality as I recall.

LukasA83, 2025-09-27:
My DDF is also fully functional. Can we now get the legacy code removed so the valve reporting bug (not apply the parse expression) gets fixed, please?

SwoopX, 2025-09-28:
That JSON blob is hard to go through on a mobile screen when you rush over it 😉 Regardless, nice work and it is pretty identical to what I've put together back in the days.

However, a couple or remarks:

- Please remove `config/preset` as it isn't exposed by legacy code
- For setting the heatsetpoint (`state/temperature`), we have kind of an issue as older versions of the thermostat **only** work with attribute 0x4003 while newer versions apparently only work with attribute 0x0012. There is unfortunately no way to differentiate, so the DDF would break working devices
- I read somewhere that the device can also report when the window is open, this code was advertised:
```
        {
          "name": "state/windowopen",
          "parse": {
            "at": "0x4008",
            "cl": "0x0201",
            "ep": 1,
            "eval": "Item.val = (R.item('config/hostflags').val >> 3) & 0x01;",
            "fn": "zcl",
            "mf": "0x1037"
          },
          "read": {
            "fn": "none"
          },
          "default": false
        }
```
- Please also include the thermostats error reporting
```
        {
          "name": "state/errorcode",
          "refresh.interval": 3660,
          "parse": {
            "at": "0x4002",
            "cl": "0x0201",
            "ep": 1,
            "eval": "var res = ''; if (Attr.val & 0x08) { res = 'E1'; } else if (Attr.val & 0x10) { res = 'E2'; } else if (Attr.val & 0x20) { res = 'E3'; } else { res = 'none'; } Item.val = res;",
            "fn": "zcl",
            "mf": "0x1037"
          },
          "read": {
            "at": "0x4002",
            "cl": "0x0201",
            "ep": 1,
            "fn": "zcl",
            "mf": "0x1037"
          },
          "default": "none"
        }
```
- Based on the suggestions above, the following reporting changes would be advised:

1.  - Remove 0x0010 (not reportable, but it will be polled hourly)
2.  - Remove 0x4000
3.  - Remove 0x4001
4.  - Add 0x4002 with 1, 600
5.  - Add 0x4003 with 1, 600

SwoopX, 2025-09-28:
As for `state/valve` and the "parsing issue", that is more or less a safeguard. Many thermostat DDFs do not reflect what has been put in the code back in the days. The thermostat cluster is kind of sensitive in that regard, so legacy code is only skipped when going into strict mode https://github.com/dresden-elektronik/deconz-rest-plugin/blob/e8606cf98f125349856e0083bb7775ea2185ec2d/de_web_plugin.cpp#L1321

mattreim, 2025-09-30:
Hello everyone, I would like to contribute here as this is my last device without DDF. @SwoopX's suggested changes are already included, and the formatting makes it much easier to read. Please check again to see if everything is included.

For example, isn't this also necessary?
```lua
      "meta": {
        "values": {
          "config/mode": {"auto": 0, "heat": 1, "off": 5}
        }
      },
```

DDF:
```lua
{
    "schema": "devcap1.schema.json",
    "manufacturername": "Eurotronic",
    "modelid": "SPZB0001",
    "vendor": "Eurotronic",
    "product": "Spirit radiator thermostat",
    "sleeper": false,
    "status": "Gold",
    "subdevices": [
        {
            "type": "$TYPE_THERMOSTAT",
            "restapi": "/sensors",
            "uuid": [
                "$address.ext",
                "0x01",
                "0x0201"
            ],
            "fingerprint": {
                "profile": "0x0104",
                "device": "0x0301",
                "endpoint": "0x01",
                "in": [
                    "0x0000",
                    "0x0001",
                    "0x0201"
                ]
            },
            "items": [
                {
                    "name": "attr/id"
                },
                {
                    "name": "attr/lastannounced"
                },
                {
                    "name": "attr/lastseen"
                },
                {
                    "name": "attr/manufacturername"
                },
                {
                    "name": "attr/modelid"
                },
                {
                    "name": "attr/name"
                },
                {
                    "name": "attr/swversion"
                },
                {
                    "name": "attr/type"
                },
                {
                    "name": "attr/uniqueid"
                },
                {
                    "name": "config/battery",
[... 282 lines cut ...]
                    "at": "0x4002",
                    "dt": "0x18",
                    "mf": "0x1037",
                    "min": 1,
                    "max": 3600
                },
                {
                    "at": "0x4003",
                    "dt": "0x29",
                    "mf": "0x1037",
                    "min": 1,
                    "max": 300
                },
                {
                    "at": "0x4008",
                    "dt": "0x22",
                    "mf": "0x1037",
                    "min": 1,
                    "max": 3600
                }
            ]
        }
    ]
}
```

LukasA83, 2025-10-03:
> As for `state/valve` and the "parsing issue", that is more or less a safeguard. Many thermostat DDFs do not reflect what has been put in the code back in the days. The thermostat cluster is kind of [...]

Is there a way to start deconz always in strict mode?

SwoopX, 2025-10-05:
> 

> For example, isn't this also necessary?
> 
>       "meta": {
>         "values": {
>           "config/mode": {"auto": 0, "heat": 1, "off": 5}
>         }
>       },

Good catch @mattreim. Yes, this part indeed needs to be added, as setting the mode wouldn't work otherwise.

> Is there a way to start deconz always in strict mode?

@LukasA83 No, I don't think so. There is a parameter to start in hybrid mode, however, it's been ages since I played around with it and I don't recall how it behaves https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/deCONZ-command-line-parameters

> Alternative, do we still need this in thermostat.cpp? It should be fully covered by DDF.

Unless the DDF is committed, this code must remain. Once done, most of the Spirit related code can go. However, some device specific code obviously must remain, as the firmware has issues.

As this is not the suited means to track additions/changes on a to be included DDF, I'd appreciate the creation of a corresponding PR to follow up. Meanwhile, I'll close the issue, thanks in advance.

closed 2025-10-05 by SwoopX