dresden-elektronik/deconz-rest-plugin #8335: "HZ-Smart-Device HaiHao Shenzhen Electronic" Smoke Detector

ICarryTheDustOfAJourney, 2025-09-30:
### Is there already an existing issue for this?

- [x] I have searched the existing issues and there is none for my device

### Product name

HZ-Smart-Device

### Manufacturer

HaiHao Shenzhen Electronic

### Model identifier

unknown

### Device type to add

Sensor

### Node info

[image]

[image]

### Endpoints and clusters

[image]

### Basic

Its a smoke detecor, which pairs to my ConBee II,  shows up in deConz, but not in PhosCon

Physical label photo:

[image]

### Further relevant clusters

### Power Configuration

### On/Off

### Level Control

### Color Control

### Thermostat

### Simple Metering

### Electrical Measurement

### Any other cluster of relevance/interest

ICarryTheDustOfAJourney, 2025-09-30:
Is it this what you asked for? (I activated the device's node and went to "Cluster Info")

[image]
[image]
[image]

ICarryTheDustOfAJourney, 2025-09-30:
Thanks a lot! 

Maybe I'm doing it wrong, but when I try to open it in deConz using Ctrl+E -> "File" -> "Open" it says "Failed to open C:/..." and no change.

I have reset the "don't open files downloaded from internet" flag and can open the file in my Notepad++

How do I create a file (.zip, .ddf etc) required for Phoscon, btw?

mattreim, 2025-09-30:
You can also upload this DDB directly to Phoscon.

[photoelectric_smoke_sensor.zip](https://github.com/user-attachments/files/22621728/photoelectric_smoke_sensor.zip)

ICarryTheDustOfAJourney, 2025-09-30:
That worked, thanks a lot! You saved my day.

Was not far from giving up, now I consider buying more of those.
They detected at least my incense stick's smoke quite reliably, are small and easy to connect now.

Just in case someone in the future find this:

- manufacturer id is _TZE284_rccxox8p, product id TS0601
- install the DDF file provided above first, for now.
- just press the button for like 5 seconds, start the pairing function in phoscon and wait for the next beep, red LED stops to blink when successful
- a flame appears as a symbol for the smoke detector in Phoscon

closed 2025-09-30 by ICarryTheDustOfAJourney